jgromes/LoRaLib #1: dublex support?

halukmy, 2017-12-20:
is it?

jgromes, 2017-12-20:
This is a terrible way to create an issue. What exactly do you want me to do? Did you misspell the word "duplex"? Or do you want to implement some new feature?

Please try to provide as much details as you can when creating an issue. It will speed up the whole process.

halukmy, 2017-12-20:
you speak much, just a little bit shut up

just learn "duplex" what means?

tx and rx same time

jgromes, 2017-12-20:
And you are being very rude and unhelpful.

With SX127x chips, you can't transmit and receive at the same time, that's not the way the chip works.

closed 2017-12-20 by jgromes

halukmy, 2017-12-20:
sorrry for being rude but you being rude too..

i dont mean same time transmitting

rx for addressing send send message

also recieve while income

jgromes, 2017-12-20:
Quoting from [SX1278 datasheet](http://www.semtech.com/images/datasheet/sx1276_77_78_79.pdf) (page 23): "SX1276/77/78/79 is a half-duplex, low-IF transceiver."

This is not an issue of the library, SX127x chips can't receive and transmit at the same time. This library is simply a control interface for the chip, running on Arduino platfrom. I can not implement something that the chip can't do.