jgromes/LoRaLib #2: Teensy 3.2 support

Sjord, 2017-12-21:
I have used this library with some success on a [Teensy 3.2](https://www.pjrc.com/store/teensy32.html). The [SPI pinout](https://www.pjrc.com/teensy/td_libs_SPI.html) is the same as for the Arduino Uno. I changed the line in [Module.h](https://github.com/jgromes/LoRaLib/blob/master/src/Module.h#L39) to include a Teensy definition:

    #if defined(__AVR_ATmega168__) || defined(__AVR_ATmega328P__) || defined(__MK20DX256__)
      int _sck = 13;
      int _miso = 12;
      int _mosi = 11;

So the `__MK20DX256__` is new here. I got constant's name this by enabling verbose output during compilation:

    "arm-none-eabi-g++" ... -mcpu=cortex-m4 ... -D__MK20DX256__ -DTEENSYDUINO=130 -DARDUINO=10611 -DF_CPU=72000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH ...

I don't know for sure whether `__MK20DX256__` is the correct constant to check on.

I used the [TX example](https://github.com/jgromes/LoRaLib/blob/master/examples/LoRaLib_Tx/LoRaLib_Tx.ino) and verified that it sent something using an RTL-SDR.

Sjord, 2017-12-21:
One other thing I found out that with the Teensy I have to add a delay after `Serial.begin`, or the prints don't show in the serial monitor:

```
void setup() {
  Serial.begin(9600);
  delay(500); 
```

I think things are printed to the USB port while it hasn't switched from programming mode to serial mode yet.

jgromes, 2017-12-21:
The library uses Arduino SPI and EEPROM libraries. If the ones on Teensy have the same API, the library should work just fine.

I think that the `__MK20DX256__` should be the correct one. I don't have any experience with Teensy, but if it was wrong, compilation should end with error `"Unsupported board selected, please select Arduino UNO or Mega"`.

I haven't tested the library with Teensy (since I don't have one). It would be nice to add Teensy support, could please test both examples, to see if data are transmitted successfully?

Sjord, 2018-01-03:
Module.h tries to determine which board you have, in order to determine which SPI pins to use, in order to set the right pinMode (input or output) on it. However, I am not sure this is necessary. It seems that the Arduino SPI library already knows the SPI pins, and setting the correct pinMode is not needed. Maybe this detection code can be removed, making this library work on any board with an SPI interface.

jgromes, 2018-01-03:
Just checked the SPI library, you're absolutely right. That check is redundant, the correct pins are saved in file pins_arduino.h for all supported boards and the pin mode is set in SPI.cpp.

I'll remove the check and verify that the library still works on Uno and Mega boards.

jgromes, 2018-01-03:
Works on Arduino Uno and Mega. Thanks again, please let me know if the library works on Teensy now.

Updated in commit 0f444a50ac4e7956a14aae034fe654dee044344d

Sjord, 2018-01-04:
At least it compiles and `lora.begin()` succeeds. So that seems to me like it succesfully communicated over SPI. I was unable to confirm whether radio communication works.

jgromes, 2018-01-04:
You can enable debug output (uncomment #debug definition in [TypeDef.h](https://github.com/jgromes/LoRaLib/blob/master/src/TypeDef.h#L10)) and Serial output should look something like this:

```
LoRa node address string: <randomly generated address here>
SX1278 not found! (1 of 10 tries) SX1278_REG_VERSION == 0x00
Found SX1278! (match by SX1278_REG_VERSION == 0x12)
```

Then you can be sure that the SPI communication works and register SX1278_REG_VERSION contains correct value (0x12).

One other thing: what is your wiring setup, especially DIO0 and DIO1 pins on your SX1278 module? By default, the library expects DIO0 connected to Arduino D2 and DIO1 to Arduino D3. These have to be connected for the transmit/receive functions to work correctly.

BlueCoal69, 2018-02-05:
Is there a way to map the nss pin to the SS pin that's in pins_arduino.h and not ask for it in the LoRa function? Would that work?

jgromes, 2018-02-05:
The SS pin in pins_arduino.h is used as Slave Select signal when Arduino is operating as a slave on that bus, and that is not this case. The basic SPI interface is implemented in hardware of the MCU on all Arduino boards and tied to exact pins, so I would suggest not messing with the configuration in pins_arduino.h.

Side note: I think you might be commenting on the wrong issue ;)

BlueCoal69, 2018-02-06:
Correct, if the MCU is operating as the slave the SS is the correct pin to use. But in most cases the MCU is the master talking to the slave (SX1278) and the SS may be used to drive signal low before sending or you can use other I/O pins for that task. Right?

As you noted above there are files that keep the pin definitions for the boards usually in the pins_arduino.h file. My question/idea was not to change the pin definition file but for the code to look at it (somehow) and get the pin number for SS and use it for NSS. This would work fine if its only 1 to 1 with the MCU as the master.

The way you have it now gives more flexibility to the users.

jgromes, 2018-02-06:
The main problem with that approach is that it would prevent you from using the Arduino as SPI slave. You are correct that the usual use case will be 1 to 1 communication Arduino - SX1278, but I can't assume that will always be the case.

The library was originally written for LoRenz shield, which has NSS connected on Arduino pin 7  - that's why that is used as the default setting. Fundamentally, you would only change the default pin from 7 to pin defined as SS in the header file. It is presumed that users will want to use different pins as NSS - that's why it is the second parameter in the LoRa constructor, you can set the chip type, the NSS pin and leave the rest as default.

Flexibility is one of the core aims for this library, so I'd rather keep it the way it is set up now.

closed 2018-05-16 by jgromes