jgromes/LoRaLib #4: ESP32 Support

BlueCoal69, 2018-01-27:
I have been trying to get the libr to work with a TTGO/HelTec  ESP32 SX1278 module. It seems to be init ok using the provided examples after getting the pins set

LoRa node address string: 0:0:0:0:0:0:0:0
Found SX1278! (match by SX1278_REG_VERSION == 0x12)
Initialization done.

This may be just confusion on my part.  The TX example is sending 01:23:45:67:89:AB:CD:EF for a destination but the RX example doesn't seem to set its node address to 01:23:45:67:89:AB:CD:EF to match and ends up with 0's for its node address string.

jgromes, 2018-01-27:
It's definitely not confusion on your part: the packet addresses are not compared by the library, that's up to the user.

I think the problem might be the interrupt pins. Currently, the library uses both DIO0 and DIO1 pins on SX1278. I'm not sure both of those are available on your module. Can you confirm that?

On a side note: It is strange that the node address string seems to be set to 0:0:0:0:0:0:0:0, this should be generated randomly and then written into EEPROM. Then again, I'm not sure how to use EEPROM on ESP32.

BlueCoal69, 2018-01-27:
Thanks for the pointer. I fixed the EEPROM issue.  I had to init the eeprom area

LoRa node address string: D6:AA:A9:3E:28:C2:51:F9
Found SX1278! (match by SX1278_REG_VERSION == 0x12)
Initialization done.

But you seem to be right on the pins. It looks like I may not have access to DI01.

// GPIO5  -- SX1278's SCK
// GPIO19 -- SX1278's MISO
// GPIO27 -- SX1278's MOSI
// GPIO18 -- SX1278's CS
// GPIO14 -- SX1278's RESET
// GPIO26 -- SX1278's IRQ(Interrupt Request)

#define SS      18
#define RST     14
#define DI0     26

I'll post back if I a solution

BlueCoal69, 2018-01-27:
Found a post where DI01 is on pin 33. I'm getting a packet after resetting the module sometimes but never more than one.

LoRa node address string: 46:BC:F3:38:3B:52:3A:8A
Found SX1278! (match by SX1278_REG_VERSION == 0x12)
Initialization done.
success!
Source:		00:00:00:00:00:00:00:00
Destination:	46:BC:F3:38:3B:52:3A:8A
Length:		28
Data:		Hello World!
Datarate:	512.59 bps
RSSI:		-44 dBm

jgromes, 2018-01-28:
The serial output of the receiving module should look something like this:

LoRa node address string: 46:BC:F3:38:3B:52:3A:8A
Found SX1278! (match by SX1278_REG_VERSION == 0x12)
Initialization done.
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... success!
<packet data>
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!

Did you just remove all "Waiting for incoming transmission ... timeout!" output, or was it not there in the first place?

I suppose you're transmitting the packets from another HelTec module, could you check that it actually sends them? The transmitter example code should be sending a packet every second (roughly), so the receiver should report some timeouts.

BlueCoal69, 2018-01-29:
For some reason the "Waiting for incoming transmission ...  got commented out and the timeout never comes. So if I am reading the lib correctly that tells me that DIO1 must not be connected to pin 33 on my module. I may be getting the one packet thru just on the timing of the module reset.

I had also purchased two LoRa-02 modules from Aliexpress but they seem to be DOA.  I will try again after getting a "working" non-intergrated Lora module. Thanks for your help

BlueCoal69, 2018-02-04:
The lib works great with ESP32 and the arduino-esp32 libraries with the RA-02 SX1278 from (https://www.aliexpress.com/item/Elecrow-2pcs-lot-SX1278-LoRa-Module-433M-10KM-Ra-02-Ai-Thinker-Wireless-Spread-Spectrum-Transmission/32817296119.html?spm=a2g0s.9042311.0.0.01tlkh) 

The only change besides the pins #s was to initialize the EEprom area in the LoRaLib.cpp under LoRa::begin with some size (1000 was way more than needed). This was needed per the ESP32 SPI lib.

uint8_t LoRa::begin(uint16_t addrEeprom) {
  #ifdef DEBUG
    Serial.begin(9600);
    Serial.println();
  #endif

  if (!EEPROM.begin(1000))
  {
    Serial.println("failed to initialise EEPROM");
  }

The TTGO integrated modules work fine with the other libs (like arduino-LoRa) that do not use DIO1. 
Thanks

jgromes, 2018-02-05:
I added the EEPROM.begin and DIO setting in the latest commit, could you please check that it works? I would have done it myself, but I don't have any ESP32 board, thanks.

BlueCoal69, 2018-02-06:
Worked great!! Just need to remove leading underscore on dio1 in SX1278.cpp

SX1278::SX1278(int nss, Bandwidth bw, SpreadingFactor sf, CodingRate cr, int dio0, int dio1) {
  _nss = nss;
  _dio0 = dio0;
  _dio1 = _dio1;  <---------

jgromes, 2018-02-06:
Fixed the typo in 96a9181d96e312e29e72c6acbbed7820645c05c2, thanks for testing it.

I'll probably add a new example explaining the settings and leave the basic ones as they are.

BlueCoal69, 2018-02-06:
Do you have any current plans to add to the lib.  I saw your to do CAD note. Any need to use RXCONTINUOUS mode?

jgromes, 2018-02-06:
I might implement the CAD mode at sime point in the future, although it's not my priority right now. As far as RxContinuous mode is concerned, I'm not sure it's entirely necessary, I might implement it if I find some use for it.

At this point, I think it's time to release this library as version 1.0 and get it included in Arduino Library Manager, to allow easy installation.

BlueCoal69, 2018-02-12:
Thanks, I think I'll try adding RXContinuous as part of my project.  I have a remote building sending sensor changes only when needed. So it could be 1 minute or ten days between transmissions (may add heartbeat signal to make sure its working). For this RXContinuous would free up the micro-controller more to do other things since it does not need to keep resetting after RXTimeout.

jgromes, 2018-02-12:
Cool, if you manage to get it working, start a pull request, so that we can merge it into this project :)

closed 2018-02-12 by jgromes