jgromes/LoRaLib #5: Hi, I am using two SX1276RF1IAS-ND modules o ne as a transmitter and one as a receiver.

kayani009, 2018-02-09:
Hi, I am using two SX1276RF1IAS-ND modules o ne as a transmitter and one as a receiver. I want to transmit the data and receive the data only.
can any one guide how to configure this module as a transmitter and as a receiver to send and receive the data.
thanks

jgromes, 2018-02-09:
To transmit a packet, call the lora.transmit function. To receive one, call the lora.receive function. Please refer to the example code provided in /examples/ directory. For a more detailed API reference, there's the [wiki](https://github.com/jgromes/LoRaLib/wiki).

kayani009, 2018-02-09:
I need your help bro. Thanks

jgromes, 2018-02-09:
You really don't, using the library is just as simple as using Serial.println and Serial.read. If you run into some trouble, post your code and we can discuss it. You can try modifying the examples so that they suit your needs. But please, at least try first.

kayani009, 2018-02-10:
but before sending ad receiving the data i need to configure the
module by using configuration  registers but i dont have any idea
about that how to configure the registers to set frequency, bandwidth
and spreadinf factor and others elements.

jgromes, 2018-02-10:
You don't have to set anything, unless you want settings other than the default ones. If you want to adjust range and datarate, [this wiki article](https://github.com/jgromes/LoRaLib/wiki/LoRa-modem-settings) has the information about that.

jgromes, 2018-02-10:
Try running the example sketches, and if you want to set non-default settings, pass them to the Lora calss contructor, like so:
`Lora lora(CH_SX1276, 7, BW_250_00_KHZ, SF_8, CR_4_6)`
This will force SX1276 chip with NSS connected to Arduino pin 7, bandwidth 250 kHz, spreading factor 8, coding rate 4/6. All of that is a part of the documentation [here](https://github.com/jgromes/LoRaLib/wiki/LoRa-Class).

kayani009, 2018-02-10:
Thanks i will try and let you know

kayani009, 2018-02-10:
If i use default setting so will i connect to Miso Mosi and Ck pin or left
it unconnected only use Tx/Rx?

jgromes, 2018-02-10:
LoRaLib uses SPI to connect to the module, so use MOSI, MISO, clock and slave select pin.

kayani009, 2018-02-10:
Thanks a lot

closed 2018-02-10 by jgromes

kayani009, 2018-02-12:
hi bro
i have tested this program thee are no error at all but it is not sending
the data.
which pins are to be connected to transmit the data.

jgromes, 2018-02-12:
You can set all the connections however you like - see the Settings example. The default is Arduino pin 7 as SPI slave select, Arduino pin 2 as SX1276 DIO0 and Arduino pin 3 as SX1276 DIO1.

kayani009, 2018-02-12:
I have to connect Sx1276 DIO0 pin with Arduino pin 2 and S1276 DIO1 pin
with arduino pin 3 .
Am i right?

jgromes, 2018-02-12:
That's the default setting - so yes if you use `LoRa lora;`
However, as I've told you multiple times already, you can customize it. See the Settings example.

kayani009, 2018-02-12:
Sorry bro my aim is to send and receive the data by using default setting.
If it works then i will change the setting.
The program stuck on sending packet and nothing happens after that.

jgromes, 2018-02-12:
Can you post/attach your code? It is impossible to debug something I can't see...

kayani009, 2018-02-12:
I am using your code in examples transmit and receive.

jgromes, 2018-02-12:
In that case there is probably something wrong with your wiring. You should check all the connections.

kayani009, 2018-02-12:
i connect  pin of SX1276  VR_ANA,  VBAT_ANA,  VR_DIG,  VBAT_RF,  VR_PA to
3.3 V
DIO0  with arduino pin 2
DIO1/DCLK  with arduino pin 3
pin 15,23,26 of SX1276 to ground
MOSI of sx1276 to pin 11 of arduino
MISO of sx1276 to pin 12 of arduino
sck of sx1276 to pin 13 of arduino
NSS of Sx1276 with pin 9 of arduino
i am using ardiuno uno board

kayani009, 2018-02-12:
i didnt connect pin 20 RX/TX to arduino

jgromes, 2018-02-12:
Why are you connecting NSS to Arduino pin 9? I told you the default setting is pin 7, so try that one instead.

kayani009, 2018-02-12:
i try that as well but not working

kayani009, 2018-02-12:
RFI_HF (GND)  not sure about this pin????

jgromes, 2018-02-12:
What is the logic level of your module? SX1276 by itself is 3.3 V only, so unless there are level converters on your module, you might have blown the SX1276.

kayani009, 2018-02-12:
but there is a power supply of 3.3 v on arduino board i am using that tom
power up SX1276.

jgromes, 2018-02-12:
Logic level has nothing to do with the power supply. Arduino Uno has 5 V logic - if you set one of the digital pins high, it will have a value of 5 V. SX1276 has 3.3 V - logic high is 3.3 V. If you apply 5 V directly to a 3.3 V logic device, there's a good chance you will destroy the chip.

kayani009, 2018-02-13:
Thanks bro.
I have ordered logic level converter. But how to check whether my chip is
destroyed or not?

jgromes, 2018-02-13:
No idea, I don't know this specific board,  check with your distributor or the board manufacturer. But the logic inputs are probably connected directly to the chip, and since you applied 5V repeatedly, I'd say the probability of the chip being damaged is fairly high.

I suggest that you first learn a bit more about Arduino and electronics in general, there's plenty of tutorials online. Once you have that, you can try using more advanced modules like this one.

kayani009, 2018-02-22:
Hi,
its working now but the receiver is giving CRC Error. Transmitter is
working fine. can you please help?

jgromes, 2018-02-24:
CRC error means that the CRC16 checksum of the received message does not match the expected CRC16 checksum - the packet was damaged while on-air. That can be either incorrect LoRa modem settings (bandwidth, spreading factor and coding rate of transmitter and receiver don't match), or not enough power for the module. The usual power requirements of SX1276 is low enough for the Arduino regulator to handle, but since this one is on a breakout board, there might be other components that increase it.

First of all, check all the settings and maybe try a different combination. If it still reports CRC error, get a 3.3 V regulator that can provide at least 1 A and power the module from that.

jgromes, 2018-02-25:
Look in the SX1276 datasheet: on page 11, RXTX pin is marked as output. I suspect it goes high when the module is transmiting. The library doesn't use this pin, because it is not routed on the modules this library is written for, like [this one.](https://www.ebay.com/itm/433-LoRa-SX1278-Long-Range-RF-Wireless-Transceiver-Module-SX1276-5Km-For-Arduino/142465838566?epid=2259922846&hash=item212ba02de6:g:JN8AAOSwP6pZn9vc)
I would suggest that you get simple LoRa module like the one in the link and try with that. Only then use more complicated modules.

kayani009, 2018-02-26:
Hello brother,
I have checked 3.3v pin on Arduino board and it gives me 2.45V on DMM and
module can operate at 1.7 V as well.
Can you suggest some other library that i can use?

jgromes, 2018-02-26:
Are you measuring the voltage with the module connected? If so, your reading is very inaccurate - the regulator voltage output voltage will drop because of the load.

You can try [this library](http://bfy.tw/GoII).

kayani009, 2018-02-26:
Which libaray sandeepmistry one?
Is there any connection difference between transmitter and receiver?

jgromes, 2018-02-26:
What do you mean by "connection difference"? I told you about five times already - I don't know your specific module.

kayani009, 2018-02-26:
brother i need to only transmit data from transmitter and receive the data
from the receiver. no gateway is involved here i am not using LORAwan. its
simple transmitter and receiver. will LORa support this?

jgromes, 2018-02-26:
I don't think you understand the purpose of GitHub Issues - this is not a forum. GitHub issues are meant to suggest improvements, report bugs, or ask the repository authors specific questions about the code. Your questions do not fall into any of these cathegories, so please go to Arduino forums to ask these very basic questions there or better yet - try it yourself.

I am locking this conversation now. Good luck with your project.

PS: Do not call people you have never met "brother". It comes out as somewhat weird.