jgromes/LoRaLib #6: SX1278 1742 161510

jakecorn, 2018-03-04:
Hi please help me with my problem, I have two transceivers  SX1278 1742 161510  and SX1278 1733 151144 but my first transceiver will only work as receiver. I tried to upload the Trasmit.ino, it says initialization done but it is stuck on sending packet....

When I tried to upload the Receive.ino to the first transceiver, it works. My second transceiver able to send but i haven't tried it to receive since my other transceiver will not work as sender.

They have different numbers in the chip.
1st transceiver:  
**SX1278**
     1742
     161510 

2nd Transceiver:
**SX1278**
    1733
    151144
 I followed this connection
// create instance of LoRa class with default settings
// chip:                SX1278
// NSS pin:             7
// bandwidth:           125 kHz
// spreading factor:    9
// coding rate:         4/7
// DIO0 pin:            2
// DIO1 pin:            3

and since it will not work to have just the above connection , added 

| SCK | SCK |
| MISO | MISO |
| MOSI | MOSI |

This is the module that I'm using just don't mind the number in the chip
[image]

This is the arduino board I'm using

[image]

I'm not sure if the first transceiver was designed to be just a receiver not a sender.
Please help me.

Thank you.

jgromes, 2018-03-04:
The numbers on the chip are just the date of production, e.g. the first one (1742 161510) was made during the 42nd week of 2017, lot number 161510. This is pretty much irrelevant for now.

Try enabling the debug output (uncomment the debug definition in [TypeDef.h](https://github.com/jgromes/LoRaLib/blob/master/src/TypeDef.h#L10)) and see what shows up on the serial monitor. Also check all the connections - especially the DIO0, since the library uses that to check if the packet was sent. If there's a faulty connection, that signal will never arrive.

jakecorn, 2018-03-04:
thank you for the reply.  I already uncommented the debug definition and i got this

LoRa node address string: 1A:22:E8:B8:FF:FF:FF:FF
Found SX1278! (match by SX1278_REG_VERSION == 0x12)
Initialization done.
Source:		1A:22:E8:B8:FF:FF:FF:FF
Destination:	01:23:45:67:89:AB:CD:EF
Length:		28
Data:		Hello World!
Sending packet

still stuck on sending packet. I already checked DIO0. thank you

jgromes, 2018-03-04:
Does it print out a whole bunch of dots like this?

`Sending packet ................................................` (and so on forever)

If so, it is stuck in an infinite loop waiting for DIO0 signal to arrive.

jgromes, 2018-03-04:
The way it is written now the library waits for the DIO0 to go high which signals that packet was transmitted. However, if the signal never arrives, it will be stuck there forever. I will add a timeout to that loop so that after waiting long enough, the loop will terminate and won't get stuck this way.

This doesn't solve your issue though. Just to make sure: is the SX1278 module the only thing that is connected? It would help if you could get an oscilloscope to see if there's a signal coming out of DIO0 when transmission is finished.

jgromes, 2018-03-04:
One more thing: how is the SX1278 module connected?

jakecorn, 2018-03-04:
I followed this connection
// create instance of LoRa class with default settings
// chip: SX1278
// NSS pin: 7
// bandwidth: 125 kHz
// spreading factor: 9
// coding rate: 4/7
// DIO0 pin: 2
// DIO1 pin: 3

and since it will not work to have just the above connection , added

| SCK | SCK |
| MISO | MISO |
| MOSI | MOSI |

Actually im testing it now. when i tried to remove DIO0 is says success but let me check if it really sends data, thank you for quick reply.

jgromes, 2018-03-04:
What I meant is how is it connected physically: did you solder wires to the module, or do you have some connectors?

jakecorn, 2018-03-04:
Yes, i solder wires directly to the module

jgromes, 2018-03-04:
I still suspect faulty wiring: it's odd that the module seems to be running fine, except the DIO0 signal. Try carefully resoldering the connection on DIO0.

jakecorn, 2018-03-04:
Okay I will try that. Is there any other way that  I can use my module to send and receive without using DI0O?

jgromes, 2018-03-04:
Unfortunately not: the DIO0 is used to signal both transmission _and_ reception end; see SX1278 [datasheet](https://www.semtech.com/uploads/documents/sx1276_77_78_79.pdf), page 46, table 18.

jakecorn, 2018-03-04:
Okay, thank you so much for the reply. I will update you after i re-solder the wires. thank you.

jgromes, 2018-03-09:
I added Tx timeout in the latest commit d38797213e37019ebfe3f4316b676e337f898f5c. I'll also release a new version of the library.

Right now, Tx timeout only depends on packet length - it will wait 1.5 seconds per byte of packet length. That should cover the worst case scenario with lowest datarate - 7.8 kHz bandwidth, spreading factor 12 and coding rate 4/8. It would be much better if the timeout was calculated dynamically based on the modem settings, I'll probably implement that later.

jgromes, 2018-03-09:
Just published [Release 1.0.1](https://github.com/jgromes/LoRaLib/releases/tag/1.0.1) with the Tx timeout. If you're using Arduino library manager, it should let you update the library in a few hours.

jakecorn, 2018-03-09:
Thank you for the update, actually I still not able to fix my problem. I have another question, if  DIO0 is used to signal both transmission and reception, does the module is using that pin during receiving a signal?if so, then the problem is not in the wire which not properly soldered.

Also, with regards to setting the bandwidth, what is the advantage or disadvantage of setting the module to high or lowest bandwidth especially with regards to the distance that the data can be transmitted or the size of the data that can be transmitted.

Another thing, I successfully modified your module specifically on Packet(const char dest[24], c**onst char dat[240]**). Instead of using char in the second parameter I changed it to String because I find it hard to manipulate char, is it a bad idea?

THANK YOU.

jgromes, 2018-03-09:
1. DIO0 is used to signal Rx done or Tx done. If you can successfully receive data with the module that is not transmitting, then you are correct and the problem is not the wiring.

2. See the [wiki](https://github.com/jgromes/LoRaLib/wiki/LoRa-modem-settings) page. Basically, higher bandwidth means higher data rate, but lower range. The size remains the same, 256 bytes (16 bytes of addresses, 240 bytes of payload) is the maximum that will fit in a single packet.

3. Personally, I try to avoid Arduino String class whenever I can, it can be really unpredictable sometimes and it uses large amounts of RAM - of which there's never enough on Arduino.  I highly recommend that you learn the basics of C-strings. It's much easier than it looks - basically just working with an array of characters. Besides, you can always get the C-string out of Arduino String by calling the `.c_str()` method. So if you want to create a LoRaLib packet using some String data, you can do this:
```
String myString = "text";
Packet pack("01:23:45:67:89:AB:CD:EF", myString.c_str());
```

jakecorn, 2018-03-09:
Thank you for information,
High likely that its not the wiring, so, what could be possibly the
problem? the module itself?, Does the module sensitive heat caused during
soldering?
Another thing, I'm worried about the other module, the one that is working
in receiving and transmitting, When I use the module for 15-20 minutes as
transmitter, it gets so hot and I notice the other side of the board gets
darken,seems like a sign of overheating, is it normal? I notice that the
transmitter easily gets hot than the receiver. I connected the module to
the 3.3v pin of arduino, is this a good practice?
I'm just curious about the transmission range of the module, is it possible
to transmitter at the distance of 5km? I haven't seen any documents  that
supports that claim,it's only the  seller says that the range is 5km.

I really appreciate for taking time to answer my queries. Thank you so much.

jgromes, 2018-03-09:
I've never encountered overheating issues with these modules - in fact, LoRa modulation was designed for low-power systems. You should check the current draw on your module, looking through the datasheet, it should not draw more than 90 mA when transmitting. The 3.3 V regulator on Arduino should be able to handle that. If the modules draws significantly more, it is possible that both the modules are faulty. Try to get new ones from a different seller.

The range depends on many different factors, some of them are:

1. LoRa modem settings - bandwidth, spreading factor and coding rate. If you optimize all of those for the highest range possible, the data rate will drop to just a few bits per second.

2. The antenna you are using. If you have no antenna conneted to the module, the range will be much lower.

3. The type of terrain between the transmitter and the receiver. If you're transmitting in a city, the range will obviously be much shorter than in open space. Weather will play a role as well.

Under ideal conditions, with no additional amplifier between the module and antenna, I'd say you can push the range up to 2 - 3 km. While testing the library, I got to about 500 m with the highest bandwidth and a small 10 cm antenna for both the transmitter and the receiver.

I'm going to close this issue for now, feel free to re-open it if/when you get the new modules and if you run into trouble again.

closed 2018-03-09 by jgromes

jgromes, 2018-07-12:
I have to advise against using LoRa modules to control RC plane - or anything that requires pretty much instant reactions. The whole point of LoRa modulation is maximum range with minimum power requirements. This is achieved by lowering data rate. You can see an extreme example here: https://www.youtube.com/watch?v=nfH6tO0VnJc This is a recording of LoRa transmission with the maximum possible range. Transmitting 28 bytes takes almost 30 seconds.

The highest data rate you can get from LoRa is 37.5 kbps (according to the datasheet), but at that point, you have basically defeated the point of the modulation, because the range will drop drastically.

I recommend trying out different wireless modules - XBee, for example. Or better yet, why reinvent the wheel? You could use standard RC receiver to control the plane, and then some other wireless module for non-critical communication (data from sensors etc.).

jgromes, 2018-07-12:
You could probably use RFM69. These modules are similar to Semtech LoRa modules, but don't use LoRa modulation, and can reach data rates up to 300 kpbs. I don't know what's the range, though.

No worries ;) If you encountered a bug or a problem, please open a new issue, we can discuss it there.

jakecorn, 2018-07-21:
Hi, i just want to share that my 2 lora module are now capable of sending and receiving. What i did , i just get the source address (module 1) and use it for the destination of the packet of the module 2. thank you so much.

jgromes, 2018-07-21:
Did you mean to post this at issue #18?