jgromes/LoRaLib #8: Set Frequency

athtest800, 2018-05-19:
Hi!
How can I set the frequency it transmit?
Thank you.

jgromes, 2018-05-19:
Sorry, LoRaLib doesn't support frequency setting yet, it will be implement in the next release sometimes during the next month.

In the meantime, you can use the following workaround:

1. In LoRalib.h, line 35, move `Module* _mod;` from private methods to public.
2. This will allow you to directly access SPI functions, so you can set the frequency manually in the Arduino sketch, like so:
```
lora._mod->setRegValue(SX1278_REG_FRF_MSB, 0x6C);
lora._mod->setRegValue(SX1278_REG_FRF_MID, 0x80);
lora._mod->setRegValue(SX1278_REG_FRF_LSB, 0x00);
```
This will set the carrier frequency to 434 MHz. The frequency value in MHz is calculated according to the equation

f = (32 * SX1278_FRF_VALUE) / 2^19

where SX1278_FRF_VALUE is the 3-byte value set into the registers (0x6C8000 in the above example).

renatocarusos, 2018-06-08:
@jgromes This library is compatible with 915Mhz modules? If yes, in this method, whats the hex value in "lora._mod-> setRegValue" to set 915Mhz, because I loaded the standard sketch. Did I mess up my module with this?

jgromes, 2018-06-08:
The default carrier frequency is written to the module at startup, and depends on the type of chip - 434 MHz for SX1278, 915 MHz for SX1272. Which one are you using? If it's the latter, use `LoRa lora(CH_SX1272);` and the library will take care of the frequency setting.

If you try to set a value that is out of range for your module, it won't damage it. The module will try to transmit, you just can't predict the frequency. I tried forcing 434 MHz on 915 MHz SX1272 module, and the resulting transmission was at 946 MHz (not an ISM band, transmitting here might not be legal in your country).

You can get the register value using the following formula:

SX1278_FRF_VALUE = (f_rf * 2^19)/f_osc

where f_rf is your target frequency (915 MHz) and f_osc is the oscillator frequency (should be 32 MHz).
Plucking in the numbers for you: SX1278_FRF_VALUE = 0xE4C000, which is the default value for SX1272.

PS: Forcing 915 MHz on 434 MHz SX1278 worked fine. I'm not sure if there's any difference between modules labeled 434 MHz and 915 MHz, provided they are using the same chip.

renatocarusos, 2018-06-08:
I am using a 915MHz NiceRF SX1276 module which is the frequency allowed in my region. What do you think is better? Change the MSB MID LSB from SX1278.h to 0xE4C000, or set the object as CH_SX1272 ?  
I cant find information on functional differences between versions, but my intuition tells me to change the hex instead of setting it to 72. Ty for attention

jgromes, 2018-06-08:
Since SX1276 is a variation of SX1278, it is probably better to set the frequency registers.

I will add separate classes for all the SX127x chip variants in the next release, since there are some differences between the chips in the parameters they accept (different frequency/bandwidth ranges etc.).

renatocarusos, 2018-06-08:
Dude, thanks a lot for help, you're doing a great job. Let me know when you are coming to the Rio de Janeiro. You're welcome.

jgromes, 2018-06-17:
@athtest800 @renatocarusos Frequency setting is now implemented in the base-class branch, so you can fork that branch and check it out.

You can set frequency by calling method `LoRa::setFrequency(float f)`. The argument is a floating-point frequency in MHz, e.g. `LoRa::setFrequency(433.8)` to set frequency to 433.8 MHz.

Notes:

1. The frequency setting isn't exact - from what I can measure with SDR, the actual frequency is about 10 kHz lower than what you set. Also, the register itself has a fixed resolution (61 Hz according to datasheet) Not a huge problem, just something to be aware of.
2. Different modules have different accepted frequency ranges, if you try to set a frequency out of range for that module, you will get error 0x33.

jgromes, 2018-06-20:
Just released version 1.1.0 with all of the above implemented, closing this issue for now.

closed 2018-06-20 by jgromes