jgromes/LoRaLib #17: RST pin on SX12**

musgravehill, 2018-07-12:
What about RST pin on SX12**?  
Some users say that this pin should be On Off in some cases. Before or after initializations for robust work. 
And some libs use Dio0 only, but RST pin...

jgromes, 2018-07-12:
From SX1278 datasheet, it seems that the only function of RESET pin is to provide manual reset. I don't know how this would improve robustness, and I never encountered hardware instability. In any case, users can easily implement manual reset themselves - there's no need to add this to the library, in my opinion.

closed 2018-07-12 by jgromes

musgravehill, 2018-07-13:
May be, after several hours of continuous work it must be rebooted. But much easier way - is automatic rebooting microcontroller once a day and making a fresh initialization.

jgromes, 2018-07-13:
I can set up a little bit of a stress test - transmitting packet each second for a few hours, see if there are any issues with long-term stability. If it's stable, then implementing automated resets is beyond the scope of the library.

musgravehill, 2018-07-13:
I'm still waiting for the modules from the store. And then I will make the telemetry with packet each 3 seconds for a some years...

jgromes, 2018-07-13:
You should check on your local regulations then. Where I am, you cannot transmit more than 10% of the time (i.e. our device can only transmit 6 minutes out of every hour). 3 seconds between tranmissions might be too much.

PS: I'm running the stress test right now, currently at packet 6600(-ish) and counting :) No instability yet, I think we don't need reset pin.

jgromes, 2018-07-13:
So after nearly 10 hours and about 33 000 packets sent, everything is still running perfectly. The only unexpected thing that happened was that the `int` counter I used to add an ID to each packet had overflown at 32 767. It still sends packets with negative IDs just fine though :) 

I stand by my original opinion - there's no need to implement RESET support into the library. If the user needs it for some (unknown) reason, he will have to implement it himself.