jgromes/LoRaLib #18: Transmission stops after a several secons

jakecorn, 2018-07-13:
Hi, please help with my new problem using Lora Module SX1278. My previous problems are already discussed in separate thread, this may help you from determining the problem. https://github.com/jgromes/LoRaLib/issues/6#issuecomment-371758939

During the transmission after 30-50 seconds the transceiver will stop transmitting and it work again by resetting my arduino board. What do you think the cause of this? I tried to you the default code and still the same. I already tried my 2 modules but same problem. Is there any way to restart the module without resetting the controller?

Thank you.

jgromes, 2018-07-13:
1. What version of the library are you using?
2. What type of MCU? Arduino Uno/Mega/ESP8266/...?

jakecorn, 2018-07-21:
Hi, I just want you to know that this problem is now fixed. I had this problem because the instance of the packet class is created inside the loop() function but when i tried to transfer it out that function, it works well. i have another question. Currently im using arduino uno and I need to change the default pins  DIO0 pin:  2
DIO1 pin:  3 for lora SX1278?
In your sample  below, you are using SX1272 , but when i tried to use it using SX1278 lora = new LoRa(7,11,12), i got an error saying  "Initialization failed, code 0x1"

SX1272 loraSX1272 = new LoRa(6, 4, 5);

Please help me.thank you

jgromes, 2018-07-21:
You're probably using outdated version of the library - the Packet class was discontinued at v3.0.0. Please update and try if the current issue persists.

jakecorn, 2018-07-22:
okay I will redownload your library. is there any option in your code that allows me to use any digital pin for DIO0 and DIO1? thank you

jgromes, 2018-07-22:
It's that constructor:

`SX1278 lora = new LoRa(6, 4, 5);`

You were doing the right thing, you just can't use pins 13, 12 and 11 on Uno, because those are already used for SPI.

jakecorn, 2018-08-05:
Hi, Good evening. 

I need your help again.

I was trying to change the spreading factor to 6 using below code:

if (lora.setSpreadingFactor(6) == ERR_INVALID_SPREADING_FACTOR) {
    Serial.println("Selected spreading factor is invalid for this module!");
    while (true);
  }

I can successfully change the config, However, as per documentation the CRC needs to be disabled when using SF6, the problem is I don't know to disabled it. I see some codes in the library  to change the settings of the module using SPIsetRegValue() function. I tried the below code but i got an error saying "'class SX1278' has no member named 'SPIsetRegValue'". Can you provide an example to change the settings of the module?

 lora.SPIsetRegValue(SX127X_REG_MODEM_CONFIG_2,SX127X_REG_MODEM_CONFIG_2,2,2); 

OR

 lora._mod->SPIsetRegValue(SX127X_REG_MODEM_CONFIG_2,SX127X_REG_MODEM_CONFIG_2,2,2);

Thank you so much.

jgromes, 2018-08-05:
You can't access SPI manipulation methods, because they're private - public API is unable to access them, which is very much intended.

Also, you don't need to disable CRC manually - LoRaLib does it automatically when you set spreading factor to 6.

jakecorn, 2018-08-06:
Thank you so much for the reply.
Currently, I'm now playing the settings by combining the different configurations by changing the frequency,bandwidth, spreading factor and coding rate. What i want to achieve is fast data transmission with considerable distance since I'm using this to RC plane. As what you have said in the other thread, you don't recommend this module, however, I will to do some configurations and see how far this module can go.

Thank you again.