jgromes/LoRaLib #20: SX1272 fails to initialize

bibaswan, 2018-07-18:
**Describe the bug**
When trying to initialize the LoRa Semtech sx1272 board the program throws error code 0x1 after 2 seconds of showing initializing.
 
**To Reproduce**
Steps to reproduce the behavior:
run default transmit example using settings from screenshot or any other combination of pins for the NCC, DIO0, DIO1

**Expected behavior**
trying to run default example at 868 MHz but get error code 0x1

**Screenshots**
[image]
[image]

**Additional info (please complete):**
 - MCU: Arduino NANO ATmega328p
 - LoRa module type SX1272
 - Arduino IDE version  1.8.1
 - Library version 3.0.1

jgromes, 2018-07-18:
Could you try connecting your SX1272 to the default pins (NSS -> 7, DIO0 ->2, DIO1 -> 3)? Then try to run the example with default constructor `SX1272 lora = new LoRa;`

 If the problem presists there might be a fault in your wiring; if it goes away there's something wrong in the library.

bibaswan, 2018-07-18:
[image]

here is the output I have checked my wiring again and it seems to be all fine.

jgromes, 2018-07-18:
Thanks for the bug!

It is fixed in release 3.0.2. Since it will take a few hours until Arduino Library Manager detects the new version, you can just download the source from [here](https://github.com/jgromes/LoRaLib/releases/tag/3.0.2)

closed 2018-07-18 by jgromes

bibaswan, 2018-07-18:
thanks for the quick fix really appreciate it, you've helped out on a big project.