jgromes/LoRaLib #22: sx1276/77/78 errata

pheex, 2018-07-21:
Hi,

Are you aware of this ?
https://www.semtech.com/uploads/documents/sx1276_77_78-errata.pdf

jgromes, 2018-07-21:
When writing the library, I used revision 5 of the SX1278 datasheet, dated August 2016. This errata is dated September 2013, so everything in the datasheet *should* already be fixed. I'll double-check though. Thanks for the tip!

pheex, 2018-07-21:
The errata note still on the sx1278 ressource page 
https://www.semtech.com/products/wireless-rf/lora-transceivers/SX1278

And i don't see any use of register 0x36 and 0x3a (errata 2.1) in rev 5 datasheet, so i don't know if they managed errata another way.

jgromes, 2018-07-21:
2.1 - Seems like an easy fix.
2.2 - Errata doesn't provide any workaround.
2.3 - Seems fixable as well.
2.4 - LoRaLib doesn't use the counter, so no need to fix that.
3.x - LoRaLib doesn't use FSK modem, no need to fix that either.

It'll be fixed it in the next release, probably tomorrow.

pheex, 2018-07-21:
ok, thank you

jgromes, 2018-07-22:
Fixed in release 3.0.3. I tested fixes for 2.1 and 2.3, both should be working fine.

closed 2018-07-22 by jgromes