jgromes/LoRaLib #25: RX TX without freeze (interrupt on D2)

musgravehill, 2018-07-25:
I send and receive 14 bytes data every 15s. 
I put .receive((uint8_t*)&DATA_RAW, sizeof(DATA_RAW))  in LOOP, and Arduino froze by permanent rx_timeout.  
TX also causes freezing about 1 seconds. 
No buttons and OLED do not work on RX-TX-while-timeout...

May be, remove it?
`while(!_mod->getInt0State()) {
    if(_mod->getInt1State()) {`
and 
`while(!_mod->getInt0State()) {
    if(millis() - start > timeout) {`

And make interrupt D2 or polling DIO0 state on RX-TX-DONE... 
May be, you need to save the LoRa state: RX or TX until you get interrupt (or polling dio) state) and know is it RX-DONE or TX-DONE. 

It's just a suggestion. I apologize if there is something wrong.

jgromes, 2018-07-25:
This is literally the first item for my LoRaLib TODO list :)

The way LoRaLib is written now is called "blocking" approach - Arduino continually polls DIO0 and DIO1 states, so the program is "blocked" until packet is received, or timeout occurs.

Using interrupts is called "non-blocking" approach, which doesn't halt the program. I'm aware of this, I just haven't figured out the best way to implement this in the public API.

jgromes, 2018-07-25:
What do you think about this as the public API:

```c++
#include <LoRaLib.h>

SX1278 lora = new LoRa;

void setup() {
  Serial.begin(9600);

  // initialization with default settings
  int state = lora.begin();

  // set the function that will be called when new packet is received
  state = lora.onReceive(handlePacket);

  // start listening for LoRa packets
  state = lora.listen();
}

void handlePacket(String& str) {
  Serial.print("Received packet: ");
  Serial.println(str);
}

void loop() {
  // nothing here

}
```

This has the advantage of not changing the current `receive()` method, so it's backwards compatible. Also, the current `receive()` method doesn't need interrupts, so you can use any free pin for DIO0 and DIO1. The version with interrupts requires Arduino pins 2 and 3 for DIO, so it's less flexible. That's why I want to keep them both.

There will also be two different versions for the callback: one for Arduino `String` class and for arbitrary byte array.

musgravehill, 2018-07-25:
0. after each lora.send()  execute  lora.listen();  again for listening
1. `void LORA_interruptOnReceive(uint8_t dataSize){
   uint8_t dataSize_iNeed = 12; //12 bytes
   if(dataSize == dataSize_iNeed ){
       //get data, else skip
    }
}`
2. may be, you need d2 interrupt only? RX_DONE on dio0---d2    Other interrupts (TX_DONE, RX-TX-tomeout) skip?  

PS sorry for my english, Im study it, but not learn, because I don`t use it at a real life, I read books\datasheets only.

jgromes, 2018-07-25:
0. That is implied  - `listen()` method will put the radio to RxContinuous mode. It will stay there until the mode is changed, either by trying to transmit or by calling `standby()` directly. I'll explicitly mention it in the example code as well.
1. I'm not sure what you mean, can you explain it?
2. That is true - the library can (and does) calculate the expected time-on-air locally. That can be an improvement once the basic interrupt support is in.

musgravehill, 2018-07-25:
1.  I use struct someStructVar 12 bytes. So, I know valid datasize = 12.  OnReceive I compare received datasize and size that I expect. If they are equal, I will use received data, else skip this data. 

I suggest put the size of the received data to function OnReceive. May be OnReceive(receivedData, receivedDataSize). 
Or I can calculate it manually:
` uint8_t receivedDataSize = sizeof(receivedData);  
if(receivedDataSize  == sizeof(someStructVar)){
     lora.receive(&someStructVar, sizeof(someStructVar)); 
}

`

jgromes, 2018-07-25:
I'm going to have to change the planned API a bit:

```c++
// include the library
#include <LoRaLib.h>

// create instance of LoRa class using SX1278 module
SX1278 lora = new LoRa;

void setup() {
  Serial.begin(9600);
  pinMode(4, OUTPUT);

  // initialize SX1278 with default settings
  int state = lora.begin();

  // set the function that will be called when new packet is received
  lora.onReceive(handlePacket);
  
  // start listening for LoRa packets
  state = lora.listen();
}

void handlePacket() {
  String str;
  int state = lora.readData(str);
  Serial.println(str);
}

void loop() {
  // nothing here
}
```

What do you think about this version? Basically, user will always have to call `readData` method inside the callback function to get the packet data. Don't worry, you'll still get your `readData(uint8_t*, size_t)` overload ;)

musgravehill, 2018-07-25:
Jan, I like your version. It looks simple and flexible.

jgromes, 2018-07-25:
I added the support in fc547476cf32da0f47aa2d93c0fdbe490d0402f7 so you can test it with your setup. Let me know if you have some more ideas for tweaks. I'll release the next version (4.1.1) sometime tomorrow.

jgromes, 2018-07-26:
I just got another idea: in your original post, you said that transmitting takes about 1 second. The time-on-air depends on packet length and modulation settings, but in general use should not be longer than 1 - 2 seconds. 

However, under extreme circumstances, it can take quite a bit longer. The absolute worst case scenario is bandwidth 7.8 kHz, spreading factor 12, coding rate 4/8 and 256 bytes long packet - transmission takes 229 seconds.

What are the LoRa modem settings you are using? I could implement interrupt-driven transmit method, if that would be useful to you.

musgravehill, 2018-07-26:
I use "Lora modem calculator" for time on air calculation and etc. 
The config is: F 434, BW 125, CR 8, SF 12.  Payload size is 12 or 14 bytes. 

I'm afraid to use BW 62.5 or lower cause a temperature shift of quartz. Cheap modules do not use TCXO, but cheap crystal ±25 ppm, so for 433MHz offset is about ±10kHz. In winter it's even worse, when the first module at home +25C, the second - in a street -35C.

jgromes, 2018-07-26:
That's true - maybe I can add some frequency error calibration in the future? The SX1278 can save frequency error of the last received packet.

jgromes, 2018-07-26:
Issue resolved in release 4.1.0.

closed 2018-07-26 by jgromes