jgromes/LoRaLib #26: Frequency error calibration

jgromes, 2018-07-26:
SX127x modules have the ability to measure frequency error of received LoRa signals. This can be used to add automated calibration procedure, which would allow to use very narrow bandwidths even with cheap LoRa modules without temperature-calibrated oscillators.

musgravehill, 2018-07-26:
If packet not received (time out), than Lora can change frequency to catch packet. For example, BASIC_FREQ++ 8kHz in a cycle, then BASIC_FREQ-- 8kHz in a cycle until catch.  BASIC_FREQ = 433.000, the packet lost, time out. BASIC_FREQ set to 433.008, then 433.016, etc. Then 433.000, 432.092, 432.086.

jgromes, 2018-07-28:
A new method `getFrequencyError()` was added in b2c4f1d61da53c21d26c65048278bcd95131efaa

This method returns frequency error of the last received packet in Hz, so that can be used as a base for the calibration procedure.

It might be better not to have the calibration as a part of the API but rather have two examples `FreqCalibrationRx.ino` and `FreqCalibrationTx.ino` that implement the procedure using existing API methods. Calibration requires synchronization between transmitter and receiver, and it depends on the user's specific requirements (required minimum bandwidth, expected temperature range etc.).

jgromes, 2018-08-24:
I did some more testing and it seems like it's not necessary to keep adjusting center frequency while lowering bandwidth. I managed to get 7.80 kHz bandwith transmissions just fine after a single frequency correction on 500 kHz bandwidth. It might be necessary to adjust the frequency again after some time to compensate for time and tempertature drift, but that's so use-case dependent, it doesn't really make sense to create examples for it.

I added a new parameter to the `getFrequencyError` method, when called as `getFrequencyError(true)` it will trigger automatic LoRa modem data rate correction. User still has to change the cetner frequency manually though.

closed 2018-08-24 by jgromes