jgromes/LoRaLib #27: Found module with error

renatocarusos, 2018-07-31:
**Hi!**
I've been away from lora prototyping...

**The bug**
When I try to start the object, I quickly get a _"failure, code -4"_ in the Transmit, Setting and Recive sketch. I observed _(#KITELIB_DEBUG)_ that he _"Found SX127x!"_ if I initiate between SX1276 ~ 9.
With the others I have, after ten attempts _"failure, code -2"_.

**To Reproduce**
[The code](https://pastebin.com/raw/kqtLTaCA)
[LoRa Module](https://ae01.alicdn.com/kf/HTB1JUYZQpXXXXX1XFXXq6xXFXXXA/-100pcs-lot-LoRa1276-NiceRF-LoRa-Module-in-868MHz-SPI-interface-with-SX1276-Chip-4000M-Long.jpg)
[ESP32 Module](https://makeradvisor.com/wp-content/uploads/2018/04/ESP32-DOIT-DEVKIT-V1-Board-Pinout-36-GPIOs-Copy.jpg)

D18 as  CLK
D19 as MISO
D23 as MOSI
D32 as CS
D34 as D0
D35 as D1

**Expected**
I was hoping it would work as it normally does with Esp8266 + v1.1.0.

 - MCU: ESP32 DEVKITV1 
 - LoRa module type SX1276
 - Arduino IDE version 1.8.5
 - Library version 5.0.0

**Could you help me investigate?**

jgromes, 2018-07-31:
So the `begin()` method returns -4 for you?

renatocarusos, 2018-07-31:
Yes. But it finds the module if I declare it as 76~79. And after returning the error, dies in "while(true)".

jgromes, 2018-07-31:
That extremely weird: -4 is `ERR_PACKET_TOO_LONG`. I checked and the only place in the entire library that can return this is the `transmit` mtehod.

You mentioned that you tried uncommenting the `#define KITELIB_DEBUG` macro. Can you post the whole Serial output when running with the macro uncommented?

jgromes, 2018-07-31:
Basically, three things can go wrong with the `begin` method:
1. It won't find the chip (version number in register doesn't match the expected) - error code -2
2. SPI write fails - error code -16
3. Some of the arguments provided by user is incorrect

renatocarusos, 2018-07-31:
The debug output is:
>"Found SX127x!
>failed, code -4
Weird...

jgromes, 2018-07-31:
So it's not a failed SPI write - that would have printed bunch more information about the register that failed.

I'm going to try with my ESP32, see what I can figure out.

jgromes, 2018-07-31:
I found out what is the issue, there are two parts to it:
1. There's a mistake in the way module classes inherit common methods, so when the `begin` method is executed for class SX1276, it calls configuration methods from class SX1278 (instead of SX1276).
2. The reason the returned code was -4 is that the returned codes get ORed together at one point. Since more than one configuration method failed, this is what happened: -12 | -8 = -4

jgromes, 2018-07-31:
There's also another problem: on ESP32, there seems to be some inconsistence in float constants. For example, the 7.8 passed to `begin` method becomes 7.8000001907, while 7.8 inside `setBandwidth` method is interpreted as 7.8000000000. No idea why, but apparently I'm going to compare the float values properly, i.e. not exact match, but within small range.

jgromes, 2018-07-31:
All three bugs should be fixed in release 5.0.1. Thanks for reporting!

closed 2018-07-31 by jgromes

renatocarusos, 2018-07-31:
Good! This kind of inconsistency in the variables is pretty crazy. I'm having serious issues with int at attiny85. The library has progressed a lot, good work!