jgromes/LoRaLib #30: ESP32 "CORRUPT HEAP"

renatocarusos, 2018-07-31:
**Hi**

**The bug**
In receiver sketch, when the transmitter starts to transmit, the receiver, which was running normally with 

> "Waiting for incoming transmission ... timeout!"

 gives crash:

> CORRUPT HEAP: Bad tail at 0x3ffb2420. Expected 0xbaad5678 got 0xbaad5600
> assertion "head! = NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 205, function: multi_heap_free
> abort () was called at PC 0x400d57f3 on core 1
> 
> Backtrace: 0x400883f8: 0x3ffb1e10 0x400885fb: 0x3ffb1e30 0x400d57f3: 0x3ffb1e50 0x400880e1: 0x3ffb1e80 0x400840d2: 0x3ffb1ea0 0x400845e1: 0x3ffb1ec0 0x4000bec7: 0x3ffb1ee0 0x400e73b1: 0x3ffb1f00 0x400e7739: 0x3ffb1f20 0x400d2097: 0x3ffb1f40 0x400d0b32: 0x3ffb1f70 0x400e79ee: 0x3ffb1fa0
> 
> Rebooting ...

**To Reproduce**
[The code](https://pastebin.com/raw/z6cHe8G3)
[LoRa Module](https://ae01.alicdn.com/kf/HTB1JUYZQpXXXXX1XFXXq6xXFXXXA/-100pcs-lot-LoRa1276-NiceRF-LoRa-Module-in-868MHz-SPI-interface-with-SX1276-Chip-4000M-Long.jpg)
[ESP32 DEVKITV1](https://makeradvisor.com/wp-content/uploads/2018/04/ESP32-DOIT-DEVKIT-V1-Board-Pinout-36-GPIOs-Copy.jpg)

D18 as CLK
D19 as MISO
D23 as MOSI
D32 as CS
D34 as D0
D35 as D1

**I Think ...**
this is kind a [overwriting a memory buffer](https://github.com/espressif/arduino-esp32/issues/1101#issuecomment-365730136).
I used the ESP Decoder and got line 145 of SX127x.cpp, as in the screenshot:
[image]

This is a problem that will probably be solved by the ESP32 community, but do you imagine some way to circumvent this exception?

 - MCU: ESP32 DEVKITV1
 - LoRa module type SX1276
 - Arduino IDE version 1.8.5
 - Library version 5.0.1

jgromes, 2018-08-01:
You mean this line [145](https://github.com/jgromes/LoRaLib/blob/a9cf92996ecd67e91b15ba18650030bd9f17ea71/src/modules/SX127x.cpp#L145)?

That will cause memory leak - dynamically allocated array `data` will no longer be de-allocated, so eventually, you will run out of RAM.

jgromes, 2018-08-01:
As a temporary workaround, you can use the byte array version of `receive` - that is working fine on ESP32, because it doesn't do any dynamic allocations. Downside is that you need to know the packet length in advance.

jgromes, 2018-08-01:
I think I know what's causing this bug: in the `receive(String&)` method, I'm using a zero-length character array as a temporary object. This works fine on Arduino, but for some reason not on ESP32. I'll rework the method and release the fix in v5.0.2.

jgromes, 2018-08-01:
Turns out it was much simpler than that; there was a [dynamic allocation](https://github.com/jgromes/LoRaLib/blob/dac6423b29cd7a02c46878987d7357ab1521d1c9/src/modules/SX127x.cpp#L194) that was one byte smaller than it should be. Subsequently, a null character was written to the unallocated space, hence the crash on ESP.

Fixed in update 5.0.2, thanks for reporting the bug!

closed 2018-08-01 by jgromes