jgromes/LoRaLib #32: Duplex

juliantech5, 2018-08-04:
Hey! would it be possible to see a simple code with both receive and transmit even if it is with a delay, ping pong between lora modules.

Thanks

jgromes, 2018-08-04:
I don't think it's necessary to add that example - the purpose of examples is to show off library features, as simply as possible. What you're suggesting is not a _feature_, but a _usage_ of the library.

It is also unclear what is meant by duplex here. Since SX127x series doesn't support true duplex, I can see at least two different possibilities how to emulate it:
1. **Time-division duplex** -  You could have a single LoRa chip that listens for incoming transmissions 50% of the time and transmits 50% of time.
2. **Frequency-division duplex** - Since LoRaLib can control multiple SX127x on a single Arduino, you could have two chips in different channels - one always transmitting, one always receiving.

Which one user wants to implement is entirely up to him, because it's use-case dependent.

juliantech5, 2018-08-04:
Ah ok, Was not aware of the fact it did not support duplex, thanks

closed 2018-08-04 by juliantech5