jgromes/LoRaLib #34: FSK recieve

juliantech5, 2018-08-13:
Would it be possible to receive FSK packets with this library?

jgromes, 2018-08-13:
FSK modem support is not currently implemented. LoRaLib only supports the LoRa modem (as the name suggests).

Is there a reason why you want to use FSK specifically on SX127x chips? You could use RFM69 chips with [KiteLib](https://github.com/jgromes/KiteLib) for FSK, and you could use the same public interface as LoRaLib.

juliantech5, 2018-08-13:
We are currently developing a satellite with a LoRa downlink using the sx1278, we require an FSK uplink due to regulations though.

jgromes, 2018-08-13:
And I suppose you don't want to have SX1278 only for downlink and RFM69 only for uplink?

juliantech5, 2018-08-13:
Yeah exactly, we can implement it but we already have a near complete prototype

jgromes, 2018-08-13:
I will investigate a bit. Can't promise anything though.

juliantech5, 2018-08-13:
Do you have an email address or other means of contact?

jgromes, 2018-08-13:
Sorry, I'd rather keep all GitHub-related communication here.

juliantech5, 2018-08-13:
Sure

Bambofy, 2018-08-13:
Hi, allow the modem mode to be switched between FSK and LoRa at runtime please!

jgromes, 2018-08-13:
I'm counting on that, this was obvious from the single-chip requirement. It will require substantial changes to the public API though.

jgromes, 2018-08-13:
I made a new branch [fsk-modem](https://github.com/jgromes/LoRaLib/tree/fsk-modem), all the development will be there until everything is tested properly.

There will be a lot of changes, both in public API and in the backend, so it will take me a few weeks to implement the support.

juliantech5, 2018-08-13:
Thanks jgromes, really appreciated :) Please keep us updated!

jgromes, 2018-08-14:
First draft of the API:

Call `beginFSK();` to start the FSK modem and disable LoRa modem. Similarly, calling `begin()` will start LoRa modem and disable FSK. All calls to methods like `transmit()`, `receive()` etc. will be routed through the currently active modem.

What do you guys think about this?

Bambofy, 2018-08-14:
Yep sounds perfect.

jgromes, 2018-08-15:
Basic FSK transmission is in 736f709a505c6cb1960f4be0377446fc43dffbff, currently only for SX1278. On SDR receiver it seems to be transmitting, although that will have to be tested with receiving SX1278.

What do you need in terms of the features? Do you need interrupt-driven Tx and Rx as well as the default blocking ones?

What about configuration? Currently you can set bit rate, receiver bandwidth, frequency deviation and all the settings that aren't exclusive to LoRa (gain, overcurrent protection etc.). But there are other FSK settings, e.g. modulation shaping. Currently it's left to the default Gaussian filter with BT=0.3. Do you need access to that in the public API as well?

juliantech5, 2018-08-15:
Thanks for your quick work jgromes! I'll test it out now on the satellite and let you know, In theory, we should be fine with the current settings but it would be great to have the option to change modulation shaping and other settings in the Public API.

I don't think FSK uses address filtering like LoRa does, we don't personally need it but it may come in useful. 

Thanks again!

Bambofy, 2018-08-15:
Hi James, ISR functions aren't needed (bit skeptical, simpler the better) also the configuration is great we don't need to change anything else. Is there a chance you can add timeouts to the receive and transmit functions? and how does the receive and transmit function handle timing out at the moment?

Cheers

jgromes, 2018-08-15:
Right now timeouts are set to 150% of expected transmission time based on the bit rate and payload size. Receive function isn't implemented yet, I'm working on that right now.

Bambofy, 2018-08-15:
Thanks, awesome library, thanks for making it. will credit you in our project FOSSASAT-1 (http://fossa.systems)

jgromes, 2018-08-15:
Basic FSK reception is implemented in f903e2908f26497f7eab1bc3fceca5ae2e05955f, so all the basic FSK methods are done for SX1278. Next up is finishing configuration, examples and code cleanup.

Side note: The crystals on the cheap SX1278 modules are pretty bad, even at room temperature the carrier frequency is 6-8 kHz off. LoRa can receive packets that are off by less than 25% of bandwidth, and that gets pretty narrow for the lower bandwidth values (which you need in order to maximize distance). How do you plan on adddressing that?

juliantech5, 2018-08-15:
We are trying to use the highest bandwidth possible as allowed by the IARU
and the ITU, due to these duty cycle limitations we have integrated a
tuning packet system that sends high bandwidth packets every set amount of
time. We will need to account for this on the ground station and are also
looking at sourcing or developing our own SX1278 modules with low tolerance
crystals. Stuart integrated AFC tuning packets on his Lora tracker library
at the beginning of each sketch, would that be something that you could
integrate?

Thanks

jgromes, 2018-08-15:
So you want to add some form of automatic frequency calibration?

juliantech5, 2018-08-15:
Yes Indeed, right now we sort of have a working version that changes the ground station centre freq when it hears the satellite using the freq offset.

juliantech5, 2018-08-15:
We are doing testing in low temperatures though and it seems like it will work regardless, the drift is not immense. A TCXO would be ideal though just to be safe.

jgromes, 2018-08-15:
See #26 

Basically, I don't want to add it into the API, because LoRaLib is intended only as a control interface for SX127x series. Anything above what the chip itself can do is out of scope in my opinion.

juliantech5, 2018-08-15:
Yeah of course, No problem. I'll test out the FSK now, are the settings supposed to be working btw?

jgromes, 2018-08-15:
Basic settings like carrier frequency, bit rate, receiver bandwidth, frequency deviation and sync word should be working. Output power and over current protection should work too (because they are not dependent on modulation type).

juliantech5, 2018-08-15:
Great, Seems to be working perfectly. Thank you for the help! FSK functionality is not needed when you have LoRa but due to current ITU regulations, we need to have a backup modulation in case LoRa does not work for emergency procedures during orbit.

jgromes, 2018-08-16:
The most important functions of FSK modem are now implemented. The current support for FSK modem on SX127x chips in LoRaLib is now the same as on RF69 chips in KiteLib.

I will be doing a bit of testing, and in case no major bugs are found within a few days, I will merge fsk-modem branch into master. If you find any bugs on your end, please report them in a separate issue. Thanks!

Bambofy, 2018-08-16:
Thank you for implement it :) do you have any idea on the power consumption of switching modem modes?

jgromes, 2018-08-16:
The modem change itself is just an SPI write to a single register while the module is in sleep mode. Then, configuration is performed, which is done in standby mode and takes just a few milliseconds. I don't think power consumption will be particularly increased when switching from LoRa mode to FSK or vice-versa. Haven't measured it though.

Bambofy, 2018-08-16:
alright so i assume we would be alright to switch between FSK and LoRa every loop? thanks, can i contact you for advise through email?

jgromes, 2018-08-16:
It should be fine. You can also put the module into sleep mode by calling `sleep()` to minimize current draw down to just a few uA. The module will wake up automatically when you try to transmit or attempt to receive.

Sorry, I'd rather keep all GitHub-related communication here. If you have any questions in the future, feel free to open a new issue.

jgromes, 2018-08-17:
@Bambofy 
> Additionally, on the receiving arduino program, will it be able to listen to both FSK and LoRa transmission at the same time, or will we have to switch that receiver modem too? if we have to switch the receiver modem that will make the timing of everything very difficult?

There's no way to achieve that on SX127x, because only one of the modems can be active at any given given time.

juliantech5, 2018-08-17:
Slightly off-topic to FSK,
Do you know how we could use transmit FM tones using the sx1278? (say morse for example)

Thanks

jgromes, 2018-08-17:
I don't think you can. SX1278 has something called "continuous mode", in which it transmits anything that is read on DIO1 (clock) and DIO2 (data) pins. It _might_ be possible to use this to generate FM tones, but I really don't know enough about FM and FSK modulations to tell you for sure. Continuous mode isn't implemented and I'm not planning on adding it.

Besides, the module was designed for digital data transmission - not analog FM tones.

juliantech5, 2018-08-17:
Ah ok No problem was just wondering. 
And with the current state of the FSK modem would it be possible to transmit FSK RTTY with some modifications?

jgromes, 2018-08-17:
I'm not sure about that: how fast can RTTY signals be? The lowest possible bit rate of the module is 1.2 kbps, I think that is way too fast.

juliantech5, 2018-08-17:
Stuart has a library that transmits RTTY FSK using Lo-Ra, https://github.com/LoRaTracker/LoRaTracker-Library
Would this be hard to integrate into this library?

jgromes, 2018-08-17:
Seems to be using the aforementioned continuous mode.

No idea, since I don't have any experience with RTTY. I would suggest contacting the author of that library.

Bambofy, 2018-08-18:
Hi, doesn't the SX1278 transmit analog FSK modulated signals?

If yes then shouldn't a RTTY terminal process the FSK transmissions anyway?

jgromes, 2018-08-18:
I don't know how exactly are the data encoded in the FSK output of SX1278, but it's not just plain binary FSK (high frequency = 1; low frequency = 0). If that was the case then yes, you could receive it with RTTY terminal (provided you manage to get the timings right).

jgromes, 2018-08-18:
Since all the bugs I found during testing were resolved, I merged the fsk-modem branch into master.
LoRaLib now has FSK support in release 5.1.0

Feel free to open new issues for bugs/change requests.

closed 2018-08-18 by jgromes

jgromes, 2018-08-18:
@juliantech5 @Bambofy since you guys asked for it ... https://www.youtube.com/watch?v=a0CyXdC6qJg :smile:

juliantech5, 2018-08-18:
Please give this man an award, your name will be in space my friend.

retzbeats, 2019-10-30:
Hi jgromes! we are actually using the sx1278 for 433MHz FSK with RadioLib, but we need the signal strength information, we have already check the register map for FSK in the sx1278 and RSSI value is available. Do you have a function to read this in the receiver? or can you make that available?

jgromes, 2019-10-30:
@retzbeats you're replying to an issue that was resolved well over a year ago. Please open a new issue either in RadioLib or here.

retzbeats, 2019-10-31:
> @retzbeats you're replying to an issue that was resolved well over a year ago. Please open a new issue either in RadioLib or here.

Sorry, you are right. I have opened the issue in RadioLib thanks