jgromes/LoRaLib #36: FSK direct mode

jgromes, 2018-08-19:
FSK modem supports direct operation (aka continuous mode), in which all data received on DIO1 (data) and DIO2 (clock) pins are directly transmitted.

This allows to transmit FM notes (by feeding Arduino `tone()` function output directly to DIO2 (DIO1 left floating). Proof of concept: https://www.youtube.com/watch?v=a0CyXdC6qJg

Direct/continuous mode could also be used for FSK RTTY (or AFSK RTTY, using the `tone()` function).

jgromes, 2018-08-19:
Basic direct mode support is now in `direct-mode` branch.

Bambofy, 2018-08-19:
any ETA on when this will be merged into the master branch? thanks for this, really awesome work.

jgromes, 2018-08-19:
I'm still struggling with RTTY a bit, hopefully within the next few days.

jgromes, 2018-08-20:
Finally managed to get RTTY [working](https://www.youtube.com/watch?v=I5VkxEPc2zw) on `direct-mode`, so I will merge that branch and release a new version.

Since RTTY is a protocol, it won't be directly supported by LoRaLib (as in: there will not be a public API implementing the protocol, e.g. `lora.sendRTTY()`). User can either write his own implementation using LoRaLib as a driver or use KiteLib. That already separates protocols from drivers, so I will write full RTTY implementation with public API there.

jgromes, 2018-08-20:
Implemented in release 5.2.0

closed 2018-08-20 by jgromes

juliantech5, 2018-08-20:
Hey jgromes do you have the code that you used with rtty on this library?
thanks!