jgromes/LoRaLib #38: FSK Direct Mode RTTY Compatable?

Bambofy, 2018-08-20:
Hi, when using the FSK direct mode, how do we communicate with a RTTY terminal?

jgromes, 2018-08-20:
I'm currently working on integrating RTTY into KiteLib, so you will have to use that if you want simple API. I can give you an Arduino sketch I used as a proof of concept.

Bambofy, 2018-08-20:
Yes please, any information is welcome. i have 0 ideas on how to implement the RTTY protocol.

jgromes, 2018-08-20:
```c++
#include <LoRaLib.h>

// baud rate    delay [ms]
// 45           22222
// 45.45        22000
// 50           20000
#define BAUD_DELAY    22222

SX1278 lora = new LoRa(10, 2, 5);

uint32_t base = 0x6C9999;
uint32_t shift = 3; // 3 * 61 Hz = 183 Hz
void setup() {
  Serial.begin(9600);
  lora.beginFSK(434.4);
  lora.setFrequencyDeviation(0);
}

void loop() {
  lora.directMode();

  // lead in 500 ms
  mark();
  delay(500);
  
  sendStr("Hello World!");
  
  lora.standby();
  delay(1000);
}

void sendStr(const char str[]) {
  for(int i = 0; i < strlen(str); i++) {
    sendByte(str[i]);
  }
  sendByte('\r');
  sendByte('\n');
}

void sendByte(byte b) {
  space();
  
  for(byte m = 0x01; m; m <<= 1) {
    if(b & m) {
      mark();
    } else {
      space();
    }
  }

  mark();
}

void mark() {
  unsigned long start = micros();
  lora.directMode(base + shift);
  while(micros() - start < BAUD_DELAY);
}

void space() {
  unsigned long start = micros();
  lora.directMode(base);
  while(micros() - start < BAUD_DELAY);
}

```

Terminal settings: 434.4 MHz center, 182 Hz shift, 45 baud, 8 data bits, 1 stop bit.

But if you wait a few more hours, you will get API that will be about as complicated as Arduino `Serial` class ;)

`begin(shift, baudrate, databits, stopbits)` to setup RTTY,
`print(string)` to send string etc.

Bambofy, 2018-08-20:
nice code, that's a super nice implementation, please can you add RTTY to LoRaLib?

juliantech5, 2018-08-20:
A temporary branch

jgromes, 2018-08-20:
As per my comment in #36 

> Since RTTY is a protocol, it won't be directly supported by LoRaLib (as in: there will not be a public API implementing the protocol, e.g. lora.sendRTTY()). User can either write his own implementation using LoRaLib as a driver or use KiteLib. That already separates protocols from drivers, so I will write full RTTY implementation with public API there.

LoRaLib does not implement protocols, that's in [KiteLib](https://github.com/jgromes/KiteLib).

Bambofy, 2018-08-20:
ok thank you for implementing the direct mode, that's really opened our options up a lot.

jgromes, 2018-08-20:
KiteLib now has basic RTTY support - see [RTTY example](https://github.com/jgromes/KiteLib/blob/master/examples/SX127x_RTTY_Transmit/SX127x_RTTY_Transmit.ino) for details.
The API works exactly like Arduino Serial class.
Right now it will only `print` and `println` strings, the overloads will be implemented tomorrow.

closed 2018-08-20 by jgromes