jgromes/LoRaLib #40: Example sketch for switching between modem modes?

Bambofy, 2018-08-20:
Hi I'm trying to switch an sx1278 between lora to fsk to fsk direct mode in a loop, it doesn't seem to work properly, can you make a quick sketch to show me how it's done? Thanks

jgromes, 2018-08-21:
> it doesn't seem to work properly
What exactly does that mean? All the methods return status codes, it might be helpful to know them. And it would be really helpful to know your code.

Bambofy, 2018-08-21:
All the status codes return success but the receiver doesn't receive anything.

This is where all transmission call through:
https://github.com/Bambofy/FOSSASAT-1/blob/master/Code/Beacon/communication.cpp#L334

jgromes, 2018-08-21:
This is the code I used for testing

```c++
#include <LoRaLib.h>

#define DIO2  6

SX1278 lora = new LoRa(10, 2, 5);

void setup() {
  Serial.begin(9600);
  pinMode(DIO2, OUTPUT);
}

void loop() {
  Serial.println("LoRa");
  lora.begin();
  delay(1000);
  for(int i = 0; i < 5; i++) {
    Serial.println("Sending LoRa");
    lora.transmit("Hello World!");
    delay(1000);
  }
  
  Serial.print("FSK\t");
  Serial.println(lora.beginFSK());
  delay(1000);
  for(int i = 0; i < 5; i++) {
    Serial.println("Sending FSK");
    lora.transmit("Hello World!");
    delay(1000);
  }

  Serial.print("AFSK\t");
  Serial.println(lora.directMode());
  delay(1000);
  for(int i = 0; i < 5; i++) {
    Serial.println("AFSK tone 400 Hz");
    tone(DIO2, 400);
    delay(1000);
    Serial.println("AFSK tone 1000 Hz");
    tone(DIO2, 1000);
    delay(1000);
  }
  noTone();
  lora.packetMode();
}
```

It sends 5 LoRa packets, 5 FSK packets, transmits 10 FM tones and then sends RTTY string. On SDR, all transmissions are sent, although you are right that I can't receive FSK packets after the switch from LoRa modem to FSK.

jgromes, 2018-08-21:
You're also evaluating return codes wrong: e.g. [here](https://github.com/Bambofy/FOSSASAT-1/blob/master/Code/Beacon/communication.cpp#L37) you rewrite the value of `err_check` over and over again, so the ony thing you know is `setSyncWord` return code, all the previsou are lost.

Why not just pass all the configuration to `beginFSK()`?

Bambofy, 2018-08-21:
Yep will fix the err_check lazyness, i can't tell which part of the example code above is RTTY transmitted? I saw you used lora.directMode( INT VALUE FOR FREQ ) in a previous code snippet is that still usable? or do I have to use tone() to send data through directMode?

jgromes, 2018-08-21:
Sorry, I posted wrong code, here's the correct version:

```c++
#include <KiteLib.h>

#define DIO2  6

SX1278 lora = new Module(10, 2, 5);
RTTYClient rtty(&lora);

void setup() {
  Serial.begin(9600);
  pinMode(DIO2, OUTPUT);
}

void loop() {
  Serial.println("LoRa");
  lora.begin();
  delay(1000);
  for(int i = 0; i < 5; i++) {
    Serial.println("Sending LoRa");
    lora.transmit("Hello World!");
    delay(1000);
  }
  
  Serial.print("FSK\t");
  Serial.println(lora.beginFSK());
  delay(1000);
  for(int i = 0; i < 5; i++) {
    Serial.println("Sending FSK");
    lora.transmit("Hello World!");
    delay(1000);
  }

  Serial.print("AFSK\t");
  Serial.println(lora.directMode());
  delay(1000);
  for(int i = 0; i < 5; i++) {
    Serial.println("AFSK tone 400 Hz");
    tone(DIO2, 400);
    delay(1000);
    Serial.println("AFSK tone 1000 Hz");
    tone(DIO2, 1000);
    delay(1000);
  }
  noTone(DIO2);
  lora.packetMode();

  Serial.print("RTTY\t");
  Serial.println(rtty.begin(434.0, 183, 45));
  delay(1000);
  rtty.leadIn(500);
  rtty.println("Hello World!");
  lora.standby();
  delay(1000);
}
```

Bambofy, 2018-08-21:
thanks i'll let you know how i get on with this.

jgromes, 2018-08-21:
I'm pretty confused by this bug. It seems that after switching when switching to FSK from LoRa modem, something in the chip changes and the FSK transmissions can no longer be received by a module with the same settings. Here's what I know so far:
1. I did a register dump before and after switching modems (FSK -> LoRa -> FSK). It looks like all the configuration registers are the same, with the notable exception of register 0x30 (RegPacketConfig1). Before the change, this register reads 0x90 (expected value) but afterwards, it reads 0x00. The weirdest part is that it seems stuck there - writing to the register is ignored, regardless of chip mode (standby/sleep).
2. When comparing the FSK spectrum before and after the modem change, it seems changed - there's one extra spike on the lower end of the band. I think this is the reason why the receiver doesn't pick up the packet, but I'm not sure.
[image] [image]
3. I tried resetting the chip both by pulling the RESET pin low (as per datasheet), as well as power cycling and neither seems to resolve the issue.

The only attempt at solution that I can think of now is trying to configure every single SPI register each time user calls either one of the `begin()` methods. Right now, not all registers are configured, only the necessary ones (in previous versions of LoRaLib this wasn't an issue because it was assumed that user would only call `begin()` method once after startup, so I could rely on the default values). Downside is that I don't know if this actually resolves anything and even if it does, it will prolong execution time of the `begin()` methods.

I will update you once I know more.

jgromes, 2018-08-21:
First bit of progress: I managed to trace the issue to `setBandwidth()` method called inside `begin()` when switching to LoRa mode. When `setBandwidth()` isn't called, receiver can pick up FSK packets just fine. No idea why.

Bambofy, 2018-08-21:
Does direct mod fsk take the setBandwidth() function? Might be because the bandwidth changing is not supported by the cake direct mode?

jgromes, 2018-08-21:
See the [wiki](https://github.com/jgromes/LoRaLib/wiki/FSK-Modem)

FSK modem does not support `setBandwidth()` that method is exclusive to LoRa, This is not a direct mode issue, it happens when configuration LoRa modem.

jgromes, 2018-08-21:
Weirdness intensifies: inside `setBandwidth()`, method `setBandwidthRaw()` is [called](https://github.com/jgromes/LoRaLib/blob/master/src/modules/SX1278.cpp#L185). When I comment out everything from inside `setBandwidthRaw()`, it doesn't work. When I comment out _call_ to `setBandwidthRaw()`, the issue magically disappears. The crazy part is this: when I take the code that was inside `setBandwidthRaw()` and just paste it into `setBandwidth()` instead of calling `setBandwidthRaw()`, it works.

I am seriously doubting my sanity right now. Is there something really obvious I'm missing? I will try debugging in Atmel Studio, maybe that will help.

Bambofy, 2018-08-21:
Is it because you have defined setBandwidthRaw() as a protected member, but when calling it you are using the format SX1278::setBandWidthRaw() when it should be something like this::setBandWidthRaw() ?

jgromes, 2018-08-21:
I can just hotfix it for now by pasting the code from `setBandwidthRaw()` into `setBandwidth()` but that will only work for SX1278 and not the other modules. I don't like the idea of not fixing the root cause; it will probably rear its ugly head again somewhere further down the line ...

Bambofy, 2018-08-21:
yep for sure, what happens when you change setBandwidthRaw() function to public?

jgromes, 2018-08-21:
It doesn't resolve the issue and as a bonus, end user gains the ability to set invalid bandwidth values ;)

I'll commit the hotfix but won't create a new release yet. I'd rather only release once it's been fully resolved.

Bambofy, 2018-08-21:
very strange! ok sounds good.

jgromes, 2018-08-21:
Hotfixed in 5de5e1ba15806e95c45698cf0e3d360838c76c7d

You know what would be stranger? If it had absolutely nothing to do with any sort of SPI write/module configuration.

I just commented out [this line](https://github.com/jgromes/LoRaLib/blob/5de5e1ba15806e95c45698cf0e3d360838c76c7d/src/modules/SX1278.cpp#L189) and it works ... still no idea why.

jgromes, 2018-08-21:
Another tiny bit of a progress: seems like it is caused by the fact that bandwidth is saved as a `float` variable. When it is replaced by some other type (e.g. `uint32_t`) it works. I'm probably going to replace all floating point member variables with integers, just to be safe.

Bambofy, 2018-08-21:
wtf haha, nice debugging! any idea why the type would matter? perhaps it's because float is a definition and uint32_t is a real type? anyway you should submit that bug report to Semtech or someone, that's a pretty big bug to be honest.

jgromes, 2018-08-21:
It's not a bug in Semtech's chip, it's in my code. I still don't know why exactly did this happen, and until I do, there's really not much to report here. Besides, I don't think the chips were designed with quick modem switching in mind.

Bambofy, 2018-08-21:
true, thanks for the bug fix, great to see your debugging method. 👍

jgromes, 2018-08-22:
I AM AN IDIOT.

Of course the FSK spectrum looks different after I've applied Errata fixes that were intended for LoRa modem only ... Type seemed to make a difference because after I've changed it, the errata fix no longer executed, hence no more changed spectrum.

jgromes, 2018-08-22:
Fixed in v5.2.2, thanks for reporting it.

closed 2018-08-22 by jgromes