jgromes/LoRaLib #42: Code Error FFE8

juliantech5, 2018-08-22:
We are currently experimenting with RTTY and FSK,
the following code appears 
`(SWITCH FSK MODE)
Unable to set configuration, code ffe8
(COMM SYSTEM END
(SWITCH LORA MODE)
`
LoRa works fine but FSK does not work, DIO2 Is hooked up to D10 (and declared to d10)

Any ideas? Will try another Transmitter now.

jgromes, 2018-08-22:
The codes are returned as 16 bit signed numbers, could you please print it as such?

juliantech5, 2018-08-22:
Code -24

jgromes, 2018-08-22:
https://github.com/jgromes/LoRaLib/wiki/Error-Codes

-24 is invalid sync word. FSK sync word must not be longer than 8 bytes and must not contain any null bytes (0x00)

juliantech5, 2018-08-22:
Ah! Same Issue as before.
Sorry mate!
Thanks

juliantech5, 2018-08-22:
Is it fixed in the example?

jgromes, 2018-08-22:
The range check was wrong, it should be working this time.

Fixed in bdc78fe637e6dd5f00688134ad11dc7dd3b66d81

closed 2018-08-22 by jgromes