jgromes/LoRaLib #43: A few typos in the text

euwind, 2018-08-23:
Hi,

I work with an IDE that generates a few warnings when compiling projects using the library (all for v.5.2.2):
1. SX127x.h. line 321. "SX127X_PREAMBLE_DETECTOR_2_BYTE" redefined. I believe that name of macro should be SX127X_PREAMBLE_DETECTOR_3_BYTE.
3. SX127x.cpp. line 266. "Unused variable 'start'". This variable is really not used anywhere and left from the earlier versions.
2. SX127x.cpp. line 708. "Type left shift count >= width of type". Type coercion
 uint32_t raw = **(uint32_t)**_mod->SPI....
will remove warning.

In your last update, the LoRa mode check is performed before the sensitivity optimization in the files for the SX1277 and SX1278 chips. For SX1276, this check is performed after the specified optimization. Is that right?

Next, regarding  Semtech Errata Note and fuctions setFrequency.

Section 2.1. Sensitivity optimization for BW=500 kHz is performed by setting the values of registers 0x36 and 0x3a.
For the frequency range from 862 to 1020 MHz: 0x02 and 0x64 respectively. This is done correctly in your code.

The frequency range from 410 MHz to 525 MHz must be set to 0x02 and 0x7F respectively, but you set the default values to 0x03 and 0x65 in files for all types of chips. I think it should be correcting.

 - Library version 5.2.2

jgromes, 2018-08-23:
All the issues should be fixed in 221d5e0136fccc93334b077e1d8bfeee610311a9

Could you please test with your IDE and see if the warnings are resolved?

euwind, 2018-08-23:
Now everithing OK. Good luck Jan!