jgromes/LoRaLib #44: ESP32 Flash Corrupt ?

BBBits, 2018-08-24:
Hi,
I've been working with a modified version of Sandeep Mistry's Lora Lib in Arduino.
I have a Number of these modules in various flavors of my project.
https://www.aliexpress.com/item/TTGO-LORA-SX1278-ESP32-0-96OLED-16-Mt-bytes-128-Mt-bit-433Mhz-for-arduino/32824209385.html
My application now needs a faster response time so decided to try your fsk lib .
I compiled this sketch and blindly uploaded it to three modules.
`#include <LoRaLib.h>
SX1278 fsk = new LoRa;
int state ;
uint32_t cnt = 0;
char dat[100];
size_t l;

void setup() {
  Serial.begin(115200);
  Serial.print(F("Initializing ... "));
  state = fsk.beginFSK();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
  state = fsk.setFrequency(433.5);
  state = fsk.setBitRate(100.0);
  state = fsk.setFrequencyDeviation(10.0);
  state = fsk.setRxBandwidth(250.0);
  state = fsk.setOutputPower(10.0);
  state = fsk.setCurrentLimit(100);
  uint8_t syncWord[] = {0x01, 0x23, 0x45, 0x67, 
                        0x89, 0xAB, 0xCD, 0xEF};
  state = fsk.setSyncWord(syncWord, 8);
  if (state != ERR_NONE) {
    Serial.print(F("Unable to set configuration, code "));
    Serial.println(state);
    while (true);
  }
}

void sendmsg(){
	l= sprintf(dat,"Msg %d",cnt);
	cnt++;
	printf("Sent %s \n",dat);
	fsk.transmit((uint8_t *)dat,l,0);
}
void loop(){
  delay(1000);
  sendmsg();
  do_rcv();
}

void do_rcv(){
  String str;
  state = fsk.receive(str);
  if (state == ERR_NONE) {
    Serial.println(F("Received packet!"));
    Serial.print(F("Data:\t"));
    Serial.println(str);
  }
}
`
========================
This is what I get from all the devices when connecting to the serial console
rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
=====================
When I try to upload a new firmware to the devices , it fails with this error.
esptool.py v2.1
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...
A fatal error occurred: Timed out waiting for packet content
============================================
I've been programming these modules the same way for over six months .
Totally confused as to what was happening , I decided to put a new flash chip on one of the modules .
I could then upload a new program to that module .
On the other Lora Lib , I Init like this .
SPI.begin(5, 19, 27, 18);
LoRa.setPins(18, 14, 26); //LoRa.setPins(SS, RST, DI0);
I didn't set any pins in your Lib & suspect that is the reason I am in this situation .
Any thoughts on how I can resolve this problem without swapping out all the Flash chips ?

Regards
Barry

jgromes, 2018-08-24:
Three code markers in a row will show code block

\`\`\`c++
void main() {
  return(0);
}
\`\`\`

Result:
```c++
void main() {
  return(0);
}
```

BBBits, 2018-08-24:
Thanks , that makes it Very readable

jgromes, 2018-08-24:
I don't know what could be the cause for this behavior; I've previously run the library on ESP32 without any issues.

SPI bus is initialized inside Module class using the default `SPI.begin()`. There's no way for users to modify that and I don't think it's a good idea to expose this part of the library to the public API. Pin assignment is done in the contructor. You have used the default one `SX1278 fsk = new Lora;` which assumes that slave select pin is pin 7, DIO0 is pin 2 and DIO1 is pin 3. To change the pins, simply pass them to the constructor `SX1278 fsk = new Lora(4, 5, 6);` or whatever matches your setup.

Also, you're not using the settings correctly; see the `Settings` example for details.

All of the above is well documented in examples as well as on the [Wiki](https://github.com/jgromes/LoRaLib/wiki).

BBBits, 2018-08-24:
ok.
Perhaps it was the default SS pin 7 that created the problem on my modules.
Are there esp32 modules that expose GPIO7 ? 
I have 0,1,2,3,4,5,12,13,14,15,16,17,18,19,21,22,23,27,26,27,32,33,34,35,36,37,38,39
Interestingly , GPIO7 doesn't have a specific pin definition but GPIO26 can be RTC_GPIO7
also SD_DATA_0 can be GPIO7 .

Haven't figured a way round the problem without removing the flash .
I have managed to erase the flash after taking it out & then put it back .
I can then reprogram the normal way .
found this little utility for a ch341 board that works on win10 
https://www.win-raid.com/t796f16-Guide-Using-CH-A-based-programmer-to-flash-SPI-EEPROM-6.html

BBBits, 2018-08-24:
just modified Module.h
Module(int cs = 18, int int0 = 26, int int1 = 3);
compiled/ loaded /ran  fine without any problems .
recon it's GPIO7 that munted the world as SS pin .
Not planning on trying all the other pins as SS , cause if it does fail , I gotta take the chip out to fix it.

jgromes, 2018-08-24:
You don't have to modify Module class definiton, just use

`SX1278 fsk = new LoRa(18, 26, 3);`

> recon it's GPIO7 that munted the world as SS pin

If pin 7 is this dangerous on ESP32 boards, it might be worth changing the default pin number. The only reason for me to use pin 7 was that this library was originaly created as a driver for LoRenz shield, which has SS on pin 7.

BBBits, 2018-08-24:
perhaps it needs to be mcu specific then if it is a generic Lib
the heltec wifi lora module & the one  i posted a link to above both use 18 for SS & 26 for INT0

BBBits, 2018-08-24:
hasty folks like myself do tend to compile first THEN read the docs if it doesn't work .......

BBBits, 2018-08-24:
thanks for the lib , will get some sleep & continue my tests in the morning

jgromes, 2018-08-24:
I could not reproduce the flash corruption issue, but I did find out that GPIO7 can not be used as SS pin (`begin()` methods return code -2 `ERR_CHPI_NOT_FOUND`).

Changed the default SS pin to GPIO18 on ESP32 boards, and it runs fine. Are you OK with GPO18 as the default?

BBBits, 2018-08-25:
Hi , SS as 18 is good for me.
I see in SX127x::beginFSK the init defaults to INT_BOTH 
I think that could cause problems as well , depending on where beginfsk is located in the user's code.

On the modules I use , there is a Reset & INT_0  but no INT_1
I haven't managed to get a tx & rx working on my modules yet.
do you have a simple sketch that works with your hardware to transmit & receive 'helloworld' ?

BBBits, 2018-08-25:
There is a typo in SX12x.cpp in 
```c++
int16_t SX127x::transmit(String& str, uint8_t addr)
return(SX127x::transmit(str.c_str()), addr);  extra bracket after str.c_str()
&
int16_t SX127x::transmit(const char* str, uint8_t addr) 
return(SX127x::transmit((uint8_t*)str, strlen(str)), addr); extra bracket after strlen(str)
```
The compiler doesn't say anything but the functions always return zero (ERR_NONE)

jgromes, 2018-08-25:
Changed the default SS pin in 1327d3be69e522440ccfcf46b9b6ed616280ba51 and fixed the typo in 8bf5481b72260add44453715bf74a71c8e093c4a

The most basic examples are `Transmit` and `Receive` - to use them with FSK modem, you only have to call `beginFSK()` instead of `begin()`.

INT1 is required for the library to work, SX1278 sends Rx timeouts on that pin only.

In FSK mode, Tx timeout occurs when SX1278 doesn't issue interrupt `PacketSent` within 150% of expected transmission (based on packet length and data rate). I could reproduce this issue on my ESP32, could you please check your wiring?

jgromes, 2018-08-25:
You could try disabling the Tx timeouts entirely: comment out [this line](https://github.com/jgromes/LoRaLib/blob/master/src/modules/SX127x.cpp#L214) to [this line](https://github.com/jgromes/LoRaLib/blob/master/src/modules/SX127x.cpp#L217) . If transmission goes into infinite loop, the module never sends out the interrupt.

jgromes, 2018-08-25:
LoRa mode will not work without acces to DIO1 - it'S the only one where SX1278 sends Rx timeouts. FSK mode timeouts are calculated locally.

BBBits, 2018-08-25:
I replaced all occurrences of
```c++  digitalRead(_mod->int1())  with int1_triggered()
```
to attempt a resolution

jgromes, 2018-08-25:
DIO1 isn't used in FSK mode.

BBBits, 2018-08-25:
yep , but was trying to get the receive  working in lora . the transmit in lora is working .

jgromes, 2018-08-25:
You can try running the examples in debug mode - uncomment [this line](https://github.com/jgromes/LoRaLib/blob/dbb3855db21fd93c69ccdffe2cde56a8c135a491/src/TypeDef.h#L10)

There will be additional output in serial monitor.

BBBits, 2018-08-25:
Already done that , there is no extra info to enlighten me to the problem.

jgromes, 2018-08-25:
That means it's not an issue in SPI communication - it would have shown any failed writes.

jgromes, 2018-08-25:
> On the other Lora Lib , I Init like this.
>` SPI.begin(5, 19, 27, 18);`

What does this line of code do?

BBBits, 2018-08-25:
void SPIClass::begin(int8_t sck, int8_t miso, int8_t mosi, int8_t ss)

jgromes, 2018-08-25:
I am unable to reproduce any of the issues you have described - both LoRa and FSK mode transmit and receive are working fine for me. My setup is just an ESP32 board with standalone SX1278 connected to it, so I suspect it has something to do with your hardware.

BBBits, 2018-08-25:
In the receive routine , you set dio0 irq to SX127X_DIO0_PACK_PACKET_SENT 
Is that correct ?
```c++
  } else if(modem == SX127X_FSK_OOK) {
    // set mode to standby
    int16_t state = setMode(SX127X_STANDBY);
    
    // set DIO pin mapping
    state |= _mod->SPIsetRegValue(SX127X_REG_DIO_MAPPING_1, SX127X_DIO0_PACK_PACKET_SENT, 7, 6);
```

jgromes, 2018-08-25:
That's correct, the name means that in packet mode (PACK), interrupt PacketSent will be sent to DIO0.
If it was wrong, it wouldn't work with any hardware.

BBBits, 2018-08-25:
it's the name that is confusing .
I would think that the name SX127X_DIO0_PACK_PAYLOAD_READY would be more appropriate in a receive routine

#define SX127X_DIO0_PACK_PAYLOAD_READY                0b00000000  //  7     6     
#define SX127X_DIO0_PACK_PACKET_SENT                  0b00000000  //  7     6

jgromes, 2018-08-25:
Thanks for reporting all the typos/incorrect names etc. It's all fixed in release 5.3.0.

Since I wasn't able to reproduce any of the problems on my ESP32, I'm closing this issue for now. Feel free too reopen/post new comments if you need some help.

closed 2018-08-25 by jgromes

BBBits, 2018-08-28:
Update: 
I reloaded the lib after the recent changes & could run the examples on the heltec clones.
Ye Ha .. Thanks.
While messing with FSK , I found that all the rest of the 433mhz gear in my house stopped working .
It turns out that the Carrier stays on after a transmit in FSK mode .
putting the radio into standby resolves it .
Thought you might like to add a note to your wiki / examples about it because you can end up with a 433mhz Jammer otherwise...

jgromes, 2018-08-28:
Leaving the carrier on in direct mode is required for RTTY support, I will mention that in the documentation once I have finalized the direct mode access methods (I'm plying around with direct mode receive a little bit).

jgromes, 2018-09-04:
In release 6.0.0, the `directMode` method was renamed `transmitDirect()` so this hopefully gives the user the idea that calling this method will turn on the transmitter. It's also mentioned in the example code and on the Wiki.

Thanks for the suggestion, it was a good idea to add that.