jgromes/LoRaLib #46: Receive directMode()/tone() transmissions with another SX1278 module?

Bambofy, 2018-08-26:


jgromes, 2018-08-26:
> No description provided.

What's the use case?

Bambofy, 2018-08-26:
Two SX1278 modules and 2 Arduino Nanos, but no Radio->PC interface.

jgromes, 2018-08-26:
You can receive data in direct mode as a stream of bits on DIO2 (with clock on DIO1).

When receiving FM tones, the stream will be alternating sequences of 0s and 1s (because the original tone() output was a square wave signal), so you can (probably) calculate the tone() frequency out of that.

I'm pretty sure you won't be able to receive RTTY signals with SX1278. That would require measuring frequency error, which can only be measured in packet mode during preamble reception.

jgromes, 2018-08-26:
Direct mode is only available for FSK modem, not for LoRa.

Bambofy, 2018-08-26:
kk thanks

jgromes, 2018-08-26:
I'm still confused why do you want to receive direct mode data and/or FM tones? Surely neither provides any advantages over FSK/LoRa packet mode?

Bambofy, 2018-08-31:
Hi,

We have to transmit direct mode RTTY data as a requirement for our project along with LoRa and normal FSK packet mode transmissions.

jgromes, 2018-08-31:
OK, I get that, but why you want to receive them with SX1278? Any basic SDR dongle can receive RTTY and FM tones.

Bambofy, 2018-09-01:
SDR dongle is on the way (taking a few weeks to arrive). RTTY for downlink, LoRa and FSK packets for uplink.