jgromes/LoRaLib #47: Large bandwidth in direct mode AFSK?

Bambofy, 2018-09-28:
We are transmitting at 2522hz and 183hz shift with direct mode FSK. But the received output is at a bandwidth of 100khz, any ideas at what might be going wrong?

Thanks,
Richard

jgromes, 2018-09-28:
Can you please share your code when opening issues? Then I'll be able to help you much quicker.

Bambofy, 2018-09-28:
`#include <LoRaLib.h>
 
SX1278 lora = new LoRa;
int RTTY_BASE = 2125; // Hz
int RTTY_SHIFT = 182; //183; // 3 * 61Hz
int RTTY_BAUD_RATE = 22222; // micro-seconds (us) 45.5 baud
int DIGITAL_OUT_SX1278_DIRECT = 10;
 
void setup() {
  Serial.begin(9600);
}
 
 
//////////////////////////////////
// RTTY Protocol implementation //
//////////////////////////////////
void Communication_RTTY_TransmitMark()
{
  unsigned long start = micros(); // MICRO seconds.
 
  tone(DIGITAL_OUT_SX1278_DIRECT, RTTY_BASE - RTTY_SHIFT);
 
  while (micros() - start < RTTY_BAUD_RATE);
 
  noTone(DIGITAL_OUT_SX1278_DIRECT);
}
 
void Communication_RTTY_TransmitSpace()
{
  unsigned long start = micros(); // MICRO seconds.
 
  tone(DIGITAL_OUT_SX1278_DIRECT, RTTY_BASE);
 
 
  while (micros() - start < RTTY_BAUD_RATE);
 
  noTone(DIGITAL_OUT_SX1278_DIRECT);
}
 
void Communication_RTTY_TransmitBit(char inChar)
{
  Communication_RTTY_TransmitSpace();
 
  // for each bit in char (8-bits) (only use 7 bits)
  char m = 0x01; // mask
  int bitIndex = 0;
  for (m; m; m <<= 1)
  {
    if (bitIndex == 7) break; // ascii is 7 bit
   
    // if the char bit is selected by the mask...
    if (inChar & m)
    {
      // send a 1 (MARK) = base - shift
      Communication_RTTY_TransmitMark();
    }
    else
    {
      // send a 0 (SPACE) = base
      Communication_RTTY_TransmitSpace();
    }
   
    bitIndex++;
  }
 
  Communication_RTTY_TransmitMark();
}
 
/*
 * @brief send a string through the RTTY protocol.
[... 822 lines cut ...]
  Communication_RTTY_TransmitSpace();
  Communication_RTTY_TransmitSpace();
  Communication_RTTY_TransmitMark();
  Communication_RTTY_TransmitSpace();
 
  Communication_RTTY_TransmitMark();
 
  // LF
  Communication_RTTY_TransmitSpace();
 
  Communication_RTTY_TransmitSpace();
  Communication_RTTY_TransmitMark();
  Communication_RTTY_TransmitSpace();
  Communication_RTTY_TransmitSpace();
  Communication_RTTY_TransmitSpace();
 
  Communication_RTTY_TransmitMark();
    */
 
  lora.standby();
 
  delay(1000);
}`

[image]

jgromes, 2018-09-29:
You're initializing the module without any parameters and never changing the bandwidth. Default value is 125 kHz, so this is output seems fine to me. What bandwidth value do you expect?

Bambofy, 2018-09-29:
Hi we just expect a single peak at 434mhz not sure why there's another peak at the 100khz bandwidth limit?

jgromes, 2018-09-29:
Because your frequency deviation is (by default) set to 50 kHz - each peak is offset 50 kHz from center frequency. You must set frequency deviation to 0 to get a single peak.

Side note: Judging by your (quite convoluted) code, you're trying to transmit AFSK RTTY. Why not just a normal FSK RTTY? That's the more "standard" approach.

Bambofy, 2018-09-29:
Thanks for the advice, chose asfk rtty for more backwards compatibility and also kitelib implementation does not decode with mmtty

jgromes, 2018-09-29:
Were you using ASCII (7 or 8 data bits) or ITA2 (5 bits)?

See https://github.com/jgromes/KiteLib/issues/2

Bambofy, 2018-09-29:
ITA2 5 bit

jgromes, 2018-09-30:
ITA2 support was added to KiteLib after initial RTTY implementation; make sure you're using the latest version.

closed 2018-09-30 by jgromes