jgromes/LoRaLib #48: lora

bakhshizadeh, 2018-10-20:
please help me.how can i transmit data after receive data in receiveIntterupt.ino

jgromes, 2018-10-20:
Have you tried `lora.transmit()`? Seems rather obvious to me.

bakhshizadeh, 2018-10-20:
yes

jgromes, 2018-10-20:
And? Was there an error code? Did it not transmit? I need more information in order to help you.

bakhshizadeh, 2018-10-20:
/*
   LoRaLib Receive with Inerrupts Example
   This example listens for LoRa transmissions and tries to
   receive them. Once a packet is received, an interrupt is
   triggered. To successfully receive data, the following
   settings have to be the same on both transmitter 
   and receiver:
    - carrier frequency
    - bandwidth
    - spreading factor
    - coding rate
    - sync word
    - preamble length
   For more detailed information, see the LoRaLib Wiki
   https://github.com/jgromes/LoRaLib/wiki
*/

// include the library
#include <LoRaLib.h>
#include <avr/sleep.h>

int wakePin = 3;                 // pin used for waking up

void wakeUpNow()        // here the interrupt is handled after wakeup
{
  // execute code here after wake-up before returning to the loop() function
  // timers and code using timers (serial.print and more...) will not work here.
  // we don't really need to execute any special functions here, since we
  // just want the thing to wake up
}

// create instance of LoRa class using SX1278 module
// this pinout corresponds to LoRenz shield: 
// https://github.com/jgromes/LoRenz
// NSS pin:   7 (18 on ESP32 boards)
// DIO0 pin:  2
// DIO1 pin:  3
// IMPORTANT: because this example uses external interrupts,
//            DIO0 MUST be connected to Arduino pin 2 or 3.
//            DIO1 MAY be connected to any free pin 
//            or left floating.
SX1278 lora = new LoRa;

void setup() {
  pinMode(wakePin,INPUT);
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("Initializing ... "));
  // carrier frequency:           433.0 MHz
  // bandwidth:                   125.0 kHz
  // spreading factor:            9
  // coding rate:                 7
  // sync word:                   0x12
  // output power:                17 dBm
  // current limit:               100 mA
  // preamble length:             8 symbols
  // amplifier gain:              0 (automatic gain control)
  int state = lora.begin();
  lora.setFrequency(433.0);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // set the function that will be called 
  // when new packet is received
[... 157 lines cut ...]
      Serial.println(" dBm");

      // print frequency error
      Serial.print("Frequency error:\t");
      Serial.print(lora.getFrequencyError());
      Serial.println(" Hz");
  
    } else if (state == ERR_CRC_MISMATCH) {
      // packet was received, but is malformed
      Serial.println("CRC error!");
  
    }
    // we're ready to receive more packets,
    // enable interrupt service routine
    enableInterrupt = true;
   // delay(2000);
    state = lora.startTransmit("Mahmood"); 
    if (state =! ERR_NONE){
      Serial.print(F("failed, code = "));
      Serial.println(state);
    }
    //delay(2000);
  }

}

bakhshizadeh, 2018-10-20:
in below code state variable show 1
Serial.print(F("failed, code = "));
Serial.println(state);

bakhshizadeh, 2018-10-20:
i use lora.transmit() but i have the same problem.

bakhshizadeh, 2018-10-20:
if Error codes be negative, that works correct?

jgromes, 2018-10-20:
https://github.com/jgromes/LoRaLib/wiki/Error-Codes

bakhshizadeh, 2018-10-20:
may you send me a transiver code with lora?please

jgromes, 2018-10-20:
Sorry, but this is not how GitHub issues work. This is a place you go to if you have found a bug in the library, or if you want to suggest new features. For help with Arduino code in general, go to Arduino formus.

I was unable to reproduce your problem, which leads me to believe it is not a bug in the library code. That means I have to close this issue now.

Few pointers to help you:
1. Don't try to do everything at the same time. Start with code that can transmit after reception. Only add new features after you're confident that is working.
2. Read the examples and the [wiki](https://github.com/jgromes/LoRaLib/wiki). One of the comments in ReceiveInterrupt says that listen mode will be disabled by calling - among other things - `lora.transmit()`. Do you re-enable the listen mode?

bakhshizadeh, 2018-10-20:
i think so.i download LoRalib-master folder today. do i re-enable it?

bakhshizadeh, 2018-10-27:
hi.excuse me.i transmit data and receiver side receive data but i have error code -16 in transmit side.may you help me?i know error -16 is faulty spi wiring.

bakhshizadeh, 2018-10-27:
> Code -16 means that expected value in some SPI register did not match expected value after writing. This can be caused by too long SPI wires, or general isntability of the system. You can get additional information (such as which register failed) by running LoRaLib in debug mode (uncomment this line in TypeDef.h and re-upload sketch).

can you help me that where is my problem?address   0x1bits    2  0value   0b11current 0b10001001mask 0b111new 0b10001011read 0b0

jgromes, 2018-10-27:
This error occured when switching the LoRa module from standby mode to transmit mode. For some reason, the value read after writing was 0b00000000 instead of the expected 0b10001011. Value 0b00000000 seems very suspicious - it looks like the chip stopped responding to SPI read commands. In any case this is most likely hardware related: too long wires, strong EM noise from the background or power supply issues.

One workaround I can think of is increasing delay between SPI write and the vrite verification. Go to [this line](https://github.com/jgromes/LoRaLib/blob/032bd60184f55b4b7c0eaa57e092935257c7ecdc/src/Module.cpp#L63) in Module.cpp and change the delay to 10 ms.

bakhshizadeh, 2018-10-27:
is my hardware correct in attached photo?

jgromes, 2018-10-27:
Please reply on GitHub, not to the email ...

jgromes, 2018-10-27:
What exactly am I supposed to tell from that photo? None of the pins on the LoRa module are labeled, and thanks to that flash I can't even tell which wire goes where.

With that in mind, you are probably using module with 3,3 V logic on 5 V Arduino. It might be (kind of) working now, but eventually, it will start smoking.

jgromes, 2018-10-27:
Also, none of this has any relation to the library. Go to Arduino forums to get help with Arduino in general.

closed 2018-10-27 by jgromes

bakhshizadeh, 2018-10-28:
hi.can you help me that how connect arduino uno r3 board to sx1278 in lora mode?

jgromes, 2018-10-28:
From my previous comment:
> Go to Arduino forums to get help with Arduino in general.

bakhshizadeh, 2018-11-19:
Hi.excuse me.may you help me? i receive data in interruptreceive and send data to other lora up to 4 character very well. But when i want to send 5 or more character it doesnot work well.

bakhshizadeh, 2018-12-23:
hi.excuse me.how to put lora sx1278 in standby or sleep mode and listen mode? i want to decrease power consumption.

jgromes, 2018-12-29:
Once again, this has nothing to do with the library.

From [SX1278 datasheet](https://www.semtech.com/uploads/documents/DS_SX1276-7-8-9_W_APP_V5.pdf), page 14, supply current in receive mode will always be 11 - 12 mA. You can decrease the overall power consumption by putting the module to sleep mode when you don't need it.

Locking as off-topic.