jgromes/LoRaLib #49: Logo proposal

Batarian711, 2018-10-27:
Hello. I design a logo for open source projects voluntarily. Do you want to help?

jgromes, 2018-10-27:
Hi, that would be cool! Can you show me some examples of your work? Also, what sort of license would the logo have?

Batarian711, 2018-10-29:
My logos can visit my profile. Also licensed: http: //creativecommons.org/licenses/by/4.0/

jgromes, 2018-10-29:
Guess the license won't be an issue then, if the logo is licensed under CC and the rest under MIT.

Do you have some ideas for the logo? I was thinking something like the original LoRa logo. We probably won't be able to use that directly though, since it's trademarked.

jgromes, 2018-11-12:
Closed due to inactivity, feel free to reopen later.

closed 2018-11-12 by jgromes