jgromes/LoRaLib #50: Incomplete RSSI calculation

GillesC, 2018-10-30:
According to the [SX127X datasheet](https://www.semtech.com/uploads/documents/DS_SX1276-7-8-9_W_APP_V5.pdf) the RSSI formula depends on the SNR.
Only the case for SNR > 0 is implemented [here](https://github.com/jgromes/LoRaLib/blob/8a3aa2cf6ed203860b102e5dc407cf578e303c4e/src/modules/SX127x.cpp#L299).

jgromes, 2018-10-30:
I don't think it does; the RSSI calculation only depends on whether HF or LF port is used. The thing that depends on SNR is something the datasheet calls "packet strength".

Although you're right that the calculation doesn't take into accout used port or the difference between SX1272 and SX1278:

```
RSSI(SX1272) = -139 + REG_RSSI_VALUE
RSSI(SX1278_HF) = -157 + REG_RSSI_VALUE
RSSI(SX1278_LF) = -164 + REG_RSSI_VALUE
```

Currently the only thing that's implemented is HF SX1278.

jgromes, 2018-10-30:
After studying the datasheets a bit more, it looks like you were right after all. There's a difference between _Rssi_ and _PacketRssi_, so the correct equations look like this:

RSSI(SX1272, SNR >= 0)  = -139 + _PacketRssi_
RSSI(SX1272, SNR < 0)  = -139 + _PacketRssi_ + _PacketSnr_ * 0.25

RSSI(SX1278, HF, SNR >= 0)  = -157 + _PacketRssi_
RSSI(SX1272, HF, SNR < 0)  = -157 + _PacketRssi_ + _PacketSnr_ * 0.25

RSSI(SX1278, LF, SNR >= 0)  = -164 + _PacketRssi_
RSSI(SX1272, LF, SNR < 0)  = -164 + _PacketRssi_ + _PacketSnr_ * 0.25

jgromes, 2018-10-31:
Should be fixed in release 7.0.0, thanks for reporting!

The original member variables `lastPacketRSSI` and `lastPacketSNR` are now replaced with methods `getRSSI()` and `getSNR()`.

closed 2018-10-31 by jgromes