jgromes/LoRaLib #51: wemos d1 mini and sx1278 connection

alemori, 2018-11-13:
I would like to know if someone made a connection between wemos d1 mini and a lora sx1278 with its corresponding scketch of sending and receiving. Thank you

jgromes, 2018-11-13:
Hi, I don't have a wemos board, so I can't check, but it shouldn't be all that hard. Just connect the SPI bus, the power and DIO0 + DIO1 pins and try the example sketches.

Side note: this can be considered off topic. Please use Arduino forums to ask generic questions, and only use GitHub issues for library-specific questions. Thank you.

alemori, 2018-11-13:
Thank you for your answer. Connect the interface accordingly and the pins for DIO0 and DIO1, the problem is that it gives me Waiting for incoming transmission ... -6

jgromes, 2018-11-13:
We have a wiki here you know ;) https://github.com/jgromes/LoRaLib/wiki/Error-Codes

Code -6 is receiver timeout, unless you're constantly transmitting that's expected.

alemori, 2018-11-13:
Estoy utilizando un arduino nano y un wemos mini, las conexiones creo que estan bien hecha. Para el caso del arduino nano utilizo el default NSS pin:   7  DIO0 pin:  2  DIO1 pin:  3. y utilizo los ejemplos de transmit y receive. Si el que transmite es el wemos y el que recibe es el nano, entonces funciona bien, pero si es a la inversa me da un error de timeout y un vuelco de memoria. Que puede ser?

alemori, 2018-11-13:
I'm using an arduino nano and a wemos mini, the connections I think are well made. For arduino nano I use the default NSS pin: 7 DIO0 pin: 2 DIO1 pin: 3. and I use the examples of transmit and receive. If the one that transmits is the wemos and the one that receives is the nano, then it works well, but if it's the other way around it gives me a timeout error and a memory dump. What can it be?

jgromes, 2018-11-14:
> it gives me a timeout error and a memory dump
Probably should have mentioned that when creating the issue.

Stack traces from ESP8266 can be decoded using https://github.com/me-no-dev/EspExceptionDecoder
Can you post the decoded trace here?

alemori, 2018-11-15:
I send you the trace that was generated. Thanks for the help
0x40202abc: Module::SPIwriteRegister(unsigned char, unsigned char) at C:\Users\alejandro_2\Documents\Arduino\libraries\LoRaLib\src/Module.cpp line 147
0x40202a00: Module::SPIreadRegister(unsigned char) at C:\Users\alejandro_2\Documents\Arduino\libraries\LoRaLib\src/Module.cpp line 138
0x402037be: SX127x::clearIRQFlags() at C:\Users\alejandro_2\Documents\Arduino\libraries\LoRaLib\src\modules/SX127x.cpp line 304
0x40203940: SX127x::receive(unsigned char*, unsigned int) at C:\Users\alejandro_2\Documents\Arduino\libraries\LoRaLib\src\modules/SX127x.cpp line 177
0x40100584: malloc at C:\Users\alejandro_2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc/umm_malloc.c line 1668
0x402047e6: String::operator=(String&&) at C:\Users\alejandro_2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/WString.cpp line 232
0x4020398b: SX127x::receive(String&, unsigned int) at C:\Users\alejandro_2\Documents\Arduino\libraries\LoRaLib\src\modules/SX127x.cpp line 187
0x4020477a: String::String(char const*) at C:\Users\alejandro_2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/WString.cpp line 41
0x402040c0: Print::print(char const*) at C:\Users\alejandro_2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Print.cpp line 102
0x402027ce: loop at C:\Users\ALEJAN~1\AppData\Local\Temp\arduino_modified_sketch_168713/Receive.ino line 55
0x40204939: esp_schedule at C:\Users\alejandro_2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/core_esp8266_main.cpp line 95
0x402049a4: loop_wrapper at C:\Users\alejandro_2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/core_esp8266_main.cpp line 125
0x401000e5: cont_wrapper at C:\Users\alejandro_2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/cont.S line 81

alemori, 2018-11-15:
I also took out the comment to //#define KITELIB_DEBUG to do the debug and it gives me 
SX127x not found! (1 of 10 tries) SX127X_REG_VERSION == 0x0000, expected 0x0012
Found SX127x!
success!
Waiting for incoming transmission ... 
Exception (28):
epc1=0x40206076 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffffd50 end: 3fffffd0 offset: 01a0

jgromes, 2018-11-15:
I think you're using old version of the library. Stack trace is reporting lines that don't exist in the current code (e.g. Module.cpp line 147, but current Module.cpp only has 132 lines).

There was a bug similar to this on ESP32 (#30) that was fixed in 5.0.2.

alemori, 2018-11-15:
Hi, I did have an old library, update it to the latest version, and I'll leave the trace with the error again. Thank you

Decoding 12 results
0x40202b20: Module::SPIwriteRegister(unsigned char, unsigned char) at C:\Users\ALEJAN\Documents\Arduino\libraries\LoRaLib\src/Module.cpp line 132
0x40202a25: Module::SPIgetRegValue(unsigned char, unsigned char, unsigned char) at C:\Users\ALEJAN\Documents\Arduino\libraries\LoRaLib\src/Module.cpp line 47
0x402040da: SX127x::clearIRQFlags() at C:\Users\ALEJAN\Documents\Arduino\libraries\LoRaLib\src\modules/SX127x.cpp line 605
0x402041fd: SX127x::receive(unsigned char*, unsigned int) at C:\Users\ALEJAN\Documents\Arduino\libraries\LoRaLib\src\modules/SX127x.cpp line 605
0x40100584: malloc at C:\Users\ALEJAN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc/umm_malloc.c line 1668
0x402055fa: String::operator=(String&&) at C:\Users\ALEJAN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/WString.cpp line 232
0x4020435b: SX127x::receive(String&, unsigned int) at C:\Users\ALEJAN\Documents\Arduino\libraries\LoRaLib\src\modules/SX127x.cpp line 605
0x4020558e: String::String(char const*) at C:\Users\ALEJAN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/WString.cpp line 41
0x40204f4c: Print::print(char const*) at C:\Users\ALEJAN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Print.cpp line 102
0x402027de: loop at C:\Users\ALEJAN\AppData\Local\Temp\arduino_modified_sketch_726786/Receive.ino line 59
0x402057b8: loop_wrapper at C:\Users\ALEJAN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/core_esp8266_main.cpp line 125
0x401000e5: cont_wrapper at C:\Users\ALEJAN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/cont.S line 81

jgromes, 2018-11-16:
I  think I know what's going on, but I need you to verify it. Go to src/modules/SX127x.cpp and change line [234](https://github.com/jgromes/LoRaLib/blob/3cac138e1081a96661d940510c14fdd6b3e0a9ee/src/modules/SX127x.cpp#L234) from this:
```c++
char* data = new char[len];
```
to this:
```c++
char* data = new char[len + 1];
```

Upload the sketch again and see if it still crashes.

alemori, 2018-11-16:
Hello, I made the correction, and indeed began to transmit but with errors between shipments. I copy the trace and the successful transmission. Thank you

Decoding 14 results
0x40100615: _umm_realloc at C:\Users\ALE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc/umm_malloc.c line 1499
:  (inlined by) realloc at C:\Users\ALE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc/umm_malloc.c line 1727
0x40202b18: Module::SPIwriteRegister(unsigned char, unsigned char) at C:\Users\ALE\Documents\Arduino\libraries\LoRaLib\src/Module.cpp line 132
0x402054bb: String::changeBuffer(unsigned int) at C:\Users\ALE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/WString.cpp line 41
0x401002ac: _umm_free at C:\Users\ALE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc/umm_malloc.c line 1295
0x401002ac: _umm_free at C:\Users\ALE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc/umm_malloc.c line 1295
0x4010077c: free at C:\Users\ALE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc/umm_malloc.c line 1755
0x40205680: operator delete[](void*) at C:\Users\ALE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/abi.cpp line 98
0x40204361: SX127x::receive(String&, unsigned int) at C:\Users\ALE\Documents\Arduino\libraries\LoRaLib\src\modules/SX127x.cpp line 605
0x40205586: String::String(char const*) at C:\Users\ALE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/WString.cpp line 41
0x40204f44: Print::print(char const*) at C:\Users\ALE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Print.cpp line 102
0x402027de: loop at C:\Users\ALE\Documents\Arduino\Receive2\Receive2.ino/Receive2.ino.ino line 59
0x40205745: esp_schedule at C:\Users\ALE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/core_esp8266_main.cpp line 95
0x402057b0: loop_wrapper at C:\Users\ALE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/core_esp8266_main.cpp line 125
0x401000e5: cont_wrapper at C:\Users\ALE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/cont.S line 81

requency error:	1444.94 Hz
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... success!
Data:			Hello World!
RSSI:			-43 dBm
SNR:			13.25 dB
Frequency error:	1432.35 Hz
Waiting for incoming transmission ...

jgromes, 2018-11-16:
How often does it happen? Is it before each timeout or before successfully receiving packet?

jgromes, 2018-11-16:
Strange, now the exception is 3 instead of 28. It's also weird that it seems to be crashing at random.

Two more things for you to check out:
1. Try using the binary version of receive(). You have to create the buffer that will store the received 
  data yourself:
```c++
byte byteArr[12];
int state = lora.receive(byteArr, 12);
```
and to print it after packet was received:
```c++
for(int i = 0; i < 12; i++) {
  Serial.write(byteArr);
}
Serial.println();
```

This will bypass all dynamic memory allocation.

2. Remove all references to the flash string macro `F()`, e.g. on line 34 in the Receive example replace 
```c++
Serial.print(F("Initializing ... "));
```
with
```c++
Serial.print("Initializing ... "));
```
Same on lines 46 and 48. Also, disable the debug mode here. From what I've seen ESP exception 3 is relating to storing/loading flash data.

jgromes, 2018-11-17:
Sorry, I meant

for(int i = 0; i < 12; i++) {
  Serial.write(byteArr[i]);
}
Serial.println();

alemori, 2018-11-17:
Hello, With the change to receive the data with the vector, it doesn't give any error, but it doesn't show me Hello World and the RSSI data is at 0. 
Thanks 
Data:			RSSI:			92 dBm
SNR:			0.00 dB
Frequency error:	0.00 Hz
Waiting for incoming transmission ... ÐÚÿ?
Data:			RSSI:			92 dBm
SNR:			0.00 dB
Frequency error:	0.00 Hz
Waiting for incoming transmission ... ÐÚÿ?

jgromes, 2018-11-17:
This behavior is very strange, I see no reason why it shouldn't receive data and correctly read RSSI, SNR and frequency error now. Which pins are you using for NSS/DIO0/DIO1 on the wemos?

jgromes, 2018-11-17:
Got myself a Wemos D1 R1. It's not the mini you have, but it's still using ESP8266. I can confirm the original bug (exception 28) which was related to dynamic memory allocation. Unfortunately, I can't reproduce any other of your problems, it's running fine for me without any exceptions - even using the same connections as you and with unmodified Receive example.

Thank you for reporting the original bug, that will be fixed in the next library update. Since I can't reproduce anything else, I don't think the issue is in the library code. I suggest you check your wiring, try out other pins for NSS/DIO0/DIO1. If that's not the issue than maybe try a different wemos board.

closed 2018-11-17 by jgromes

alemori, 2018-11-17:
Thank you very much. I'll try a different wiring to see if the problem goes through there. Anyway how would it be if I want to use the byte array option and convert what I receive into string? thank you again for your time.

jgromes, 2018-11-17:
`String str = String(byteArr);`

The disadvantage is that you need to know the length of the packet in advance.

alemori, 2018-11-17:
Thank you very much!!!