jgromes/LoRaLib #56: need help to emulate Time-division duplex with SX1272 modules

Amaury-esig, 2019-01-10:
Hey, I'm trying to emulate Time-division duplex with SX1272 modules and I don't receive the messages entirely, after that I don't receive anything from one of the modules anymore. This problem completely blocks my project. Could you help me?

jgromes, 2019-01-10:
Hi, I had no issues with time division duplex on SX1272 in the past, so I doubt this is a bug in the library. Can you post your code so that I can check and make sure? Thanks.

ddrmax, 2019-01-11:
For TDDT you have to respect some kind of sync, all the information is provided in the document named AN1200.23 SX1272 Settings for LoRaWAN

you have to read the chapter 3.1 Downlink reception slots following an Uplink (LoRa Mode)
[AN1200.23 SX1272 settings for LoRaWAN.pdf](https://github.com/jgromes/LoRaLib/files/2748815/AN1200.23.SX1272.settings.for.LoRaWAN.pdf)

jgromes, 2019-01-11:
From [CONTRIBUTING.md](https://github.com/jgromes/LoRaLib/blob/master/CONTRIBUTING.md)

>5. **LoRaLib is a driver.**
This means that only features available in the SX127x chips will be implemented. LoRaLib will not implement protocols such as LoRaWAN.

It's up to the user to implement uplink/downlink slots - hence why I asked for the code. It's entirely up to him whether he want to adhere to the LoRaWAN specification or create his own.

ddrmax, 2019-01-11:
> From [CONTRIBUTING.md](https://github.com/jgromes/LoRaLib/blob/master/CONTRIBUTING.md)
> 
> > 1. **LoRaLib is a driver.**
> >    This means that only features available in the SX127x chips will be implemented. LoRaLib will not implement protocols such as LoRaWAN.
> 
> It's up to the user to implement uplink/downlink slots - hence why I asked for the code. It's entirely up to him whether he wants to adhere to the LoRaWAN specification or create his own.
it's for this that I linked that documentation as it not only benefits for lorawan, because if you read the actual doc, the RX offset and RX receive time linke (aka timeout time in single receive mode) affects link reliabilty in either LoRaWAN and non LoRaWAN modes. I have read some Thesis about hybrid TDMA / CSMA and LBT in LoRa (non LoRaWAN mode) to break the 1% or .1% limit duty cycle and boost either reliabilty and data transfer speed as I work for a retransmitting mesh network that sends a message from a principal node to other nodes

jgromes, 2019-01-11:
You are correct, however, all of this only applies to the user code, not the driver. The library has no control over when user stops transmitting and starts to receive. I can't prevent the user from trying to receive data when there's not enough preamble symbols. Again, it's up to the user to synchronize transmissions and receptions, that's how time division duplex works.

The actual receive timeout is handled internally by SX127x and is a function of spreading factor, bandwidth and coding rate. Regarding the continous receive mode - that's what the interrupt-driven `startReceive()` method does.

ddrmax, 2019-01-11:
In this case I think the same as you, it's probably a problem in is code about timing the single receive mode as it will trigger the beggining of the timeout.

The other thing is to probably implement the use of the CAD detection to launch the single receive mode (in a battery configuration it's better than continuous receive mode as the power is like 100 times less but not as good as an intervall wake up using the sleep function) 
My actual test code uses a slightly near approach to that:
` - Calculate the time sync as of AN1200.23 (AN1200.24 as I use the SX1276)
My transmit code is built like:
have a buffer list for transmitted data
call a function that do a dual timed out CAD read to see if someone transmitting (LBT mode)
Transmit data
reset the LoRa chip into contiuous CAD

My receive code is built like that:
it's called when we have a cad detect:
pass in continuous receive mode (with a built in timeout in my code)
if good lora packet ->read and prepare for retransmit
if not good (aka other sync word or SF collision) -> no data so receive mode until timeout
call a function that do a dual timed out CAD read to see if someone transmitting (LBT mode)
call my transmit function after changing hop Count (to not allow retansmitting the same data over and over)
 `

jgromes, 2019-01-25:
Closed due to inactivity.

closed 2019-01-25 by jgromes