jgromes/LoRaLib #59: LoRa SX1272 on Radino Spider

thzois, 2019-01-22:
Hello,

I am using LoRa SX1272 (radino32) on a Radino Spider. I tried to compile the code and flash it but I got several error probably due to C++ standard. One of them was ``` using LoRa = Module; ``` at LoRaLib.h. Another one was ```digitalPinToInterrupt was not declared in this scope``` at SX127x.cpp. After making some changes, I finally flashed it but I get error code -2 on Initialization. Is this library supposed to work with my components?

Thank you very much

jgromes, 2019-01-22:
Hi, it looks like your radino32 uses an ARM Cortex as the CPU. Unfortunetly, ARM architecture is not currently supported, LoRaLib only supports AVR, ESP8266 and ESP32 at this stage.

thzois, 2019-01-22:
It's a pity, you have done awesome work! Anyway, thank you very much!

jgromes, 2019-01-22:
ARM support will probably be added somewhere down the line, it's just not a priority right now. I don't even have a board to test with ;).

closed 2019-01-22 by jgromes

jgromes, 2019-01-22:
I just tried compiling the examples for Arduino Due (ARM board) and it compiled without any errors. What are you using to program the radino board?