jgromes/LoRaLib #60: Port to Linux / PR accepted?

ludiazv, 2019-02-13:
I would like to port this lib to linux SBCs.  Will you accept PRs for that purpose?

Regards,

jgromes, 2019-02-13:
That's a good question :)

Obviously it would be great to have compatibility with as many platforms as possible. At the same timethough, it is primarily Arduino library, so I can't accept anything that will compromise compatibility with Arduino platform. Also, I'm using Travis for automated testing & for doxygen, so anything we add must be compatible with that.

If you can port it with these constraints, then I will gladly accept any contributions :)

ludiazv, 2019-02-13:
Sure. This will be the approach. After a first look I think it can be ported with a minimal compatibility hal layer and just a few #if #endif inside your code. The arduino will ignore all this code. 

thnks!

ludiazv, 2019-02-14:
Hi @jgromes,
   After checking more the code, I think the port will not require a lot of time. Your code is well structured and documented. I found however a uncommon approach in the spi interface. Let me explain. You have defined 4 base methods:
- Spi read register Burst: multi byte transfer with no transaction.
- Spi read register: one byte transfer with transaction at 2Mhz-MSB ony for setting register . actual read is done outside transaction.
- Spi write register Burst: multi byte transfer with no transaction.
- Spi write register: one byte transfer with transaction at 2Mhz-MSB ony for setting register . actual read is done outside transaction.
There is any reason for implementing this that way? Why the one byte is not a special case of the burst and why one version implement transactions and other not?

As a recommendation: I think is better to better decouple the code to be able to pass the SPIClass object you are going to use in the constructor of the Module object or via a property setter setSPI. Some boards and mcu have several SPI interfaces. It will be a simple modification that will ensure flexibility.

jgromes, 2019-02-14:
The low-level SPI methods are probably the last thing that survived from the first versions of the library, so to be honest, I don't remember why some of them use beginTransaction and some not. Either way it's a good idea to pass SPIClass instead of just using the default instance. That could also allow to use software SPI implementations.

I'll add an overload to the Module constructor for setting SPI class, that's probably the safer solution since there's no need to change SPI interface at runtime.

jgromes, 2019-03-01:
Closed due to inactivity, feel free to reopen later.

closed 2019-03-01 by jgromes