jgromes/LoRaLib #61: Reduced energy consumption

GusakIurii, 2019-02-28:
Greetings, thanks for such a useful code.
I'm trying to use it on the LoRaDuino V1.2 board. (Https://www.electrodragon.com/product/atmega328p-arduino-plus-lora-sx1278-board-loraduino/)
I want to create a battery-powered receiving device.
But in order to minimize current consumption, I want my MCU to be in sleep mode, and the HPDTeK HPD14A V1.2 receiver (probably SX1278 Lora or RF95) was in standby mode and, upon detecting the preamble, woke the MCU, through an interrupt, process the incoming packet data in the future switch back to low consumption mode.
I added the MCU sleep function to the code (ReceiveInterrupt Sample), it works great. I successfully send the MCU to sleep (((LowPower.powerDown (SLEEP_FOREVER, ADC_OFF, BOD_OFF);)), by interrupt DIO0 - the MCU wakes up the data and then falls asleep. But in this mode the LoRaDuino V1.2 board consumes about 15 mA. For me 15 mA is a lot, and I also want to reduce consumption ..
When I call the lora.standby () function after the MCU I send to sleep by interrupt (LowPower.powerDown (SLEEP_FOREVER, ADC_OFF, BOD_OFF);), the board consumes about 5 mA, which is much better. But when receiving new packets, the interruption from the DIO0 or DIO1 pins does not go to the MCU and is not activated for processing ...
When I use the lora.scanChannel () function; (which reads the preamble) the data is lost and I cannot read them, and I do not understand how the MCU can be sent to sleep with this function.

Tell me, please, how with lora.standby () get the “CAD Detect” interrupt on DIO0 or DIO1?

If it is not difficult, you can make an example for this mode, I think it will be useful for many. Thanks again for the help and understanding.

[ReceiveInterrupt.zip](https://github.com/jgromes/LoRaLib/files/2916566/ReceiveInterrupt.zip)

jgromes, 2019-02-28:
By calling `lora.standby()`, you put the SX1278 chip into standby (idle) mode, which stops any ongoing operations, including channel activity detection. That's the reason for the current draw droppping from 
15 mA to 5 mA.

From [SX1278 datasheet](https://cdn-shop.adafruit.com/product-files/3179/sx1276_77_78_79.pdf), page 45, you can see that current draw is about 11 - 14 mA when in CAD mode. The way I see it, the only way to decrease power consumption in CAD is to lower the bandwidth.

divya0056, 2019-02-28:
Similar request, if possible could you please make an example for CAD driven interrupt to wake up MCU from sleep. Many Thanks

GusakIurii, 2019-02-28:
Oh, thanks for the quick reply. But this consumption is in continuous
reception mode (Full Rx). And I mean for the mode which is described in
clause 7.3.1. Wake on Preamble Interrupt, 7.3.2. Wake on SyncAddress
Interrupt (On page 117).

GusakIurii, 2019-03-01:
I understand that for this mode in the library you need to configure the Sequencer registers - page 101, 102 (RegSeqConfig) and timers (as I understand RegTimerResol), after which DIO4 will trigger an interrupt, detecting the preamble as mentioned on page 118.

jgromes, 2019-03-01:
@divya0056 The examples are supposed to show how to use the library features, not what to use them for. Furthermore, different platforms might have different ways of putting the MCU to sleep and waking it up. It's just too broad to write a meaningful example.

jgromes, 2019-03-01:
@GusakIurii The sequencer you're refering to is used in FSK mode. However, channel activity detection can only be used in LoRa mode. Which one are you using? LoRa or FSK?

GusakIurii, 2019-03-01:
> @GusakIurii Секвенсор, на который вы ссылаетесь, используется в режиме FSK. Однако обнаружение активности канала можно использовать только в режиме LoRa. Какой вы используете? ЛоРа или ФСК?

I plan to use the LoRa mode, but if in FSK mode you can achieve less consumption, then FSK mode can also be.

https://lowpowerlab.com/forum/low-power-techniques/lora-low-power-cad/ - It looks like the CAD mode in the RadioHead library has been implemented, which I need, but here the controller interrupt in time, why not interrupt it Preamble detection, I do not understand yet. Thanks for the help, I will deal with this example.

closed 2019-03-01 by jgromes

GusakIurii, 2019-03-04:
Greetings. The chip maker has a separate document [(Reading channel RSSI during a CAD)](https://www.semtech.com/uploads/documents/an1200.21_std.pdf) where it says that: "Even if the end platform does not support antenna diversity, the energy saving made while reading the RSSI during a CAD made this technique essential for all battery powered systems.
2. Channel Activity Detection principles. The Channel Activity Detection mode is designed to detect a LoRa preamble on the radio channel with the best possible power efficiency. Once in CAD mode, the SX127x will perform a very quick scan of the band to detect a LoRa packet preamble."

those. This mode of operation is provided for LoRa modulation, and it is simply necessary for battery power.
Please help me figure it out and make a separate function in the library for working in CAD mode.

jgromes, 2019-03-04:
CAD mode *is* implemented - see the ChannelActivityDetection example.

GusakIurii, 2019-03-04:
I use two examples:
On the first module: "LoRaLib Transmit with Inerrupts Example"
On the second module: "LoRaLib Channel Activity Detection Example"
LoRa module default settings.
But the second module does not receive a single packet, the message is
constantly "channel is free!"

jgromes, 2019-03-04:
I was unable to to verify this - my module can detect preamble fine in each transmission. I suggest you check the wiring (especially DIO0 and DIO1 pins).