jgromes/LoRaLib #62: changeModulation method in SX127x class

kiseitai3, 2019-03-13:
**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/LoRaLib/wiki). You might find a solution to your issue there.

**Is your feature request related to a problem? Please describe.**
I would like an official release in which you can switch the modem modulation to OOK after/during initialization of FSK.

**Describe the solution you'd like**
I already implemented the above method in SX127x. I included the files for reference. The reason for this feature is that I am attempting to build an embedded version of OpenAPS that would be as peripheral module/shield agnostic as possible. LoRaLib already covers all of the SX127x family of chips/modems including the more commonly used rfm95 in OpenAPS. However, I want to allow future users of my implementation to use whatever radio system they already have/can afford. Unfortunately, one of the target communication devices (insulin pumps) uses OOK modulation rather than FSK. Allowing the user to choose the modulation would be excellent.
[keywords.txt](https://github.com/jgromes/LoRaLib/files/2963681/keywords.txt)

[SX127x.h.txt](https://github.com/jgromes/LoRaLib/files/2963686/SX127x.h.txt)
[SX127x.cpp.txt](https://github.com/jgromes/LoRaLib/files/2963687/SX127x.cpp.txt)

jgromes, 2019-03-14:
Please use pull requests to contribute code. That way it much easier to see which parts of the library source were changed.

I'm not against adding OOK mode, but first we need to investigate the impacts: which FSK parameters will change and which ones will remain the same? If a significant portion of them would change, then it might be better to have a `beginOOK()` method, similar to `beginFSK()`.

kiseitai3, 2019-03-14:
Understood! Let me check again and compare with a driver built for the rfm95 module found here ( https://github.com/ecc1/gnarl/blob/master/lib/radio/ ).

kiseitai3, 2019-03-14:
Here's an excerpt of how this person handled OOK modulation.
```
static void set_mode(uint8_t mode) {
	write_register(REG_OP_MODE, FSK_OOK_MODE | MODULATION_OOK | mode);
	for (int w = 0; w < MAX_WAIT; w++) {
		if (read_mode() == mode) {
			return;
		}
	}
	printf("set_mode(%d) timeout in mode %d\n", mode, read_mode());
}

static inline void set_mode_sleep() {
	set_mode(MODE_SLEEP);
}

static inline void set_mode_standby() {
	set_mode(MODE_STDBY);
}

static inline void set_mode_receive() {
	set_mode(MODE_RX);
}

static inline void set_mode_transmit() {
	set_mode(MODE_TX);
}
```
Overall, it seems he handled it when setting up the mode, but your library takes care of the modulation at begin time. It looks to me that it is possible to tell the module to switch modulation scheme "on the fly." However, if you think it is best to do it at begin time, I'll leave that decision to you. I did it as a separate method so I would not have to edit all of the device drivers. :)

P.S. I noticed I introduced a mistake. Going to do a pull request so you can analyze my changes!

jgromes, 2019-03-14:
My main worry that the meaning of some FSK modulation parameters will change (e.g. section 4.2.1 of SX1278 datasheet). I will have to read through the datasheet and check what the actual transmission spectrum looks like.

You can call `begin()`/`beginFSK()` at any point during runtime, there's no reason why it should only be called in `setup()`.

kiseitai3, 2019-03-14:
In that case, I will check the datasheet as well. In the least, I can say the above snippet works for the people using the GNARL project, so hopefully the transmission spectrums are not so different. That or I will learn something new today! XD

jgromes, 2019-03-17:
Merged in #65

closed 2019-03-17 by jgromes