jgromes/LoRaLib #63: 64 byte limit on packet size for transmit under FSK/OOK

kiseitai3, 2019-03-13:
**Describe the bug**
As a follow up to my previous post about OpenAPS, some of the code that talk to devices (insulin pumps) expect up to 107 bytes of data transfer at any point in time. Is the 64 byte limit specific to FSK? For now, I removed the limit in my copy of the library, so that code talking to the target device in OOK modulation can handle the expected maximum packet size. 

**Additional info (please complete):**
 - MCU: ESP8266
 - LoRa module type expected to be rfm95 but users may have others
 - Arduino IDE version  1.8.8
 - Library version 7.1.1

jgromes, 2019-03-14:
The 64 byte limit is a hardware limitation (see section 4.2.10 of SX1278 datasheet). If you need to send packets longer than that, you will have to split them in your code.

closed 2019-03-14 by jgromes

kiseitai3, 2019-03-14:
Thanks! I will try that!