jgromes/LoRaLib #67: CRC works incorrectly, restarting the microcontroller

TomekRC, 2019-03-18:
In the case of a low signal level where  are many transmission errors and many frames are lost, the wrong data frame is returned by the function lora.readData (); but the program returns a transmission error very rarely (maybe 1 error at 100). In addition, in such a situation, the microcontroller sometimes stop working, reboots or still works properly, but the reception of data is suspended. If the quality of the received signal is high, the above problems do not occur.

Tested at:
 - MCU: Arduino Nano - ATMega328p
 - LoRa module type - SX1278 (RA-02)
 - Arduino IDE version  1.8.5 and 1.8.9
 - Library version 7.1.1 and some below version
 -Settings - lora.begin(433.5, 125.0, 7, 7, 0x14, 1, 100, 8, 0); 

[RXforTested.txt](https://github.com/jgromes/LoRaLib/files/2979699/RXforTested.txt)
[TXforTested.txt](https://github.com/jgromes/LoRaLib/files/2979700/TXforTested.txt)
[image]
[image]

jgromes, 2019-03-18:
>Library version 7.1.1 and some below version

What does it mean? Different version of the library aren't guaranteed to work together.
Also, in the Rx sketch you're only reading a single byte.

TomekRC, 2019-03-18:
My mistake in RX sketch... But If i correct the line at "int state = lora.readData(byteArr, 2);" problems still occur. The same applies to example sketches, no matter if it is set 1 or 8 bytes. The library versions in tx and rx are the same. I tested it on different versions of libraries but allways in both modules (tx and rx) I used the same versions of libraries.

jgromes, 2019-03-18:
The CRC calculation is done by the chip itself, so if there's some issue with CRC at lower power levels, I might have to disable it. I'll check with my setup.

TomekRC, 2019-03-18:
At 17 dBm and Spreading Factor = 9 with few km between building is the same problem.

jgromes, 2019-03-19:
Could you check if the issue persists with output power set to 2 dBm? When power is set lower than 2, the RF output is on different pin, which might not be connected on your module - I suspect it's not connected on most of the cheap SX127x modules.

I was unable to test with lower than 2 dBm, but everything seems stable at 2 - messages are delivered reliably and CRC checks pass.

Regarding the reset/hangs, I would suspect a hardware issue.

TomekRC, 2019-03-19:
I'll check it in the evening at 2dBm.
I have modules RA-02, they have only one output rf. It is possible that the pin from low power is not connected, the more that when setting below 2db the range drops drastically to several meters.
Reset occurs on both plates I test. Over the weekend I will check it on the version 3.3v arduino mini.
Have you tested your modules on the edge of the range where many frames are lost? When snr > -6db (SF=7) problems do not take place

jgromes, 2019-03-19:
Are you using levle shifter between the lora module and Arduiino nano? Nano has 5 V logic, but the lora module has 3.3 V logic - connecting them directly can damage the lora module.

TomekRC, 2019-03-19:
Off course i am using level shifter.

jgromes, 2019-03-19:
Good, just had to check because most people don't ;)

jgromes, 2019-03-19:
To be honest, I didn't much testing at the range limits - it's a bit uncomfortable to write code out in the field. Perhaps you could try different bw/sf/cr settings, see if the problem is tied to a specific setting(s), or just the general snr level.

TomekRC, 2019-03-19:
You can apply the dummy load instead of the antenna :) In the evening I will do more tests and let you know about the results

TomekRC, 2019-03-19:
I'm after testing. With different power, SF, BW, coding rate, frequency, sync world and preamble length, the problem is the same. Of course, with higher SF for obvious reasons there are fewer errors.

jgromes, 2019-03-19:
I think I found the cause for the incorrect CRC behavior: https://github.com/jgromes/LoRaLib/blob/master/src/modules/SX1278.cpp#L432

**SOMEHOW** CRC was disabled entirely. I'm not sure when this got into the library, but it must have been there for a looooong time. I feel really dumb now :)

I tested it indoors with output power set to 1 (so the output goes into the unconnected pin), now I'm getting mostly CRC errors, occasionally a packet makes it with SNR at about -10 dB.

jgromes, 2019-03-19:
Should be fixed in ef8efbc9e300fd0d61eb882bd5d6a6df1000c4ad, please let me know if it resolves the issue.

TomekRC, 2019-03-19:
It's great that you found a bug !!! The error in the code can always happen. Unfortunately, but I stay until Friday outside the home :( As soon as I come back, I will check if the code works correctly on my boards.

TomekRC, 2019-03-22:
In version 8.0.1 crc works fine. The problem is still with the microcontroller hangs and reset.

jgromes, 2019-03-23:
I checked your receiver code with debugger, I don't think there are any memory leaks that could cause this. I also don't think that the reset has anything to do with signal quality/power - I got one with SNR around 4 dB (modules just a few cm away from each other). On the other hand, when I forced the `readData()` method to always return CRC error, no reset occured for about half an hour.

I'm going to create a separate issue for the reset, though I'm really not sure what the cause could be.

closed 2019-03-23 by jgromes