jgromes/LoRaLib #68: FSK modem never ends transmitting (SX1276 TTGO Lora32)

BigNerd95, 2019-03-21:
**Goal**
I'm trying to transmit with FSK modem using a TTGO Lora32  
The serial console prints "Packet transmitted successfully!"  

[image]

**Problem**
But using my SDR (HackRF One) i see this
[image]
It never ends transmitting (I'm sending a packet every 5 seconds)
Is this behavior right?

**To Reproduce**
```c
#include <LoRaLib.h>
#include <SPI.h>

#define SCK     5    
#define MISO    19   
#define MOSI    27   
#define SS      18   
#define RST     14   
#define DI0     26   
#define DI1     33   
#define DI2     32

SX1276 fsk = NULL;

void check(int state){
    if (state == ERR_NONE) {
      Serial.println(F("Success!"));
    } else {
      Serial.print(F("Failed, code "));
      Serial.println(state);
      while (true);
    }
}

void setup() {
  Serial.begin(115200);
  while (!Serial);

  Serial.println(F("Initializing ... "));
  
  pinMode(SS, OUTPUT);
  digitalWrite(SS, HIGH);

  SPI.begin(SCK, MISO, MOSI, SS);
  fsk = new LoRa(SS, DI0, DI1, SPI);
  
  check(fsk.beginFSK());
  check(fsk.setFrequency(868.0));
  
  Serial.println(F("All ok!"));

}

void loop() {

  // transmit FSK packet
  int state = fsk.transmit("Hello World!");
  
  if (state == ERR_NONE) {
    Serial.println(F("Packet transmitted successfully!"));
  } else if (state == ERR_PACKET_TOO_LONG) {
    Serial.println(F("Packet too long!"));
  } else if (state == ERR_TX_TIMEOUT) {
    Serial.println(F("Timed out while transmitting!"));
  } else {
    Serial.println(F("Failed to transmit packet, code "));
    Serial.println(state);
  }

  delay(5000);
}
```

Additional info:
 - TTGO Lora32 V2.0 [SX1276] (https://github.com/LilyGO/TTGO-LORA32-V2.0)
 - Arduino IDE version  1.8.8
 - Tested with LoRaLib present in Ardino IDE repo and downloaded from github

jgromes, 2019-03-21:
Looks like the radio never turns off the carrier - I suppose that could be an issue for other 868 MHz devices in the area. Could you add `fsk.standby()` after `fsk.transmit()`, see if the carrier goes away? If so, I will add that directly to the `transmit()` method, so that the module doesn't turn into a jammer.

BigNerd95, 2019-03-21:
So now it stops transmitting, but then it never transmits again
[image]

Changing the the delay to 500ms it not transmit every 1 or 2 seconds
[image]

Maybe it is slow to reactivate the module and sometimes it loose the chance to transmit?

BigNerd95, 2019-03-21:
It returns ERR_NONE
My board has NOT DIO1 and DIO2 wired [TTGO LoRa32 V2.0 Pinout v3.0.pdf](https://www.thethingsnetwork.org/forum/uploads/default/original/2X/4/4eb4d5846cf0a7fbaeabbad3c0b0cdde2faecb23.pdf)
But I connected them to pin 32 and 33 like they say here https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-3/18436

Extract:
[...]
#### TTGO LoRa32 V2:
```
ESP32          LoRa (SPI)      Display (I2C)  LED
-----------    ----------      -------------  ------------------
GPIO5  SCK     SCK
GPIO27 MOSI    MOSI
GPIO19 MISO    MISO
GPIO18 SS      NSS
EN     RST     RST
GPIO26         DIO0
GPIO33         DIO1 (see #1)
GPIO32         DIO2 (see #2)
GPIO22 SCL                     SCL
GPIO21 SDA                     SDA
GPIO22                                        useless (see #3)
```
1) Required (used by LMIC for LoRa).
Not on-board wired to any GPIO. Must be manually wired.
2) Optional (used by LMIC for FSK but not for LoRa).
Not on-board wired to any GPIO. When needed: must be manually wired.
3) GPIO22 is already used for SCL therefore LED cannot be used without conflicting with I2C and display.  

[...]

Maybe this is creating some problem?

jgromes, 2019-03-21:
I doubt that this is the issue - in `transmit()`, only DIO0 is used. DIO1 is only used in LoRa mode, to detect reception timeout and channel activity.

The only other cause I can think of is power, FSK transmission might be drawing more than LoRa. You can check with e.g. `setOutputPower(5)` to set the output power to 5 dBm.

BigNerd95, 2019-03-21:
Ok thank you!
So I just used setOutputPower(5) but nothing change (it is transmitting with less power, but not regularly).
About DIO1, do I need to "disable" it?
What about DIO2? `(used by LMIC for FSK but not for LoRa)`

jgromes, 2019-03-21:
No, DIO1 doesn't need to be disabled. I checked with DIO1 unconnected on my setup, still regular transmissions. DIO2 is unused.

Could you try sending LoRa packets, to check if the issue is only affecting FSK modem?

BigNerd95, 2019-03-21:
Just adapted Transmit example
It looks like is working correctly
[image]

BigNerd95, 2019-03-21:
I was trying to replay the transmitted data with my SDR, but the board rebooted
```
Initializing ... success!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... timeout!
Waiting for incoming transmission ... CORRUPT HEAP: Bad head at 0x3ffbdb44. Expected 0xabba1234 got 0x3ffbdb54
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x400d65cb on core 1

Backtrace: 0x40085494:0x3ffb1e20 0x400856c1:0x3ffb1e40 0x400d65cb:0x3ffb1e60 0x40089da9:0x3ffb1e90 0x40083bf6:0x3ffb1eb0 0x4008403d:0x3ffb1ed0 0x4000bec7:0x3ffb1ef0 0x400e87ad:0x3ffb1f10 0x400e8b35:0x3ffb1f30 0x400d27e1:0x3ffb1f50 0x400d0dde:0x3ffb1f80 0x400d4231:0x3ffb1fb0 0x400895b5:0x3ffb1fd0

Rebooting...
```

I successfully made the replay test with this library https://github.com/sandeepmistry/arduino-LoRa
But this library doesn't support FSK, this is the reason I was trying your lib.

jgromes, 2019-03-21:
Checked with ESP32 + standalone SX1278, still perfectly regular transmissions. I also wasn't able to reproduce the crash either, memory usage is stable.

The issue with transmitter not shutting down will be fixed in the release, thanks for pointing that out. Since I can't reproduce any of the other issues, I'm afraid I won't be much more help. I suspect an issue in the hardware, but your guess is as good as mine at this point.

BigNerd95, 2019-03-21:
Ok! Thank you very much for your support
I'll continue to investigate :+1:

jgromes, 2019-03-22:
Fixed in release [8.0.0](https://github.com/jgromes/LoRaLib/releases/tag/8.0.0), thanks for reporting.

Since I wasn't able to reproduce the other problems, I'm closing this issue now. Feel free to re-open if you find out something more, thanks.

closed 2019-03-22 by jgromes