jgromes/LoRaLib #69: Wrong SX127X_SYNC_VALUE_x

BigNerd95, 2019-03-22:
I'm scrupulously inspecting your library and I found this
https://github.com/jgromes/LoRaLib/blob/master/src/modules/SX127x.h#L363

```c
// SX127X_REG_SYNC_VALUE_1 - SX127X_REG_SYNC_VALUE_8
#define SX127X_SYNC_VALUE_1                           0x01        //  7     0     sync word: 1st byte (MSB)
#define SX127X_SYNC_VALUE_2                           0x01        //  7     0                2nd byte
#define SX127X_SYNC_VALUE_3                           0x01        //  7     0                3rd byte
#define SX127X_SYNC_VALUE_4                           0x01        //  7     0                4th byte
#define SX127X_SYNC_VALUE_5                           0x01        //  7     0                5th byte
#define SX127X_SYNC_VALUE_6                           0x01        //  7     0                6th byte
#define SX127X_SYNC_VALUE_7                           0x01        //  7     0                7th byte
#define SX127X_SYNC_VALUE_8                           0x01        //  7     0                8th byte (LSB)
```

I think they shoul be
```c
#define SX127X_SYNC_VALUE_1         0x28
#define SX127X_SYNC_VALUE_2         0x29
#define SX127X_SYNC_VALUE_3         0x2A
#define SX127X_SYNC_VALUE_4         0x2B
#define SX127X_SYNC_VALUE_5         0x2C
#define SX127X_SYNC_VALUE_6         0x2D
#define SX127X_SYNC_VALUE_7         0x2E
#define SX127X_SYNC_VALUE_8         0x2F
```

Or you have to use these instead https://github.com/jgromes/LoRaLib/blob/master/src/modules/SX127x.h#L199
```c
#define SX127X_REG_SYNC_VALUE_1                       0x28
#define SX127X_REG_SYNC_VALUE_2                       0x29
#define SX127X_REG_SYNC_VALUE_3                       0x2A
#define SX127X_REG_SYNC_VALUE_4                       0x2B
#define SX127X_REG_SYNC_VALUE_5                       0x2C
#define SX127X_REG_SYNC_VALUE_6                       0x2D
#define SX127X_REG_SYNC_VALUE_7                       0x2E
#define SX127X_REG_SYNC_VALUE_8                       0x2F
```
Because you are using it here https://github.com/jgromes/LoRaLib/blob/master/src/modules/SX127x.cpp#L919 as a register address
```c
  // set sync word
  _mod->SPIwriteRegisterBurst(SX127X_SYNC_VALUE_1, syncWord, len);
  return(ERR_NONE);
```

Tell me if I'm wrong

jgromes, 2019-03-22:
Well spotted, thanks. What's wrong is actually the SPI write, the address should be `SX127X_REG_SYNC_VALUE_1`.

Basically, `SX127X_REG_<something>` is the register address. The rest is register variables.

BigNerd95, 2019-03-22:
Ok perfect :+1:

jgromes, 2019-03-22:
Fixed in update 8.0.1, thanks for reporting.

I checked the rest of the library, hopefully there are no more similar typos left.

closed 2019-03-22 by jgromes