jgromes/LoRaLib #71: Pairing of SX1278 LoRa module

MaazSk, 2019-05-03:
**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/LoRaLib/wiki). You might find a solution to your issue there.

**Is your feature request related to a problem? Please describe.**
I have 3 pair of LoRa modules and I want that each pair should communicate with its respective Transmitter and Receiver. As they all are in the same range they interfere and I cannot differentiate the data so is there any way so that I can pair these LoRa Modules 

**Describe the solution you'd like**
I have used NRF24L01+ modules they have pipes, channels, data transmit through which we can pair the devices, is there Something similar we can do to pair these modules.

jgromes, 2019-05-03:
I can see at least 2 options here:
1. **Frequency channels** - have each of the pairs operate in its own frequency channel, e.g. channel 1 at 433.1 MHz, channel 2 at 433.5 MHz and channel 3 at 433.9 MHz. You have to make sure that the bandwidth is reasonably narrow, so that the channels don't intrefere.
2. **Sync words** - you can keep everything in the same frequency channel and have a unique LoRa sync word for each of the pairs. Receiver will not receive transmissions if the sync words don't match.

MaazSk, 2019-05-03:
Okk I will try in some time and will let you know.
Thanks and Regards.

MaazSk, 2019-05-03:
Can you suggest the frequency and bandwidth and SF for best range
Thanks and Regards.

jgromes, 2019-05-03:
Higher SF and lower bandwidth gets you higher range with lower data rate. Frequency only influences how well the signal travels through solid obstacles.

closed 2019-05-03 by jgromes

MaazSk, 2019-05-03:
> 2. **Sync words** - you can keep everything in the same frequency channel and have a unique LoRa sync word for each of the pairs. Receiver will not receive transmissions if the sync words don't match.

Can you give me a example or how can I set a sync word

Thanks and Regards

jgromes, 2019-05-03:
I suggest you take a look at [Settings ](https://github.com/jgromes/LoRaLib/blob/master/examples/Settings/Settings.ino) example or the[ API reference](https://jgromes.github.io/LoRaLib/class_s_x127x.html#ac5c7f4584352a12390594395d9c29bde).

MaazSk, 2019-05-03:
Thank you its working fine without any interference.

MaazSk, 2019-06-17:
Hi I was trying to pair again using sync word but now there is some interference I am still receiving data when the sync words are different.
I have used 0x12 and 0x24 as sync words

jgromes, 2019-06-17:
There seems to be some issues with sync word recognition on SX127x, see https://github.com/sandeepmistry/arduino-LoRa/issues/16

I think it might be down to using similar sync words. I suggest you try ones that don't have much in common. 0x24 is 0x12 shifted left by one bit.

MaazSk, 2019-06-18:
Hi, sync word is not used for pairing as I just posted the same issue on the semtech website and got the reply just sharing the link for the same below
[https://lora-developers.semtech.com/knowledge-base/forum/viewthread/48/](url)

jgromes, 2019-06-18:
I still believe this is a problem in the hardware - after all, if the sync word didn't match, further reception should be stopped.

The issue with using 0x12 and 0x24 as sync words is that one is a bit shift of the other:

0x12 = 0b0**001 0010**
0x24 = 0b**0010 010**0

The radio picks up the bits that are highlighted and that causes the sync word to match. Perhaps sync words where one is the inverse of the other would perform better? E.g.

0x12 = 0b0001 0010
0xED = 0b1110 1101

Then there should be nothing for the radio to pick up.

>Having a different Sync Word (...) does NOT prevent interference whatsoever.

What's the point of having it at all then ... Most other FSK radios allow to set maximum number of bit errors in sync word, but the LoRa modem does not (or at least it'¨s not documented in the datasheet).

If it's critical that only messages from certain devices are received, then you should use different frequency channels. Alternatively, receive all of them and employ some sort of address-based filtering system.