jgromes/LoRaLib #73: ERR_SPI_WRITE_FAILE

GusakIurii, 2019-05-18:
Good day.
I used the library LoRaLib version = 7.1.1.
Arduino IDE 1.8.4
Developer's board TTGO LoRa 433MHZ SX1278 MEGA328 (http://www.lilygo.cn/claprod_view.aspx?TypeId=21&Id=539)
Initialization with such parameters:
 lora.begin (434.0, 125.0, 12, 5, 0x12);
In my user program, when sending data (StateTransmit = lora.startTransmit (WriteByte, 2);) the message sometimes appears: "StateTransmit failed, code -16!", I understand this is an error of sending data on SPI (ERR_SPI_WRITE_FAILE). All SPI connections are ok.
But after this error, the program refuses to continue to work, and the next time it is sent, it constantly gives the same error Transmit failed, code -16 !.
Successful sending can be restored after resetting the Mega328p controller.
Can you please tell me how you can programmatically re-initialize SPI, the Lora SX1278 chip, can something else, without waiting for the controller to restart?

11: 02: 12.661> StartTransmit!
11: 02: 12.661>
11: 02: 12.725> address: 0x1
11: 02: 12.725> bits: 2 0
11: 02: 12.725> value: 0b11
11: 02: 12.725> current: 0b10001001
11: 02: 12.725> mask: 0b111
11: 02: 12.784> new: 0b10001011
11: 02: 12.784> read: 0b1111
11: 02: 12.784>
11: 02: 12.848> StartTransmit failed, code -16!

After updating LoRaLib library to version = 8.0.3
This error appears less frequently.
But at some point, the data simply stops being sent, returning the ERR_NONE code

jgromes, 2019-05-18:
Could you share your code? Nothing in the library should be causing communication to stop after one SPI write failure.

You can reset the LoRa module by pulling the RST pin low for more than 100 us, then releasing it and waiting for 5 ms. See SX1278 datasheet section 7.2.2.

GusakIurii, 2019-05-18:
`#include <LoRaLib.h>
#include "timer-api.h"
#include <EEPROM.h>
/*
// create instance of LoRa class using SX1278 module
// this pinout corresponds to RadioShield
// https://github.com/jgromes/RadioShield
// NSS pin:   10 (4 on ESP32/ESP8266 boards)
// DIO0 pin:  2
// DIO1 pin:  3
// IMPORTANT: because this example uses external interrupts,
//            DIO0 MUST be connected to Arduino pin 2 or 3.
//            DIO1 MAY be connected to any free pin 
//            or left floating.
*/
SX1278 lora = new LoRa;

byte WriteByte[] = {0, 0};
byte ReadByte[] = {0, 0, 0, 0, 0, 0};
char buf[80];
int  StateRecive;
int  StateTransmit;
int  StateInit;
bool SendOK = false;
bool ReciveOK = false;
bool Transmitt = false;
bool Recive = false;
bool StartRecive = false;
bool WakeUp = false;

//Активация автоматического режима
bool EnableAuto ;

unsigned long  Timeloop=0;

//Выключение включенных клапанов
bool OffValve1 = false;
bool OffValve2 = false;

byte TimerOffValve1 = 0;
byte TimerOffValve2 = 0;

byte NumValveOn1=1;
byte NumValveOn2=21;

byte LastValveOn1=0;
byte LastValveOn2=0;

byte CurrentValveOn1=0;
byte CurrentValveOn2=0;

bool FirtSend1 = false;
bool FirtSend2 = false;

//Номер клапана для опроса 

byte TimerRepeat = 0;
bool RepeatSend = false;
byte CountSend = 0;
String ReadStr = "";
int _timer1 = TIMER_DEFAULT;
 

// Флаг отображает что данные отправлены. (flag to indicate that a packet was received)
volatile bool transmittedFlag = true;
// Флаг отображает что встали в прием.
volatile bool startreceivedFlag = false;
// Флаг отображает что данные пришли. 
volatile bool receivedFlag = false;
volatile bool SendData = false;
[... 456 lines cut ...]
          if (EnableAuto){ 
            if (WriteByte[0]-48 >= 21) { 
              if (NumValveOn2 >= 40) { 
                NumValveOn2 = 21;
              } else ++NumValveOn2 ;
              
            } else { 
              if (NumValveOn1 >= 20) { 
                 NumValveOn1 = 1;
              } else ++NumValveOn1 ;
            }
          }
            
        }
    }
  
}

//Общая функция
void loop() {
  ChekInData();
  WriteData();
  ReadData(); 
}
`

GusakIurii, 2019-05-18:
[Send.zip](https://github.com/jgromes/LoRaLib/files/3194192/Send.zip)

jgromes, 2019-05-18:
On line 368, you have `while(1)` - that's an infinite loop, so the code is working exactly as it should: if `startTransmit()` method returns something other than `ERR_NONE`, the program will enter that infinite loop. Also, line 361 is in itself an infinite loop. Why is there a `while` instead of an `if`?

> The code is too large

I absolutely agree - why not split it into multiple files, or even just strucutre the functions better? Debugging this seems almost impossible.

closed 2019-05-18 by jgromes

GusakIurii, 2019-05-19:
While (1) I set for debugging that this happened.
I think I understood the reason for the failure of spi:
I have several slave devices and one central gateway.
When the slave devices send data to the gateway, and at the same time, the central one also wants to send the data back and it fails. Those. when the central gateway attempts to send data at the time of receiving the data, the SPI fails.
After that, either Laura's chip, or SPI hangs up, and does not respond to commands in any way. I tried to manage the pin directly where RST Lora is connected, but it did not help me:
      delay (10);
      digitalWrite (9, LOW);
      delay (10);
      digitalWrite (9, HIGH);
      delay (10);
      digitalWrite (9, LOW);
      delay (10);
      digitalWrite (9, HIGH);
      delay (10);
      lora.begin (434.0, 125.0, 12, 5, 0x12);
      delay (10);
      lora.startTransmit (WriteByte, 2);

No data was sent after this procedure.

jgromes, 2019-05-19:
I don't think this is an issue in the library, rather I believe it is somewhere in your code. If you can provide a minimal sketch to reproduce this issue, I will be able to investigate further.