jgromes/LoRaLib #74: No communication on TTGO LORA32

marcsza, 2019-05-29:
I have been trying to get the library working on my TTGO LORA32 V1. Everything was configured according to documentation and pin mapping.

Problem basically is: The sender works fine with no error, but the receiver never gets any packet.

Has anyone worked with the library on this dev board? Any suggestions? Attached are the codes both for sender and receiver.

`/* LORA SENDER */
//include the library
#include <LoRaLib.h>
#include <U8g2lib.h>

#define OLED_SDA 4
#define OLED_SCL 15
#define OLED_RST 16

U8G2_SSD1306_128X64_NONAME_F_SW_I2C Display(U8G2_R0, /* clock=*/ OLED_SCL, /* data=*/ OLED_SDA, /* reset=*/ OLED_RST); // Full framebuffer, SW I2C

SX1276 lora = new LoRa(18, 26, 33);

void setup() {
  Serial.begin(9600);
  Serial.print(F("Initializing ... "));
  int state = lora.begin(915.0, 7.8, 12, 8, 0x13);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  Display.begin();
  Display.enableUTF8Print();    // enable UTF8 support for the Arduino print() function
  Display.setFont(u8g2_font_ncenB10_tr);

}

void loop() {
  Serial.print(F("Sending packet ... "));

  int state = lora.transmit("Hello World!");

  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));

    // print measured data rate
    Serial.print(F("Datarate:\t"));
    Serial.print(lora.getDataRate());
    Serial.println(F(" bps"));

    Display.clearBuffer();
    Display.setCursor(0, 12); Display.print("SEND OK");
    Display.setCursor(0, 26); Display.print("DATA RATE");
    Display.setCursor(0, 40); Display.print(lora.getDataRate());
    Display.sendBuffer();

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F(" too long!"));
    Display.clearBuffer();
    Display.setCursor(0, 12); Display.print("TOO LONG");
    Display.sendBuffer();

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occurred while transmitting packet
    Serial.println(F(" timeout!"));
    Display.clearBuffer();
    Display.setCursor(0, 12); Display.print("TIMEOUT");
    Display.sendBuffer();

  }

  // wait a second before transmitting again
  delay(1000);
}`

`/* LORA RECEIVER */
// include the library
#include <LoRaLib.h>
#include <U8g2lib.h>

#define OLED_SDA 4
#define OLED_SCL 15
#define OLED_RST 16

U8G2_SSD1306_128X64_NONAME_F_SW_I2C Display(U8G2_R0, /* clock=*/ OLED_SCL, /* data=*/ OLED_SDA, /* reset=*/ OLED_RST); // Full framebuffer, SW I2C

SX1276 lora = new LoRa(18, 26, 33);

void setup() {
  Serial.begin(9600);
  Serial.print(F("Initializing ... "));
  int state = lora.begin(915.0, 7.8, 12, 8, 0x13);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  Display.begin();
  Display.enableUTF8Print();    // enable UTF8 support for the Arduino print() function
  Display.setFont(u8g2_font_ncenB10_tr);

}

void loop() {
  Serial.print(F("Waiting for incoming transmission ... "));

  // you can receive data as an Arduino String
  // NOTE: receive() is a blocking method!
  //       See example ReceiveInterrupt for details
  //       on non-blocking reception method.
  String str;
  int state = lora.receive(str);

  // you can also receive data as byte array
  /*
    size_t len = 8;
    byte byteArr[len];
    int state = lora.receive(byteArr, len);
  */

  if (state == ERR_NONE) {

    Display.clearBuffer();
    Display.setCursor(0, 12); Display.print("RECEIVED");
    Display.setCursor(0, 26); Display.print(str);
    Display.setCursor(0, 40); Display.print(lora.getRSSI());
    Display.sendBuffer();

    // packet was successfully received
    Serial.println(F("success!"));

    // print data of the packet
    Serial.print(F("Data:\t\t\t"));
    Serial.println(str);

    // print RSSI (Received Signal Strength Indicator)
    // of the last received packet
    Serial.print(F("RSSI:\t\t\t"));
    Serial.print(lora.getRSSI());
    Serial.println(F(" dBm"));

    // print SNR (Signal-to-Noise Ratio)
    // of the last received packet
    Serial.print(F("SNR:\t\t\t"));
    Serial.print(lora.getSNR());
    Serial.println(F(" dB"));

    // print frequency error
    // of the last received packet
    Serial.print(F("Frequency error:\t"));
    Serial.print(lora.getFrequencyError());
    Serial.println(F(" Hz"));

  } else if (state == ERR_RX_TIMEOUT) {
    // timeout occurred while waiting for a packet
    Serial.println(F("timeout!"));
    Display.clearBuffer();
    Display.setCursor(0, 12); Display.print("TIMEOUT");
    Display.sendBuffer();

  } else if (state == ERR_CRC_MISMATCH) {
    // packet was received, but is malformed
    Serial.println(F("CRC error!"));
    Display.clearBuffer();
    Display.setCursor(0, 12); Display.print("MISMATCH");
    Display.sendBuffer();

  }

}`

jgromes, 2019-05-30:
What's the pinout of your board? If I recall correctly, DIO1 is not connected on TTGO LORA32.

marcsza, 2019-06-03:
Sorry for the spaghetti code. Wrong usage of the library from myself.

closed 2019-06-03 by marcsza