jgromes/LoRaLib #75: SX1278 with UNO ERR_SPI_WRITE_FAILED -16

mark9599, 2019-06-07:
I've verified the SPI Connections over and over and yet the error persists
Can you provide a solution?
I'm using Arduino UNO with the SX1278

jgromes, 2019-06-07:
I can't provide a solution, since you haven't provided much information.

1. Could you share your code?
2. How often is this happenning?
3. Are there any other SPI devices on the same bus the SX1278 is using?
4. How long are the wires from Arduino to SX1278?
5. What are you using to power the SX1278 and the Arduino?
6. Have you tried running the code in debug mode? Go to TypeDef.h, and uncomment the line #define RADIOLIB_DEBUG. Upload the sketch and there will be extra debug information in the terminal.

mark9599, 2019-06-07:
Code:-

#include <LoRaLib.h>

// NSS pin:   10 (4 on ESP32/ESP8266 boards)
// DIO0 pin:  2
// DIO1 pin:  3
SX1278 lora = new LoRa;

void setup() {
  Serial.begin(9600);
  pinMode(7,OUTPUT);
      digitalWrite(7,HIGH);
  // initialize SX1278 with default settings
  Serial.print(F("Initializing ... "));
  // carrier frequency:           434.0 MHz
  // bandwidth:                   125.0 kHz
  // spreading factor:            9
  // coding rate:                 7
  // sync word:                   0x12
  // output power:                17 dBm
  // current limit:               100 mA
  // preamble length:             8 symbols
  // amplifier gain:              0 (automatic gain control)
  int state = lora.begin();
  if(state == ERR_NONE) {
    Serial.println(F("success!"));
      digitalWrite(7,LOW);
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }
}

void loop() {
  Serial.print(F("Scanning channel for LoRa preamble ... "));

  // start scanning current channel
  int state = lora.scanChannel();

  if(state == PREAMBLE_DETECTED) {
    // LoRa preamble was detected
    Serial.println(F(" detected preamble!"));

  } else if(state == CHANNEL_FREE) {
    // no preamble was detected, channel is free
    Serial.println(F(" channel is free!"));

  }

  // wait 100 ms before new scan
  delay(100);
}

2. It is happening every time.

3. No, only SX1278 is connected 
SCLK->13
MISO->12
MOSI->11
NSS->10
REST->9
DIO0->2
DIO1->3

4. Not very long, the module to Arduino directly. Around 10cms

5. Power using USB of Laptop
I've tried External DC Power Supply also, used LED as an indicator still doesn't work.
SX1278 powered from 3.3V from Arduino

6. 
[image]

jgromes, 2019-06-07:
It looks like the library was unable to wake up the SX1278 from sleep mode. I see you have conneted the SX1278 reset pin, could you disconnect that?

mark9599, 2019-06-10:
[image]

I've left the REST pin floating (Not connected) still the error persists.

jgromes, 2019-06-10:
In that case, could you try a different SX1278 module? Your current one seems to be unable to wake up from sleep mode, so it could be damaged.

Forgot to ask about this: are you using level shifter between Arduino and SX1278?

mark9599, 2019-06-10:
I'm not using a Level Shifter.
I've tried with another LoRa module and yes it works. (without level shifter)
However is it necessary to use a level shifter?

And yeah I suppose this Module is damaged.

jgromes, 2019-06-10:
Then I highly suggest you use one. SX1278 is a 3.3V device, Arduino is 5V. When you connect the two directly, you are applying 5V to a 3.3V logic device. It will work for a while, but it will most likely break eventually - which is probably what happened to the first module.

Arduino input pins will handle the 3.3V signal from SX1278, so you only have to worry about the 5V signals from Arduino to SX1278: SCK, MOSI and NSS. Any 4 channel level shifter should do the job.

Closed as invalid (unrelated to library code).

closed 2019-06-10 by jgromes