jgromes/LoRaLib #76: Receive(str) in LoRa example crashes with Heap corruption exception

kiseitai3, 2019-06-08:
**IMPORTANT: Check the wiki**
https://jgromes.github.io/LoRaLib/class_physical_layer.html

**Describe the bug**
Default Receive LoRa example crash with Heap corruption exception. It's the same problem seen in issue #51 .

**To Reproduce**
Uploading vanilla LoRa receive example using an Arduino string will yield a heap corruption because the example does not contain an initialized Arduino String. The call to receive(str) is actually receiv(str, 0) due to len being deafulted to 0 if omitted. receive then calls binary receive with a pointer to temporary array of size 1 which is written to by a call to readData which calls _mod->SPIreadRegisterBurst(data, len), where len = 1. Up to this step we are still guarded from heap corruption, but the returned data overwrites what would be the null terminating character spot. When we do the following:
```
// if packet was received successfully, copy data into String
  if(state == ERR_NONE) {
    str = String(data);
  }
```
Maybe the String constructor tries to read beyond 1 character expecting to find the nul-character but hitting some protected memory area? Even to me, it doesn't make complete sense under the assumption that MCUs do not have firmwares with advanced memory management (unless we implement it at the expense of performance). I will test one of my solutions to see if my hypothesis is the case. 

Solution A: we set data[len] = 0 before creating the Arduino String. This agrees with the current architecture of your library but presents misleading behavior in the examples and your discussion in issue #51 . 

Solution B: we implement the behavior expressed in the examples and allow Arduino Strings to be the way we capture a variable length packet in a safe manner. I am a fan of this approach for the reasons outlined in my side note. In other words, we would have to turn receive(String&,size_t) into a pure virtual method that has to be implemented in SX127x where we can query the modem for the actual packet length and create the temporary buffer then.

**Facts so far***
receive called with a string as an argument crashes with a heap exception. receive called with a c-string and explicit buffer size does not crash and captures the contents of the packet correctly.

**Additional info (please complete):**
 - MCU: Heltec ESP32 LoRa Wifi Kit v2
 - LoRa module type SX1276
 - Arduino IDE version  1.8.8
 - Library version 8.0.6

**Side Note**
I have the test hardware now (2 ESP32s equipped with SX1276s). I will resume work on pull request #66 . Hopefully, I can get large packet and all three packet formats working on this driver. I still have to get the FSK modem example working on my boards. They give me the timeout error when receiving and transmitting (with only one success right before getting into a timeout loop in one of my many tests). I don't know if I should create a separate issue ticket for that until I know better what is happening and how I can help you with a pull request if it comes to that. Let me know if you are ok with Solution B. I will submit a pull request with a patch as soons as you let me know which route you want to take. :)

jgromes, 2019-06-09:
I was unable to reproduce this on my ESP32. Previously, this was caused by incorrect dynamic memory allocation size in `SX127x::readData()`. The null terminator is added right after reading data from SX1278. If anything, I would expect the binary version to crash, since that adds the null byte as well.

I see no reason why the temporary array in `PhysicalLayer::receive()` should be a problem - if len == 0, that array will be deleted and a new one will be created in `SX127x::readData()`. Array of length one is a perfectly valid C/C++ structure (as well as array of length zero).

kiseitai3, 2019-06-09:
You are right! I did not see this line in readData() in my initial investigation.
```
data[length] = 0;
```
That said, I found two areas of interest and the reason why the allocation was erroing in the past.
Hypothesis1: The wrong allocation in the past was due to allocating an array of size 0, which although is legal syntax, it is not correct because the returned pointer should be NULL. This NULL pointer was used to initialize String, which yields the exception upon usage similarly to templated std::strings. The following piece in readData never returned the correct temporary pointer. It created a local memory leak instead.
```
 // read packet data
  if(len == 0) {
    // argument len equal to zero indicates String call, which means dynamically allocated data array
    // dispose of the original and create a new one
    delete[] data;
    data = new uint8_t[length + 1];
  }
```
data in this context is a local variable in the stack that contains a copy of the address to data created in the physical layer. 
Hypothesis 2: PhysicalLayer's data gets deleted while it is a NULL pointer. This should definitely error out before allocating the string. It's the reason I always had to check for null pointers before deallocating them in my game engine and in my thesis.

Now, you sort of patched that by allocating an array of size 1 and passing to binary receive(data, len), where len = 0. receive calls readData with data and len = 0. So I think we are still trying to delete a null pointer. I don't know why no one else has had this issue, but my gut feeling is that I will be able to patch this issue once and for all with a couple of modifications. I will create a pull request once that's ready and you can give your final judgement.

Another note: it seems that readData attempts to query the modem for a packet length (I did not notice my suggestion in the previous post was implemented). However, it ignores len and uses length, which is correct iff len == length. The problem is that it risks overwriting memory not part of data if length > len. Thus, it's a buffer overflow. I think most people using binary receive never force it into this case scenarion, because we usually try to allocate a buffer bigger than the expected worst case packet size precisely to avoid situations like this.

jgromes, 2019-06-09:
I'm still confused as to why I can't reproduce this issue. Perhaps there's a difference in the ESP toolchain (e.g. different versions)?

If you have a more robust solution, that would be great. Please keep in mind though that in order to merge the PR, all changes have to compatible with [RadioLib](https://github.com/jgromes/RadioLib).

kiseitai3, 2019-06-09:
Yeah, I thought it might be a toolchain difference. Maybe, even though I pull the toolchain from the standard expressif link, each ESP32 board has some subtle built-in differences in configuration for the compiler. Maybe, your compiler does allocate a buffer of at least +1 byte in size when calling new uint8_t[len] in an area that does not collide with other memory regions and mine outright returns NULL as its memory management strategy? 

Hmm, I have an idea compatible with your driver architecture and will remove the memory leak and potential buffer overflows, but I will have to create a new error flag to alert users who rely on a direct call to binary receive that they are providing an empty c-string and we will not proceed with that. Give me about 3 hours. :)

By the way, I was wrong about the pointer being null but somehow it is detecting it is not valid
```
// read packet data
  if(len == 0) {
    // argument len equal to zero indicates String call, which means dynamically allocated data array
    // dispose of the original and create a new one
    if(!data)//not safe to delete memory when we 'know' the pointer is invalid, even if non zero. Added this for testing
      delete[] data;
    Serial.println("Avoided heap corruption!");
    Serial.println("data pointer: ");
    char* num = new char[sizeof(size_t) * 8 + 1];
    itoa(size_t(&data), num, 16);
    Serial.print(num);
    Serial.println();
    Serial.println("len = ");
    itoa(len, num, 10);
    Serial.print(num);
    Serial.println();
    Serial.println("length = ");
    itoa(length, num, 10);
    Serial.print(num); 
    Serial.println();   
    delete[] num;
    data = new uint8_t[length + 1];
  }
```
avoided the heap corruption and yielded the following
```
Initializing ... Found SX127x!
Symbol length: 0.01 ms
Symbol length: 32.77 ms
success!
Waiting for incoming transmission ... Avoided heap corruption!
data pointer: 
3ffb1efc
len = 
0
length = 
12
success!
Data:			⸮xV⸮⸮⸮⸮⸮
RSSI:			-28 dBm
SNR:			14.25 dB
Frequency error:	9216.20 Hz
Waiting for incoming transmission ... Avoided heap corruption!
data pointer: 
3ffb1efc
len = 
0
length = 
12
success!
Data:			⸮xV⸮⸮⸮⸮⸮
RSSI:			-28 dBm
SNR:			13.25 dB
Frequency error:	9214.10 Hz
```
I don't know what str is pulling but it does not look like overflowed memory... It might be the memory leak I mentioned earlier... Nonetheless, it proves that delete[] data; was the culprit. :)

kiseitai3, 2019-06-09:
I see this snippet in the PhysicalLayer
```
int16_t PhysicalLayer::readData(String& str, size_t len) {
  // create temporary array to store received data
  char* data = new char[len + 1];
  int16_t state = readData((uint8_t*)data, len);

  // if packet was received successfully, copy data into String
  if(state == ERR_NONE) {
    str = String(data);
  }

  delete[] data;
  return(state);
}

int16_t PhysicalLayer::receive(String& str, size_t len) {
  // create temporary array to store received data
  char* data = new char[len + 1];
  int16_t state = receive((uint8_t*)data, len);

  // if packet was received successfully, copy data into String
  if(state == ERR_NONE) {
    str = String(data);
  }

  delete[] data;
  return(state);
}
```
My changes will necessitate receive(String& str, size_t len) be turned pure virtual. My implemented receive will make readData(String&, size_t) unnecessary and obsolete since I will be calling readData(uint8_t*, size_t) instead. Is it being used anywhere at all? I will have to copy the changes to receive over to any other base class that inherits PhysicalLayer and is not 127x.

kiseitai3, 2019-06-09:
I checked the class hierarchy, so I am safe to make changes in SX127x.

jgromes, 2019-06-09:
You can't get rid of `readData()`, it's used in interrupt-driven receive. That's also the reason why `receive()` calls `readData()` at the end. `receive` is basically just `listen()` -> wait for packet or timeout -> `readData()`.

Also, I would prefere to keep `receive()`/`readData()` implementations at the `PhysicalLayer` level. The reason is that these methods don't need any information about what type of module is currently using them.

kiseitai3, 2019-06-09:
Hmm, the memory issue can only be fixed at the last method with a direct reference to str. The only way I could keep the memory management portion in the physical layer is if readData(uint8_t*, size_t) returns the new temporary buffer pointer instead of state, which conflicts with the general format of the library which is something I like from your work. The current versions of readData(String&) and receive(String&) are broken logic-wise. I agree they don't need to know the specifics of a module, but they have to be situated at the base of a module family. What I can do is make them regular virtual, keep my re-implementations for the SX/ HopeRF chips, and let you decide if RadioLib needs to maintain access to these methods as is instead of reimplementing on a per chip family scenario. This way, the SX/HopeRF chips would be patched and protected from the three current memory management issues. Does this sound like a good compromise? That said, I strongly recommend that they are turned pure virtual to force other radio families to make the decision of how they can process the receive steps. It would also agree with how the other methods in the PhysicalLayer are pure virtuals. 

The issue I am facing is that querying the modem for packet length is modem family specific. Without it, I cannot ask binary readData to write the correct amount of data into our buffer. The reason is I would be passing an invalid pointer. Yes, we can "guess" the size by pre-allocating a 255 byte buffer now. However, in the future, I will have to solve the same issue when I attempt to implement the other two packet formats the SX/HopeRF chips are supposed to support. We can only get the packet length after/while receiving and before reading.

jgromes, 2019-06-09:
Would it help to have a module-specific method that reads the number of received bytes? You could then call that from `PhysicalLayer` before call to `readData` to get the required buffer size.

The reason it's written this way is that it works fine on 8-bit AVR devices with no memory protection. It wasn't intended to be used on platforms like ESP, the fact that it (sort of) works is more or less a lucky coincidence.

kiseitai3, 2019-06-09:
Ah, I see. 

We can call module specific methods from the base class? Even if it worked, it breaks the inheritance hierarchy. PhysicalLayer->FamilyModule->Module. That said, because of what you just mentioned, the best compromise is to have these two methods virtual (instead of pure virtual). This way, any usage of PhysicalLayer elsewhere in RadioLib will only see the PhysicalLayer implementation and anyone using the SX127x modules will see the patched version.

kiseitai3, 2019-06-09:
I looked at https://jgromes.github.io/RadioLib/hierarchy.html , so this should be the strategy for now. I don't have the other radio chips for testing, otherwise I would have patched the other modules as well. :(

I am cleaning some syntax errors and I will be testing on my units. Hopefully, the pull request will pass Travis and won't break anything else.

jgromes, 2019-06-09:
The other modules in RadioLib use receive methods that are very similar to what's in SX127x.

What I meant was to add the getPacketLength method at the level of FamilyModule. I think we should be able to call that from PhysicalLayer, in the same way PhysicalLayer calls module-specific binary receive and transmit.

kiseitai3, 2019-06-09:
Ah, actually, yeah, that's bloody smart. I happen to have moved the modem query for packet size portion into its own method earlier today. So yeah, that's no problem at all!

kiseitai3, 2019-06-09:
Hmm, I have to look back at the datasheet.
```
else if(modem == SX127X_FSK_OOK) {
    // get packet length
    length = _mod->SPIreadRegister(SX127X_REG_FIFO);
  }
```
Implies that the first piece in the FIFO for FSK is the packet length. But we cannot call getPacketLength 2x or we lose some data. Binary receive would not know the difference and startReceive is called inside receive which comes after physicalLayer receive where I plan to call getPacketLength for the Arduino String variety. I'm going to take a break to think about I would do that without breaking current usage of the driver.

kiseitai3, 2019-06-09:
Actually, maybe an internal flag will suffice. I will still take a break.

kiseitai3, 2019-06-09:
Ok, it's working now. I hit issue #51 and #30 but that case was fixed with a simple +1 to buffer length. I should be creating a pull request in 10 minutes.

kiseitai3, 2019-06-09:
```
Initializing ... success!
Waiting for incoming transmission ... success!
Data:			Hello World!
RSSI:			-25 dBm
SNR:			12.75 dB
Frequency error:	8763.74 Hz
Waiting for incoming transmission ... success!
Data:			Hello World!
RSSI:			-25 dBm
SNR:			12.75 dB
Frequency error:	8762.16 Hz
```

kiseitai3, 2019-06-10:
One side question: if the modem returns timeout really fast with some frequent Successfully transmitted packet (even though the receiver does not print anything related to the packet in the terminal) messages, what should I look into to start troubleshooting? This is happening with the modem in FSK with and without OOK modulation using the FSK example. Does this warrant a separate issue in which I should document my troubleshooting findings or is this usually fixed with some simple procedure?

I gotta say, this is fun! XD

jgromes, 2019-06-10:
>Does this warrant a separate issue

I'm not sure. The issue with FSK is that the data rate is up to a few hundred kbps. The timeout duration is calculated as 500% of ToA for 255 B packet at the configured data rate. I would suggest to use intrrupt-driven receive for this. Blocking methods don't really make sense here.

kiseitai3, 2019-06-10:
Interesting. One of the folks at the OpenAPS project said the opposite some months back, but I am inclined to believe you are right. I think his rfm95 driver was indefinitely blocking in the while loop until transmission ended. I will test your suggestion as well. Thanks!

jgromes, 2019-06-10:
Under normal circumstances, `receive()` should be guaranteed to exit at some point - especially the FSK variant, since the timeout of that is fully controlled by MCU. LoRa timeout is controlled by the radio, so in theory, that *could* end up in an infinite loop. Not FSK though.

jgromes, 2019-06-12:
Conversation moved to PR #77

jgromes, 2019-06-13:
Hang on - how do we know something was already received here? https://github.com/jgromes/LoRaLib/blob/master/src/protocols/PhysicalLayer.cpp#L57
That isn't true, things will only be received when `receive()` is running.

jgromes, 2019-06-13:
Also, what happens when data in `PhysicalLayer::receive(String&, size_t)` is null? We then still add the null terminator and attempt to initialize String.

jgromes, 2019-06-13:
`readData()` works fine, because that will only be executed after packet was received. We can't make the same assumption for `receive()`, so the only way I can see around this issue is to set the temporary buffer to 257 bytes long (LoRa FIFO size + 1 byte for `\0`), then attempt to receive a packet, if one arrives read the length, initialize String and free the temporary buffer.

Please correct me if I'm wrong.

jgromes, 2019-06-13:
I pushed the fix into https://github.com/jgromes/LoRaLib/tree/memory-test, could you test it with your setup? Thanks.

kiseitai3, 2019-06-14:
Hmm, good catch! The only problem is that one of the packet formats I will work on implementing from the datasheet would allow you to send packets bigger than 255 (at least hypothetically). When I implement the rest of the FIFO handling, this assumption will not hold either (ok, maybe in practice it will because the packets from insulin pumps were expected to be no more than 107 bytes).

Give me a couple of days to rethink this portion. One idea I have is to uncouple the packet reception portion from the packet read call. That way, we can get the packet, call readData(String&, size_t), and let readData handle the actual string initialization. I have to explore how to do this without breaking the library's API or increasing your work on the RadioLib side.

kiseitai3, 2019-06-14:
Based on the datasheet, your approach is the simplest solution for now. I will need to rethink this portion over the weekend.

I will test your patch tonight.

jgromes, 2019-06-14:
> one of the packet formats [...] would allow you to send packets bigger than 255

Looking at the datasheet, the way that works is when payload length exceeds 64 bytes, you have to write/read FIFO during tx/rx. That approach won't be compatible with current `receive()` implementation anyway, since that waits to receive the whole packet, and only then reads.

I think that you don't have to worry about current `SX127x::receive()` implementation, since you will most likely have to write a completely new method(s) to handle the variable packets. We can then decide which one to call at the `PhysicalLayer` level.

Alternatively, if the goal is 107 bytes, how about SX126x series? These have FSK modem and 256 byte FIFO.

kiseitai3, 2019-06-14:
You are probably right. In terms of chip type, the original work was done with an rfm95. My goal was to find or expand a driver to accomodate as many radio chips as possible. That way, current and future patients have options in pricing, capabilities, and frequency band support. Basically, I want the user to have the freedom to make use of the project with minimal hardware constraints. Different medical devices use different communication standards, so some folks have worked on other chip families. Currently, the project uses the rfm95 driver from the person who originally reversed engineered one of the insulin pumps. I wanted a driver that abstracts away the whole chip family while still being flexible enough for other projects. 

I need to do one test later today. I will let you know if we are good to go.

kiseitai3, 2019-06-15:
Seems like it is running well!
```
21:21:29.835 -> Waiting for incoming transmission ... success!
21:21:30.200 -> Data:			Hello World!
21:21:30.200 -> RSSI:			-40 dBm
21:21:30.233 -> SNR:			8.50 dB
21:21:30.233 -> Frequency error:	9074.38 Hz
21:21:30.266 -> Waiting for incoming transmission ... timeout!
21:21:30.598 -> Waiting for incoming transmission ... timeout!
21:21:31.029 -> Waiting for incoming transmission ... success!
21:21:31.361 -> Data:			Hello World!
21:21:31.394 -> RSSI:			-40 dBm
21:21:31.394 -> SNR:			8.75 dB
21:21:31.427 -> Frequency error:	9074.38 Hz
21:21:31.460 -> Waiting for incoming transmission ... timeout!
21:21:31.759 -> Waiting for incoming transmission ... timeout!
21:21:32.190 -> Waiting for incoming transmission ... success!
21:21:32.522 -> Data:			Hello World!
21:21:32.555 -> RSSI:			-40 dBm
21:21:32.555 -> SNR:			8.75 dB
21:21:32.588 -> Frequency error:	9074.38 Hz
21:21:32.621 -> Waiting for incoming transmission ... timeout!
```

jgromes, 2019-06-15:
Merged memory-test, thanks for testing it.

closed 2019-06-15 by jgromes