jgromes/LoRaLib #80: scanChannel() function does not work

Y-Fakih, 2019-06-17:
Hi 
I am using your Library "LoRaLib" in my project. I am working on Dragino LoRa Shield Version 1.4 that contains the SX1276. In fact using the scanChannel function we can detect a lora preamble, but i cannot detect the LoRa preamble. I go in depth in the code

I see in the file SX127X.cpp the body of the scanChannel function and i think there is something wrong concerning the interrupts zero and one : 

"
  while(!digitalRead(_mod->getInt0())) {
	Serial.println("getInt0 checked");
    if(digitalRead(_mod->getInt1())) {
	  Serial.println("getInt1 checked");
      clearIRQFlags();
      return(PREAMBLE_DETECTED);
    }
  }
"

I think the condition must be inverted because in the first condition, 
we have getInT0 which is associated to pin D2 with respect to Dragino LoRa Shield pin mapping GPIO0; 

and 

the getInt1hich is associated to pin D6 with respect to Dragino LoRa Shield pin mapping GPIO1;

I have modified in the module.h the following line to have correspondance " Module(int cs = LORALIB_DEFAULT_SPI_CS, int int0 = 2, int int1 = 6, SPIClass& spi = SPI);"

but still after the configuration, i cannot detect a lora preamble because of the 2 conditions above ?

Any type of Help ?
is there something missing in the configuration ?
is the Logical sequence of interrupts true ?
Thank you in advance

jgromes, 2019-06-17:
I'm fairly certain the sequence is correct, since myself and others have used CAD without any issues or modifications. Why should it be inverted?

You don't have to change anything in Module.h, just provide the pins to the LoRa contructor. See the [Basics Wiki page](https://github.com/jgromes/LoRaLib/wiki/Basics).
```c++
SX1276 lora = new LoRa(10, 2, 6);
```

What are the modulation properties of the LoRa signal you're trying to detect? If it has low enough spreading factor and/or short enough preamble, you will have to run scanChannel() very often to detect it.

Also, don't put any Serial input into the CAD loop. Serial methods like print() take a long time to execute, preventing you from actually detecting anything.

Y-Fakih, 2019-06-17:
Dear Mr. Jan Gromes
The modulation properties of the LoRa signal I am trying to detect is :
- Bandwidth 125 KHz
- setSpreadingFactor(7)
- setCodingRate(5)
- setFrequency(868.1)
- setSyncWord(0x34)
- setPreambleLength(8)

I think it should be inverted because in the SX1276 documentation the flowchart 
[image]

The radio receiver captures LoRa preamble symbol of data from the channel. The modem digital processing starts. The modem searches for a correlation between the radio captured samples and the ideal preamble waveform. This correlation process takes a little bit less than a symbol period to perform. Once the calculation is finished the modem generates the CadDone interrupt. If the correlation was successful, CadDetected is generated simultaneously. The chip goes back to Standby mode. If a preamble was detected, clear the interrupt. 
the documentation of SX1276 can be found on the following link "https://www.mouser.com/ds/2/761/sx1276-1278113.pdf"

`int16_t SX127x::scanChannel() {
  // check active modem
  if(getActiveModem() != SX127X_LORA) {
    return(ERR_WRONG_MODEM);
  }
  // set mode to standby
  int16_t state = setMode(SX127X_STANDBY);
  // set DIO pin mapping
  state |= _mod->SPIsetRegValue(SX127X_REG_DIO_MAPPING_1, SX127X_DIO0_CAD_DONE | SX127X_DIO1_CAD_DETECTED, 7, 4);
  // clear interrupt flags
  clearIRQFlags();
  // set mode to CAD
  state |= setMode(SX127X_CAD);
  if(state != ERR_NONE) {
    return(state);
  }
  // wait for channel activity detected or timeout
  while(!digitalRead(_mod->getInt0())) {
    if(digitalRead(_mod->getInt1())) {
      clearIRQFlags();
      return(PREAMBLE_DETECTED);
    }
  }
  // clear interrupt flags
  clearIRQFlags();
  return(CHANNEL_FREE);
}`

if we compare the code and the flowchart we can conclude that the CadDone interrupt corresponds to InT0 and the CADDetected interrupt correspond to Int1. These two interrupts must be on if a preambul is detected ; it is not the case ??

Any Type of help ?
Thank you in Advance

jgromes, 2019-06-17:
> we can conclude that the CadDone interrupt corresponds to InT0 and the CADDetected interrupt correspond to Int1

Precisely. Let me rewrite the loop for you in "pseudo" code:

```c++
while(not cadDone) {
  if(cadDetected) {
    return(PREAMBLE_DETECTED);
  }
}

return(CHANNEL_FREE);
```

CadDone interrupt is generated when channel activity detection is finished, regardless of the outcome. CadDetected is generated only when LoRa preamble is detected.

Try adding more symbols to the preamble, increasing spreading factor, or running scanChannel() more often. The ChannelActivityDetection with default settings on both transmitter and receiver does detect LoRa packets, albeit with less than 100% success rate - most likely due to the 100ms delay between each call to scanChannel().

Y-Fakih, 2019-06-19:
Dear Mr. Jan Gromes,
First of all, thank you for clarifying the pseudocode.

I have tried the code it works but i cannot see anything on the oscilloscope because it is really very very fast. so I am trying to do it manually by writing something similar to the ScanChannel Function without using it. But I cannot access the IRQ Flag from the arduino directly ?

Is there a function to clear the IRQFlag directly ?
Any type of help ?
Thank You in advance;

jgromes, 2019-06-19:
If it works, why do you need to reimplement scanChannel? Can you post your code?

Low-level methods like clearIrq() are unavailable to the top-level API by design - there's no need to give that sort of access to the user. If you find yourself at situation where you need to use the low level methods from Arduino sketch, it might be a sign you're not using the library correctly.

On my scope, both CadDone and CadDetected are about 60 us - I wouldn't call that very fast, not even for 16 MHz Arduino.

Y-Fakih, 2019-06-19:
ok here you will find my code:

`#include <LoRaLib.h>
#define Freq 868.1
#define Bw 125.0
#define SF 7
#define CR 5
#define SyncWord 0x34
#define Power 17
#define Current 80
#define PreambLength 8
#define Gain 1

SX1276 lora = new LoRa (10,2,6);
void setup() {
  Serial.begin(9600);
  int state = lora.begin(Freq, Bw, SF, CR, SyncWord, Power, Current, PreambLength, Gain);
  if (state == ERR_NONE) {
    Serial.println("Successful Beginning");
  } else {
    Serial.print("Failed");
    while (true);
  }
  if (lora.setCRC(false) != ERR_NONE) {
    Serial.println("Unable to set CRC");
    while (true);
  }
}

void loop() {
  int state = lora.scanChannel();
  if (state == PREAMBLE_DETECTED) {
    state = lora.transmit("The Game is Amazing; The Game is Amazing; The Game is Amazing;");
    } 
  } `

I connet 
- PIN2 (CadDone Int0) on Channel 1 on oscilloscope
- PIN6 (CADDetected Int1) on channel 2 on oscilloscope
[image]

and the result is in the picture

jgromes, 2019-06-19:
That looks normal to me. What exactly did you expect to happen?

Y-Fakih, 2019-06-19:
With respect to the edit you're right. 

Channel 1 - CadDetected
Channel 2 - CadDone

Thank you for your time and your advice;

closed 2019-06-19 by jgromes