jgromes/LoRaLib #81: RFM98 error code -2

MaazSk, 2019-06-18:
**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/LoRaLib/wiki) and the [API documentation](https://jgromes.github.io/LoRaLib/). You might find a solution to your issue there. 

**Describe the bug**
Interfaced RFM98 with arduino getting failed code -2 checked SPI connection everything is fine but still the same issue changed RFM98 also but getting the same 

**To Reproduce**
`
#include <LoRaLib.h>

RFM98 lora = new LoRa;

void setup() {
  Serial.begin(9600);

  
  Serial.print(F("Initializing ... "));
  
  int state = lora.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  
  lora.setDio0Action(setFlag);

  // start transmitting the first packet
  Serial.print(F("Sending first packet ... "));

 
  state = lora.startTransmit("Hello World!");

 
  if (state != ERR_NONE) {
    Serial.print(F("failed, code "));
    Serial.println(state);
  }
}

volatile bool transmittedFlag = false;

void setFlag(void) {
 
  transmittedFlag = true;
}

void loop() {
  
  if(transmittedFlag) {
    Serial.println(F("transmission finished!"));

   
    delay(1000);

    
    Serial.print(F("Sending another packet ... "));

    int state = lora.startTransmit("Hello World!");
  
    
    if (state != ERR_NONE) {
      Serial.print(F("failed, code "));
      Serial.println(state);
    }
  }

}`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional info (please complete):**
 - MCU: Arduino Uno
 - LoRa module type RFM98
 - Arduino IDE version  1.8.9

closed 2019-06-18 by MaazSk

jgromes, 2019-06-18:
If you managed to find a solution to the problem, could you please post it here? In the future, it might help others facing a similar issue.

MaazSk, 2019-06-19:
I believe that my RFM98 is not working because the code is working with another pair of RFM98 as I have 4 of them, unfortunately 3 are not working and only 1 is working. Issue was in the hardware.

jgromes, 2019-06-19:
Just to check - you are using some logic level shifter between Uno and the LoRa module, right?

MaazSk, 2019-06-19:
No I am not using any Logic level shifter in between but now 2 of them are working I think those jumper wire created a mess and the arduino uno board also just now changed those and 2 are working will just check the rest of them and will let you know soon.
Thank You!

jgromes, 2019-06-19:
Then I suggest you use one, the inputs on RFM98 are not 5V-tolerant. A simple 4 channel logic shifter should do, for MOSI, SCK and NSS.

MaazSk, 2019-06-20:
I will just check it and will let you know, else the other are working fine with me.
There is a option of Network ID and device ID also in LoRa can you add it in your Library?

jgromes, 2019-06-20:
LoRa PHY doesn't have the concept of address - network/device ID are concepts from LoRaWAN.

LoRa packet looks like this:
[image]

If you want some address-based filtering, you have to implement it on user side.

MaazSk, 2019-06-22:
Okay I will try and lets see 
Thank You!