jgromes/LoRaLib #84: Use of D I/O

MaazSk, 2019-07-03:
Hi, 
I have a question can you please explain me the use of D I/O pins of the LoRa module as I can't understand it from the datasheet 

Thank you.

jgromes, 2019-07-03:
The DIO pins are used to signal states of the radio module. For example, in receive mode, when packet is received, DIO0 goes high. If the receive mode times out without packet reception, DIO1 goes high. This is useful as external interrupts for the microcontroller.

DIO state and configuration is handled by the library directly, and users are highly discouraged from manipulating them from Arduino code.

closed 2019-07-03 by jgromes

MaazSk, 2019-07-03:
What is the use of DIO3, DIO4, DIO5?

jgromes, 2019-07-03:
These - as well as DIO2 - are unused in the library.

It depends on the current mode and configuration. See tables 18, 29 and 30 in SX1278 datasheet.

MaazSk, 2019-07-03:
Okay 
Thank you!

MaazSk, 2019-07-03:
Can you explain me How to set DIOx, there is only one pin how can I set it to 00, 01, 10, 11.

jgromes, 2019-07-03:
Those are the values that are supposed to be written to the SPI registers. For example, to set DIO0 to CadDone event, you have to be in LoRa mode and write value 0b10 to register RegDioMapping1 (0x40) to bits 7 and 6.

MaazSk, 2019-07-03:
Ohh Okay
Thank You!