jgromes/LoRaLib #85: LoRa readData and other functions using getPacketLength in SX127X modules not working

ddrmax, 2019-07-03:
**Describe the bug**
Each function using getPacketLength in the SX127X module seems to be broken

**To Reproduce**
Board tested: 
scketch: receive interrupt  either with readdata(str) or readdata(array, lenght)
the sketch is modified like this:
#define SS 8
#define DI0 7
SX1276 lora = new LoRa(SS,DI0,5);
int state = lora.begin(868,125,7,5,0x12);

**Expected behavior**
lora.readData(str) returns a valid string and not an empty string
lora.getPacketLength() returns the size of the data and not always 256
lora.readData(array, lenght) returns the good data (exemple with a sender sending "hello world X" with X a counter, the counter stays a 1.
It shows as well that the dumping of the FIFO doesn't work either
a simple workaround and test working in my case:
created this function in SX127X.cpp/.h:
size_t SX127x::getLoRaPacketLength()
```
{
return(_mod->SPIreadRegister(SX127X_REG_RX_NB_BYTES));	
}
```
and modified the int16_t SX127x::readData(uint8_t* data, size_t len) function with:
```
  // dump bytes that weren't requested
  size_t packetLength = getLoRaPacketLength();
  if(packetLength > length) {
    clearFIFO(packetLength - length);
  }
```
I checked the original getPacketLength and I doesn't see anything wrong with it at first sight
tinkering with it (commenting private an protected)
I get 128 (the value of SX127X_LORA) from getActiveModem();
and I get 6 for _sf
so the problem comes from another function that does not set _sf properly (in my case it had to be 7) and a test with other Spread factor stays the same (with the sketch where I only change the frequency) 
**Additional info :**
 - MCU: Feather 32u4 lora board / Lora32u4II
 - LoRa module type SX1276 (HPD13)
 - Arduino IDE version  1.8.9
 - Library version 8.1.0
 - Set up of the lora System:
     Frequency: 868Mhz
     Bandwith: 125Khz
     Coding rate: 4/5
     CRC either enabled or disabled

jgromes, 2019-07-03:
Perhaps this is to blame? https://github.com/jgromes/LoRaLib/commit/70dd1dc79cda88b36d29d59d8c471def6ebfb5ff

ddrmax, 2019-07-03:
when i searched for _sf in SX127X.cpp the startReceive() function is broken here:
```
    if(_sf = 6) {
      state |= _mod->SPIsetRegValue(SX127X_REG_PAYLOAD_LENGTH, len);
    }
```
it needs to be:
```
    if(_sf == 6) {
      state |= _mod->SPIsetRegValue(SX127X_REG_PAYLOAD_LENGTH, len);
    }
```

ddrmax, 2019-07-03:
When Going down to 8.0.6 anithing works when using readData(str)

ddrmax, 2019-07-03:
with the cloned repo it works nice, the maybe time to bump to 8.1.1 to prevent other users from having the problem? and sorry ,I havent seen you replied becauseI havent refreshed the page before posting the lasts comments

jgromes, 2019-07-03:
Will do that in a few minutes, I just quite simply forgot to release new version after that commit.

Thanks for reporting!

closed 2019-07-03 by jgromes