jgromes/LoRaLib #86: SX127x not found! Error code -2

mark9599, 2019-07-22:
I am using Arduino UNO with SX1278 LoRa module with a level shifter.
I have checked connections thoroughly no problems there.

Code:-
/*
   LoRaLib Channel Activity Detection Example

   This example scans the current LoRa channel and detects
   valid LoRa preambles. Preamble is the first part of
   LoRa transmission, so this can be used to check
   if the LoRa channel is free, or if you should start
   receiving a message.

   For more detailed information, see the LoRaLib Wiki
   https://github.com/jgromes/LoRaLib/wiki

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/LoRaLib/
 */

#include <LoRaLib.h>
// NSS pin:   10 
// DIO0 pin:  2
// DIO1 pin:  3
SX1278 lora = new LoRa;

void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("Initializing ... "));
  // carrier frequency:           434.0 MHz
  // bandwidth:                   125.0 kHz
  // spreading factor:            9
  // coding rate:                 7
  // sync word:                   0x12
  // output power:                17 dBm
  // current limit:               100 mA
  // preamble length:             8 symbols
  // amplifier gain:              0 (automatic gain control)
  int state = lora.begin();
  if(state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }
}

void loop() {
  Serial.print(F("Scanning channel for LoRa preamble ... "));

  // start scanning current channel
  int state = lora.scanChannel();

  if(state == PREAMBLE_DETECTED) {
    // LoRa preamble was detected
    Serial.println(F(" detected preamble!"));

  } else if(state == CHANNEL_FREE) {
    // no preamble was detected, channel is free
    Serial.println(F(" channel is free!"));

  }

  // wait 100 ms before new scan
  delay(100);
}

Serial Monitor:-
SX127x not found! (10 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0012
No SX127x found!
failed, code -2

[image]

Can you suggest something please?

jgromes, 2019-07-22:
The debug output shows that SPI always returns either 0x00 or 0xFF, which means SPI communication isn't working properly. That's most likely a hardware problem.

What level shifter are you using and what are the connections?

mark9599, 2019-07-22:
Connections:-
UNO    LoRa
13 ----> SLCK
12 ----> MISO
11 ----> MOSI
10 ----> NSS
9 -----> RST
3.3V -----> Vcc
GND -----> GND
2 -------> DIO0
3 -------> DIO1

I'm using 4 Channel level shifter
[image]

jgromes, 2019-07-22:
1. What are the connections on the level shifter?
2. You don't have to connect RST - the library doesn't use it.
3. DIO0 doesn't have to go through the shifter - it's the SX1278 sending the data.
4. SCK, on the other hand, should - Arduino is the source of the clock signal, so it's at 5 V.

mark9599, 2019-07-22:
Level Shifter connections
UNO                                             LoRa
HV->5V                                        LV->3.3V
HV1=11(MOSI)                             LV1=MOSI
HV2=NC                                       LV2=NC
HV3=10(NSS)                               LV3=NSS
HV4=13(SCK)                                LV4=SLCK

[image]

jgromes, 2019-07-22:
Is the ground connected?

mark9599, 2019-07-22:
LoRa's Vcc is shorted with the LV pin which is connected to UNO's 3.3V
Yes the Ground is connected from UNO to LoRa

jgromes, 2019-07-22:
Wiring seems good then, so I suspect hardware issue - either poor soldering contact or damaged LoRa module.

In any case, this is unrelated to the library. I suggest asking at the Arduino forum for more general help.

closed 2019-07-22 by jgromes