jgromes/LoRaLib #88: FSK mode for RFM95

yannpom, 2019-08-09:
Hello,

I've been playing with your LoRaLib and it works great. I currently have 2 ESP32 + RFM95W. I'm building with ESP-IDF. I had some difficulties to make your Arduino code work with ESP-IDF but it's finally working.

Now I would like to switch to the FSK modem of the RFM95 but I see it's not supported by your lib. Did you have some difficulties adding FSK support for RFM95 or is it just because it was not on your priority list? If it's the latter, I'm willing to help. Because I really need FSK for RFM95.

It looks feasible, reading the RFM95 datasheet and adapting your FSK code from SX1278 (for example) to RFM95? Do you have any other advices?

Thanks,
Yann.

jgromes, 2019-08-09:
What gave you the impression FSK is not supported for RFM9x? RFM9x is the exact same chip as SX127x, only packaged by a different company. As such, all features of this library work on both SX127x series as well as on RFM9x.

closed 2019-08-09 by jgromes

yannpom, 2019-08-09:
Well, my bad, you are right. I got confused reading this wiki page: https://github.com/jgromes/LoRaLib/wiki/FSK-Modem
First line says "LoRaLib supports FSK transmissions using SX127x chips." and I thought RFM chips were excluded. But it actually works great!

jgromes, 2019-08-09:
I amended the wiki page to explicitly mention RFM9x as well as SX127x, but in general, whenever any part of the documentation says "SX127x" it usually means RFM9x too, since there's no essential difference between the two chip series.