jgromes/LoRaLib #89: Interrupt transmit does not work

bakerw71, 2019-09-01:
- MCU: ESP8266 (NODE MCU v0.9)
 - LoRa module type SX1278
 - Arduino IDE version  v1.8.8
 - Library version 8.1.1

No activity on DIO0 or DIO1 pins of LORA module
Tested using a modified version of your example "Transmit Interrupt"
DIO0 on D2 = GPIO 4
DIO1 on D1 = GPIO 5

NOTE: Minor glitch where setSyncWord returns an error despite working fine

`#include <LoRaLib.h>

SX1278 lora = new LoRa(15, 4, 5, SPI);

void setup() {
	Serial.begin(115200);
	if (lora.setBandwidth(500.0) == ERR_INVALID_BANDWIDTH) {
		Serial.println(F("Selected bandwidth is invalid for this module!"));
		while (true);
	}
	if (lora.setSpreadingFactor(6) == ERR_INVALID_SPREADING_FACTOR) {
		Serial.println(F("Selected spreading factor is invalid for this module!"));
		while (true);
	}

	// set coding rate to 6
	if (lora.setCodingRate(5) == ERR_INVALID_CODING_RATE) {
		Serial.println(F("Selected coding rate is invalid for this module!"));
		while (true);
	}

	// set carrier frequency to 433.5 MHz
	if (lora.setFrequency(433.5) == ERR_INVALID_FREQUENCY) {
		Serial.println(F("Selected frequency is invalid for this module!"));
		while (true);
	}

	// set LoRa sync word to 0x14
	// NOTE: value 0x34 is reserved for LoRaWAN networks and should not be used
	if (lora.setSyncWord(0x11) != ERR_NONE) {
		Serial.println(F("Unable to set sync word!"));
		//	while (true);
	}

	// set output power to 10 dBm (accepted range is -3 - 17 dBm)
	// NOTE: 20 dBm value allows high power operation, but transmission
	//       duty cycle MUST NOT exceed 1%
	if (lora.setOutputPower(17) == ERR_INVALID_OUTPUT_POWER) {
		Serial.println(F("Selected output power is invalid for this module!"));
		while (true);
	}

	// set over current protection limit to 80 mA (accepted range is 45 - 240 mA)
	// NOTE: set value to 0 to disable overcurrent protection
	if (lora.setCurrentLimit(200) == ERR_INVALID_CURRENT_LIMIT) {
		Serial.println(F("Selected current limit is invalid for this module!"));
		while (true);
	}

	// set LoRa preamble length to 15 symbols (accepted range is 6 - 65535)
	if (lora.setPreambleLength(6) == ERR_INVALID_PREAMBLE_LENGTH) {
		Serial.println(F("Selected preamble length is invalid for this module!"));
		while (true);
	}

	// set amplifier gain to 1 (accepted range is 1 - 6, where 1 is maximum gain)
	// NOTE: set value to 0 to enable automatic gain control
	//       leave at 0 unless you know what you're doing
	if (lora.setGain(0) == ERR_INVALID_GAIN) {
		Serial.println(F("Selected gain is invalid for this module!"));
		while (true);
	}

	// disable automatic CRC check on received packets
	// NOTE: packets that do not pass the check are discarded automatically
	if (lora.setCRC(false) != ERR_NONE) {
		Serial.println(F("Unable to set CRC!"));
		while (true);
	}

	
	Serial.print(F("Initializing ... "));
	
	int state = lora.begin();
	if (state == ERR_NONE) {
		Serial.println(F("success!"));
	}
	else {
		Serial.print(F("failed, code "));
		Serial.println(state);
		while (true);
	}

	lora.setDio0Action(setFlag);
	lora.setDio1Action(setFlag);
}

// flag to indicate that a packet was received
volatile bool transmittedFlag = true;

void ICACHE_RAM_ATTR setFlag(void) {
	// packet transmission is finished, set the flag
	transmittedFlag = true;
	Serial.println("DIO RISING!");
}

uint16_t counter = 0;

void loop() {
	// check if the previous transmission finished
	if (transmittedFlag) {
		Serial.println(F("transmission finished!"));

		// wait one second before next transmission
		delay(1);

		// send another packet
		Serial.print(F("Sending another packet ... "));

		// you can transmit C-string or Arduino string up to
		// 256 characters long
		int state = lora.startTransmit((uint8_t*)&counter,2);

		// you can also transmit byte array up to 256 bytes long
		/*
		  byte byteArr[] = {0x01, 0x23, 0x45, 0x56,
							0x78, 0xAB, 0xCD, 0xEF};
		  int state = lora.transmit(byteArr, 8);
		*/

		// NOTE: when using interrupt-driven transmit method,
		//       it is not possible to automatically measure
		//       transmission data rate using getDataRate()

		if (state != ERR_NONE) {
			Serial.print(F("failed, code "));
			Serial.println(state);
		}
		transmittedFlag = false;
		counter++;
	}

}`

jgromes, 2019-09-02:
First of all, please post the minimal required code to reproduce the issue. Second, you're calling a lot of configuration methods (`setBandwidth()` etc.) before calling the initialization method (`begin()`) - I'm a bit surprised that is working at all, the `begin()` method should always be the first thing you call (see https://github.com/jgromes/LoRaLib/wiki/Basics).

Also, what error is returned with `setSyncWord()`? It might be a good idea to open a separate issue for that, if it turns out it's unrelated to the main problem.

bakerw71, 2019-09-05:
Awsome thanks that fixed both problems!
Can't believe it was that simple.
Really appreciate the library especially after writing a small one of my own for the SX1278, Thanks!

TienHuyIoT, 2019-09-17:
Dear jgromes,

I had same problem with Tx_interrupt.

I run sketch example TransmitInterrupt.ino base on sx1278 module. It just be success the first time after run sketch or reset board. And then it always fail.

Log serial:
    Sending first packet ... 
    transmission finished!
    Sending another packet ... 
    failed, code xxxx
    transmission finished!
    Sending another packet ... 
    failed, code xxxx
....

Update:
 I think have misstake in sketch TransmitInterrupt.ino
`// check if the previous transmission finished
  if(transmittedFlag) {    
    Serial.println(F("transmission finished!"));
    .......
  }`
Should be add "transmittedFlag = flase;" after if(transmittedFlag){}

jgromes, 2019-09-17:
You're right, also the flag to disable new interrupts is missing. Well spotted, thanks.

jgromes, 2019-09-17:
Fixed in v8.1.2, thanks for reporting!

closed 2019-09-17 by jgromes

TienHuyIoT, 2019-09-17:
Dear @jgromes ,

After update, i still got the above error . I had review lib and found in file module.cpp at line 85 should be "if((readValue & mask) == (value & mask))" instead of "if(readValue == newValue)".
Because we only need to compare any bits by mask.

jgromes, 2019-09-17:
`newValue` is already masked, so it should be enough to mask `readValue` - more importantly, why do you think this is the issue?

TienHuyIoT, 2019-09-18:
Dear @jgromes ,

When reading a register maybe not same the register written before. Because a register has some bits only can read. ex, the register RegOpMode(0x01) has msb bit "LongRangeMode". "This bit can be modified only in Sleep mode. A write operation on
other device modes is ignored." 
So when we write, this bit can be 0 but then we read it can be 1 --> compared fail. I got this fail when Lora module into sleep mode, I read register 0x01 and get value of MSB bit is 0. I set Standy mode to Lora Module and then read register 0x01 again, the msb bit of register 0x01 now is 1.

If i have wrong somthing. May you tell me.
Thank so much your Lora library.

jgromes, 2019-09-18:
I think you may have misunderstood my question - why do you think the bit masking is related to your problem with TransmitInterrupt not working? I am running that example on Arduino Uno without any issues, so I don't think the bit masking is the main problem.