jgromes/LoRaLib #90: reset on ESP32 Arduino code when instance is changed

sushant-ht, 2019-09-17:
**IMPORTANT: Before submitting an issue, please check the following:**
1. **Read [CONTRIBUTING.md](https://github.com/jgromes/LoRaLib/blob/master/CONTRIBUTING.md)!** Issues that do not follow this document will be closed/locked/deleted/ignored.
2. LoRaLib has a [Wiki](https://github.com/jgromes/LoRaLib/wiki). You might find a solution to your issue there.
3. Make sure you're using the latest release of the library! Releases can be found [here](https://github.com/jgromes/LoRaLib/releases).
4. Use [Arduino forums](https://forum.arduino.cc/) to ask generic questions about SX1278 wiring, usage, etc. Only create issues for problems specific to LoRaLib!
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/LoRaLib/group__status__codes.html).
Hi, thanks for library,
I am using LoraLib to connect my esp32 Devkit(wroom devkit) with sx1278 Lora module,
i came across an issue, i am unclear whether its library or my own coding,
as per settings example i can modify IO pins for NSS, GPIO0, GPIO1 for Lora,

hence i modified below lines in transmitInterrupt example,

i have changed, 
`SX1278 lora = new LoRa;`
to
`
// include SPI library
#include <SPI.h>

//Default pins SCLK = 14, MISO = 12, MOSI = 13, SS = 15
SPIClass hspi = new SPIClass(HSPI);
// create new instance
SX1278 lora = new LoRa(4, 2, 3, (SPIClass&)&hspi);
`
now when i use this code i get in reset loop 

> 

> Rebooting...
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1100
> load:0x40078000,len:9232
> load:0x40080400,len:6400
> entry 0x400806a8
> Initializing ... /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:1443 (xQueueGenericReceive)- assert failed!
> abort() was called at PC 0x40087db5 on core 1
> 
> Backtrace: 0x4008b46c:0x3ffb1de0 0x4008b699:0x3ffb1e00 0x40087db5:0x3ffb1e20 0x400d3779:0x3ffb1e60 0x400d2981:0x3ffb1e80 0x400d0fce:0x3ffb1eb0 0x400d1059:0x3ffb1ee0 0x400d1cbc:0x3ffb1f10 0x400d20d6:0x3ffb1f30 0x400d1bcf:0x3ffb1f50 0x400d0e1c:0x3ffb1f80 0x400d432f:0x3ffb1fb0 0x40089fb9:0x3ffb1fd0

if i change back to , 
`SX1278 lora = new LoRa;`

it works but i cant work with my kit, i also wish to know, what are default wiring for 
`SX1278 lora = new LoRa;`
SPI IO pins from ESP32 to Lora 

Thanks
Sushant

jgromes, 2019-09-17:
The default SPIClass used is `SPI` - I'm not sure which pins ESP32 uses in that case. What happens when you try `SX1278 lora = new LoRa(4, 2, 3);`?

Please post the decoded stack trace.

TienHuyIoT, 2019-09-17:
Dear @sushant-ht ,

You may be declared HSPI as follows.
```
SPIClass * hspi = new SPIClass(HSPI);
SX1278 lora = new LoRa(SX1278_NSS_PIN, SX1278_DIO0_PIN, SX1278_DIO1_PIN, *hspi);
void setup() {
hspi->begin(SD_SCK_PIN, SD_MISO_PIN, SD_MOSI_PIN, SX1278_NSS_PIN);
...
lora.begin();
}
```

jgromes, 2019-10-02:
Closed due to inactivity, feel free to reopen later.

closed 2019-10-02 by jgromes